machinekit/machinekit-hal #16: machinekit (c)python module not found in machinekit-hal

luminize, 2017-08-22:
The (c)python bindings do not work with the machinekit-hal package.
I installed this on a recent image
apt-get install machinekit-hal removed the machinekit package.
```
machinekit@beaglebone:~/mk-xyz-config$ ./run.py
Traceback (most recent call last):
  File "./run.py", line 8, in <module>
    import xyzConfiguration as configuration
  File "/home/machinekit/mk-xyz-config/xyzConfiguration.py", line 5, in <module>
    from machinekit import hal
ImportError: No module named machinekit
```

happy to work on this if somebody can point me a bit in the proper direction

ArcEye, 2017-08-23:
Can you point me to where it is in a RIP build, should be just a case of changing some of the packaging rules.

luminize, 2017-08-23:
I've searched for "machinekit" in a RIP build (not very recent, but there's not a lot of changes iirc)
```
machinekit@bbb-mk-02:~/machinekit$ find ./ -name machinekit
./lib/python/machinekit
./scripts/machinekit
./bin/machinekit
./src/hal/cython/machinekit
./src/objects/hal/cython/machinekit
```

ArcEye, 2017-08-23:
I think the libraries are actually called hal.so / compat.so / rtapi.so / shmcommon.so
and should be in lib/python/machinekit

They should be copied by the .install rules

Testing now.

ArcEye, 2017-08-23:
OK found it

In the Makefile, I simplified the original testing process by 
```
ifeq ($(BUILD_PARSEL_TONGUE),yes)
SUBDIRS += hal/cython
endif
```
Then I did not reinstate it and because I never use python, did not notice the absence of slithery things🐍 

Fixed and re-testing

ArcEye, 2017-08-23:
Had to change the rules.in to ensure they got included for some reason, but that now works.

In the interim, I am going to change the Submakefile rule preventing pru_generic.bin being created
as per #17 and you can test whether that works too.

2 tweets with 1 rock :)

PR follows in a little while

PR #18 (2017-08-23): Fix python library builds and inclusion in packages
Test pru_generic.bin creation for rt-preempt armhf

Remove mkwrapper from tree

Fixes #16 and will test solution to #17
closed 2017-08-23 by luminize