machinekit/machinekit-hal #17: pru_generic.bin missing when install machinekit-hal-rt-preempt package on Beaglebone

luminize, 2017-08-22:
following this issue https://github.com/machinekit/machinekit/issues/1238 I've installed the machinekit-hal package on the Beaglebone black wireless.
However it seems pru_generic is missing.

ArcEye, 2017-08-23:
That is because you installed the rt-preempt package

There is a rule in Makefile which only builds pru_generic.bin and pru_decamax.bin for xenomai kernels.
That is exactly the same in machinekit
```
# .bin files are targets
# Adding to TARGETS builds the PRU code once and puts it in the RTLIBDIR for
# the first defined RTOS flavor (typically posix)
#TARGETS +=  $(PRU_BIN) $(PRU_DBG)
# Only build PRU code for the Xenomai RTOS flavor
ifeq ($(threads),xenomai)
modules : $(PRU_BIN) $(PRU_DBG)
endif
```

Need advice from @cdsteinkuehler as to why this is.
Is it simply historical and no-one was expected to run a BBB on rt-preempt, or is there a deeper issue?

ArcEye, 2017-08-24:
I have removed the attempt to build all flavours with pru_generic.bin etc in the first PR.

It should now build and you have your python bindings back

There were peculiar build problems, due I think to the build flow being disturbed resulting in attempts to copy the files to a rtlib flavour dir which had yet to be created.

Besides I seem to recall @cdsteinkuehler saying that these files only need building once.  
Suspect they have a binary format which is irrespective of the host kernel flavour.

So perhaps another approach is needed, I'll experiment on the development version of the CI builder.

PR #19 refers

ArcEye, 2017-08-24:
@luminize 
The pru_*.* files, built natively on armhf, are here http://mgware.co.uk/staging/

It would be interesting to see what happens when you add them to /usr/lib/linuxcnc/rt-preempt
and then run the BBB config with the cape.

If that does not work, there is no point continuing with getting them into package builds.

RobertCNelson, 2017-08-24:
@luminize in that script, we need to nuke the "kernel" side slot loading, it needs to be moved to u-boot:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

(cape universal is automatically loaded in u-boot now days)

The AIN0 check, needs to use the iio directory

Regards,

RobertCNelson, 2017-08-24:
btw, you can use:

```
grep -c bone_capemgr.uboot_capemgr_enabled=1 /proc/cmdline
```
to validate u-boot "did something", thus assume cape universal is loaded...

Regards,

luminize, 2017-08-25:
@RobertCNelson the device was still booting from the eMMC, I've used `sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k` as Daren Schwenke mentioned in the google group thread https://groups.google.com/forum/#!topic/machinekit/RrNLUo4ASP4

thus:
```
machinekit@beaglebone:/opt/scripts/tools$ grep -c bone_capemgr.uboot_capemgr_enabled=1 /proc/cmdline
1
```

I do not use the setup script anymore, and I can see that the pru_generic.bin is used because the hpg component is being properly set up in HAL
```
machinekit@beaglebone:~/mk-xyz-config$ halcmd show pin hpg
Component Pins:
  Comp   Inst Type  Dir         Value  Name                                            Epsilon Flags  linked to:
   125        s32   OUT             0  hpg.capture-position.time               		----
   125        s32   I/O             0  hpg.capture-position.tmax               		----
   125        bit   OUT         FALSE  hpg.capture-position.tmax-inc           		----
   125        bit   IN          FALSE  hpg.stepgen.00.control-type             		--l-
   125        s32   OUT             0  hpg.stepgen.00.counts                   		--l-
   125        float OUT             0  hpg.stepgen.00.dbg_err_at_match         	0.000010	--l-
   125        float OUT             0  hpg.stepgen.00.dbg_ff_vel               	0.000010	--l-
   125        float OUT             0  hpg.stepgen.00.dbg_pos_minus_prev_cmd   	0.000010	--l-
   125        float OUT             0  hpg.stepgen.00.dbg_s_to_match           	0.000010	--l-
   125        s32   OUT             0  hpg.stepgen.00.dbg_step_rate            		--l-
   125        float OUT             0  hpg.stepgen.00.dbg_vel_error            	0.000010	--l-
   125        u32   IN     0x00000064  hpg.stepgen.00.dirhold                  		--l-
   125        u32   IN     0x00000011  hpg.stepgen.00.dirpin                   		--l-
   125        u32   IN     0x00000064  hpg.stepgen.00.dirsetup                 		--l-
   125        bit   IN          FALSE  hpg.stepgen.00.enable                   		--l-
   125        float IN              1  hpg.stepgen.00.maxaccel                 	0.000010	--l-
   125        float IN              0  hpg.stepgen.00.maxvel                   	0.000010	--l-
...
...
   125        u32   IN     0x00000064  hpg.stepgen.04.steplen                  		--l-
   125        u32   IN     0x00000011  hpg.stepgen.04.steppin                  		--l-
   125        u32   IN     0x00000064  hpg.stepgen.04.stepspace                		--l-
   125        s32   OUT             0  hpg.stepgen.04.test1                    		--l-
   125        s32   OUT             0  hpg.stepgen.04.test2                    		--l-
   125        s32   OUT             0  hpg.stepgen.04.test3                    		--l-
   125        float IN              0  hpg.stepgen.04.velocity-cmd             	0.000010	--l-
   125        float OUT             0  hpg.stepgen.04.velocity-fb              	0.000010	--l-
   125        s32   OUT             0  hpg.update.time                         		----
   125        s32   I/O             0  hpg.update.tmax                         		----
   125        bit   OUT         FALSE  hpg.update.tmax-inc                     		----

```

luminize, 2017-08-25:
@ArcEye Everything seems to run by putting the .bin files in the proper directory

ArcEye, 2017-08-25:
OK.
AFAIR, moving pru_generic.bin etc into all builds was tried previously and caused problems with failed builds and had to be reverted.
It works fine as a RIP, there is something in the packaging that needs changing.

I'll have another look later.

ArcEye, 2017-08-27:
The problem with the build, was the way that the multi-flavour builds work, did not mesh with the atypical pru_generic build, which only targets one flavour (xenomai).
Removing the flavour restriction, resulted in binaries being built out of sequence ,when there was no receiving directory for that flavour yet created.  This is what also caused the issue in #1199 when the same was attempted.

As discussed in https://github.com/machinekit/machinekit/issues/1199#issuecomment-292630938 and elsewhere, the binary pru_generic.bin etc. files are flavour agnostic and only really need to be built once.

My solution is to build them for posix, which should always be selected and then copy the files across to the correct directory and tidy up in a postinst file in the package.
This may be an interim step towards the removal of the flavour dirs and just having one set of modules - but that is for another day 😄 

I will before long port it across to machinekit, which has the same requirement to use rt-preempt on BBB, without having to manually copy the pru_generic.* files across.

PR to follow once polished
Already tested on install

ArcEye, 2017-08-27:
Fixed by #26

PR #26 (2017-08-27): Move PASM generated pru_generic.bin etc. into all flavours
Was restricted to xenomai originally and there is now a requirement
to be able to run BBB on rt-preempt kernels, which also require
the files

Since they are of a binary format which is not flavour dependent
they are now built once for posix, incorporated into each
package, to be moved to the correct location upon install

Fixes #17
closed 2017-08-27 by ArcEye