machinekit/machinekit-hal #21: configserver is missing

luminize, 2017-08-24:
```python
from machinekit import launcher
launcher.check_installation()
```

gives the error:
```
configserver not found, check Machinekit installation
```

ArcEye, 2017-08-25:
Found the problem and discovered the answer to another which I fixed another way earlier.

I removed a lot of the python stuff that appeared to relate to the running of a CNC machine.
Need to reinstate some of it to the Makefile.

ArcEye, 2017-08-25:
Questions for you then.

I removed all the stuff relating to thermistor tables, fdm, hal_temp etc. because that was related to
using machinekit to run a CNC machine, ie a 3D printer.

Is it required just for HAL?

luminize, 2017-08-25:
@ArcEye that's a good one.

Regarding hal_temp_bbb for example (https://github.com/machinekit/Machinekit-HAL/tree/master/src/hal/user_comps) This is a user component, and although it is used in FDM printing, it's basically a specialized component for a beaglebone, converting ADC thermistor values to a temperature. Therefore I see this as a component and thus residing in machinekit-hal.
It refers to fdm stuff like here https://github.com/machinekit/Machinekit-HAL/blob/master/src/hal/user_comps/hal_temp_bbb.py#L54 tohttps://github.com/machinekit/Machinekit-HAL/tree/master/lib/python/fdm The directory "fdm" should go IMO, because that is used for building the configuration.

Logic/programs which builds configurations (a lot of the python fdm stuff does that) are part of the application built on top of machinekit-hal, so "stuff" which generates hal configuration or user applications  interacting with HAL should not be in machinekit-hal imo.

Tools like halscope/halmeter/hardware configuration stuff etc, should be in machinekit-hal though. As opposed to applications like Axis etc.

We should have machinekit-hal application agnostig, but filled with the building blocks for building applications. Right?

@machinekoder @cdsteinkuehler what's your take?

ArcEye, 2017-08-25:
I have just done a PR #25 which puts a lot of the python stuff back in the packages.
(it was still being built in a RIP, just not installed)

The dir 'fdm' is back for now, because it occurs that it might be used where machinekit-hal was attached to another interpreter.  It is hard to speculate what uses might be found, it can always be removed.

All the BBB and temp specific stuff is back in too.

ArcEye, 2017-08-28:
configserver back in the package

closed 2017-08-28 by ArcEye