machinekit/machinekit-hal #22: error initiating jplan component

luminize, 2017-08-25:
It seems there were some changes in protobuf.

comparing machinekit with machinekit-hal I can see that from this PR https://github.com/machinekit/machinekit/pull/1229 only this commit https://github.com/machinekit/machinekit/pull/1229/commits/e81486ab977f669254463e27237ce083a44c8c46 has been teken over into https://github.com/machinekit/Machinekit-HAL/commit/fc6834ebcc9d5208f8b8fa079dd61b78a66ecd8c but not the other commits in the PR.

I'll cherry-pick these other 2 commits and make a PR.

luminize, 2017-08-25:
@machinekoder I'm running into merge errors

```
machinekit@beaglebone:~/Machinekit-Hal$ git cherry-pick a1ff33091652f3b70143b2510c6d35929de80782
error: could not apply a1ff330... Squashed 'src/machinetalk/proto/' changes from 211433bfb..f4481f561
hint: after resolving the conflicts, mark the corrected paths
hint: with 'git add <paths>' or 'git rm <paths>'
hint: and commit the result with 'git commit'
machinekit@beaglebone:~/Machinekit-Hal$ git status
On branch merge-back-upstream-protobuf
You are currently cherry-picking commit a1ff330.
  (fix conflicts and run "git cherry-pick --continue")
  (use "git cherry-pick --abort" to cancel the cherry-pick operation)

Changes to be committed:

	new file:   MANIFEST.in
	new file:   setup.cfg
	new file:   src/machinetalk/protobuf/jplan.proto
	new file:   src/machinetalk/protobuf/ros.proto

Unmerged paths:
  (use "git add/rm <file>..." as appropriate to mark resolution)

	both modified:   .gitignore
	both modified:   README.md
	deleted by us:   dist/machinetalk-protobuf.min.js
	deleted by us:   dist/machinetalk-protobuf.min.js.gz
	deleted by us:   dist/machinetalk-protobuf.min.map.json
	both deleted:    python/setup.py
	added by them:   setup.py
	added by us:     src/machinetalk/proto/setup.py
```

can you advise how to best get machinekit-hal up to par w.r.t this?

ArcEye, 2017-08-25:
If you look at https://github.com/machinekit/Machinekit-HAL/commit/fc6834ebcc9d5208f8b8fa079dd61b78a66ecd8c and especially https://github.com/machinekit/Machinekit-HAL/commit/cb942a632be13f70a66aa2df2d62b2a7c174c467
this is probably where something was missed.

The original merge in machinekit was from a convoluted squashing and merge in another repo and just did not work.
I ended up picking out the elements

If you already have a file which the cherry pick thinks is a new one etc. it causes all sorts of problems.
jplan and ros are in there already, so it should be just modifying.
Looks like the path is different, which is right ? 

Someone buggered about with all the machinetalk paths a while back (no names no packdrill)
https://github.com/machinekit/Machinekit-HAL/blob/master/src/machinetalk/proto/src/machinetalk/protobuf/jplan.proto
https://github.com/machinekit/Machinekit-HAL/blob/master/src/machinetalk/proto/src/machinetalk/protobuf/ros.proto

luminize, 2017-08-25:
I missed your takeover. This is a pita, i hoped there was an easy way. I'll find my goggles and wet suit and dive into it later.
To be continued...
Thx

ArcEye, 2017-08-25:
I would advise having a good look at what is in machinekit and comparing to that.
The two should mirror where they contain the same.

The problem is likely to be that mk-hal was taken from mk before all the path changing and class name changing of machinetalk in mk, and some of them did not align again.

I'll look back again when I have fixed the other stuff later.

luminize, 2017-08-25:
just for historical purposes:
```
Aug 25 12:55:35 beaglebone rtapi:0: 4:rtapi_app:18806:user pbmsgs: loaded from pbmsgs.so
Aug 25 12:55:35 beaglebone rtapi:0: do_load_cmd: dlopen: /usr/lib/linuxcnc/rt-preempt/jplan.so: undefined symbol: machinetalk_JplanCommand_fields
Aug 25 12:55:35 beaglebone rtapi:0: rpath=/usr/lib/linuxcnc/rt-preempt
Aug 25 12:55:35 beaglebone rtapi:0: 1:rtapi_app:18806:user do_load_cmd: dlopen: /usr/lib/linuxcnc/rt-preempt/jplan.so: undefined symbol: machinetalk_JplanCommand_fields
Aug 25 12:55:35 beaglebone rtapi:0: 1:rtapi_app:18806:user rpath=/usr/lib/linuxcnc/rt-preempt
Aug 25 12:55:35 beaglebone rtapi:0: newinst: component 'jplan' not loaded
Aug 25 12:55:35 beaglebone rtapi:0: 1:rtapi_app:18806:user newinst: component 'jplan' not loaded
```

I'll have a look at the fields

ArcEye, 2017-08-25:
Comparing machinekit and machinekit-hal
```
/usr/src/machinekit/rtlib/posix# nm -C jplan.so | grep machinetalk_JplanCommand_fields
                 U machinetalk_JplanCommand_fields

/usr/src/machinekit-hal/rtlib/posix# nm -C jplan.so | grep machinetalk_JplanCommand_fields
                 U machinetalk_JplanCommand_fields

for x in *.so; do nm -C $x | grep machinetalk_JplanCommand_fields; echo $x; done
liblinuxcnchal.so
liblinuxcncini.so
liblinuxcncshm.so
liblinuxcnculapi.so
000000000020ddc0 D machinetalk_JplanCommand_fields
libmachinetalk-npb.so
libmachinetalk-pb2++.so
libmtalk.so
libposemath.so
librtapi_math.so

/usr/src/machinekit-hal/lib# nm -C libmachinetalk-npb.so | grep machinetalk_JplanCommand_fields
000000000020ddc0 D machinetalk_JplanCommand_fields

/usr/src/machinekit/lib# nm -C libmachinetalk-npb.so | grep machinetalk_JplanCommand_fields
000000000020dda0 D machinetalk_JplanCommand_fields
```

So the libmachinetalk-npb.so holds that routine in both

Don't know what is different in machinekit-hal, the linking?

luminize, 2017-08-25:
visual comparison of commits (apart from the jplan.c and interpolate.c) between
machinekit
https://github.com/machinekit/Machinekit-HAL/commit/cb942a632be13f70a66aa2df2d62b2a7c174c467
and
machinekit-hal
https://github.com/machinekit/machinekit/pull/1229/files
yields that this is missing in machinekit-hal (picture from machinekit repo)
[image]

I also looked at the proto defenitions themselves, what I can see differs from a not up-to-date branch on my repoo is that there's a 
```
syntax = "proto2";
package machinetalk;
```
in the [files of machinekit, machinekit-hal and machinetalk-protobuf](https://github.com/machinekit/machinetalk-protobuf/blob/17178e54fb7782c7559806ed0bdc09e592c8d398/src/machinetalk/protobuf/jplan.proto)

as opposed to
```
package pb;
```
in [the old file](https://github.com/luminize/machinekit/blob/153ddab58efbe6097435b5513bd5c105a474b257/src/machinetalk/proto/src/machinetalk/protobuf/jplan.proto) 

going to take a break.. I'm getting cross-eyed from comparing!

ArcEye, 2017-08-25:
Before you chase your own tail, I would suggest that you install the new package and try again.

I don't know what that MANIFEST.in file does but it is not referred to by any other file in Machinekit (except .git/index)

luminize, 2017-08-25:
Ok, will do!

ArcEye, 2017-08-28:
I have done a diff of `src/machinetalk` between _machinekit_ and _machinekit-hal_

All the locations appear to be the same and the only bits missing or commented out refer to NML or launching CNC.
The only difference between the entire `src/machinetalk/proto` dirs is the absence of that MANIFEST.in file
The .proto files are identical

The jplan.so component is built in the packages

But I can confirm the error which appears to be because jplan is not linked against libmachinetalk-npb.so
```
root@armv7-bb:/usr/src/machinekit-hal# halcmd loadrt jplan
<commandline>:0: insmod failed, returned -1:
do_load_cmd: dlopen: /usr/src/machinekit-hal/rtlib/posix/jplan.so: undefined symbol: machinetalk_JplanCommand_fields
```
This is from a RIP build, so nothing to do with the packaging.

ArcEye, 2017-08-28:
Also note, pbmsgs.so which also uses that symbol, is properly linked
```
root@INTEL-i7:/usr/src/machinekit-hal/rtlib/rt-preempt# nm pbmsgs.so | grep machinetalk_JplanCommand_fields
0000000000208be0 d machinetalk_JplanCommand_fields

root@INTEL-i7:/usr/src/machinekit-hal/rtlib/rt-preempt# nm jplan.so | grep machinetalk_JplanCommand_fields
                 U machinetalk_JplanCommand_fields
```

The hunt continues 😜

ArcEye, 2017-08-28:
@luminize 

Well I have fixed jplan, even though I don't know why it arose
```
mick@INTEL-i7:/usr/src/machinekit-hal$ DEBUG=5 realtime restart
mick@INTEL-i7:/usr/src/machinekit-hal$ halcmd loadrt jplan
<commandline>:0: Realtime module 'jplan' loaded
mick@INTEL-i7:/usr/src/machinekit-hal$ halcmd show pin
Component Pins:
  Comp   Inst Type  Dir         Value  Name                                            Epsilon Flags  linked to:
    78     80 bit   OUT         FALSE  jplan.0.0.active                        		--l-
    78     80 float OUT             0  jplan.0.0.curr-pos                      	0.000010	--l-
    78     80 float OUT             0  jplan.0.0.curr-vel                      	0.000010	--l-
    78     80 bit   IN          FALSE  jplan.0.0.enable                        		--l-
    78     80 float IN              0  jplan.0.0.max-acc                       	0.000010	--l-
    78     80 float IN              0  jplan.0.0.max-vel                       	0.000010	--l-
    78     80 float IN              0  jplan.0.0.pos-cmd                       	0.000010	--l-
    78     80 bit   OUT         FALSE  jplan.0.joints-active                   		--l-
    78     80 s32   OUT             0  jplan.0.update.time                     		----
    78     80 s32   I/O             0  jplan.0.update.tmax                     		----
    78     80 bit   OUT         FALSE  jplan.0.update.tmax-inc                 		----
```
`machinetalk_JplanCommand_fields` is declared `extern` in _jplan.c_ and is actually defined in _jplan.npb.c_,
which is a generated file.

I specifically added `jplan.npb.o` to the jplan-objs in Makefile  ie.
```
obj-m += jplan.o
jplan-objs := hal/jplanner/jplan.o machinetalk/build/machinetalk/protobuf/jplan.npb.o
```
and the resulting `jplan.so` is correctly linked without undefined symbols.

I am rather suspicious of this and must test all the other semi-generated components, to make sure that they too do not have incomplete linkage.

There is something else missing or wrongly placed I think, but I cannot put my finger on it right now.
The obfusticated protobuf build system does not assist at all.

luminize, 2017-08-28:
kudo's to you!

ArcEye, 2017-08-28:
By means of a crude but effective script
```
# !/bin/bash

DEBUG=5 realtime restart ;

for i in rtlib/rt-preempt/*.so ; do
	echo "`basename $i .so`"
	halcmd loadrt `basename $i .so`
	halcmd show pin
	halcmd unloadrt `basename $i .so`
done	

halrun -U
```
I have determined that all the modules load ( with the exception of pwmgen and pwmgenv2, which fail if there are no arguments)

So I will make a PR fixing the jplan issue and we can see what shakes out of the tree next 😄

closed 2017-08-28 by ArcEye

luminize, 2017-08-29:
apt-get update && apt-get upgrade got jplan component working again.
Thx

PR #27 (2017-08-28): Component jplan for some reason was not properly linked.
Symbols within the generated jplan.npb.o
were not being linked correctly

Specific linkage added to Makefile, which produces a jplan
module which loads

Fixes #22