machinekit/machinekit-hal #24: what should stay and what should go

luminize, 2017-08-25:
in addition to this comment https://github.com/machinekit/Machinekit-HAL/issues/21#issuecomment-324897228 so we can come to understanding:

what should stay and what should go?

what needs to go:
- related to GUI's, like
  - https://github.com/machinekit/machinekit/blob/master/lib/python/gladevcp_makepins.py
  - https://github.com/machinekit/machinekit/blob/master/lib/python/hershey.py
  - and so on and so forth
- related to configurations
  - https://github.com/machinekit/Machinekit-HAL/tree/master/lib/python/fdm/config

What needs to stay:
- files related to user components, tools and setup helper programs
  - https://github.com/machinekit/Machinekit-HAL/blob/master/lib/python/fdm/r2temp.py should naming/location should be discussed?

(please edit this comment by adding examples)

ArcEye, 2018-07-29:
The current status is that, machinekit-hal has all the python vcp stuff in it, so that it can use for panels as required, whilst allowing the machinekit-cnc pyvcp and gladevcp elements to function when the
-hal and -cnc packages are combined

In an installation of machinekit-cc over machinekit-hal, machinekit (axis_mm.ini) runs as does machinekit (pyvcp_demo) proving pyvcp is working.

[image]

stepconf and pncconf work too, which use both pyvcp and gladevcp

[image]

[image]

As does latency-test (pyvcp)

[image]

There is some tidying to do, update default changelog, increment version number so that it will be obvious which packages someone is asking questions about and several other odds.

But largely it is there, just needs more testing once merged and packages available.

luminize, 2018-07-29:
Great! One step further.