machinekit/machinekit-hal #28: mklauncher is missing

luminize, 2017-08-29:
Although I can find a reference in the Makefile https://github.com/machinekit/Machinekit-HAL/blob/master/src/Makefile#L206 I get the message 
`starting mklauncher... /bin/sh: 1: mklauncher: not found`
`sudo find / -name mklauncher` does indeed yield nothing

ArcEye, 2017-08-29:
That is because its name suggests that it launches `machinekit`, I took it out of the install side of the Makefile.
Why would you want to launch machinekit in a HAL only package?

Instead it appears what it does is launch `machinetalk`.

Perhaps it should be renamed?  I'll put it back in shortly.

Now you appreciate why I didn't want to do anything until it was tested.
I don't use any of this stuff, so it works fine for me 😝

luminize, 2017-08-29:
well, I don't want to run Machinekit as in "CNC stack". But I indeed do need to launch machinetalk (remote components).

I in deed appreciate the waiting for testing
And it still needs some :)

luminize, 2017-08-29:
@machinekoder since you probably know most of this, can you shed some light on this?
Do we need to rename, does this launch other stuff like the CNC stack?

closed 2017-08-30 by luminize

luminize, 2017-08-30:
it's back again!

ArcEye, 2017-08-30:
Indeed, it was always in the RIP, just removed from the package pending complete removal.

Please keep testing 👍

PR #29 (2017-08-30): Return mklauncher to the package build
Fixes #28
commit cd97aeee (2017-08-30): Return mklauncher to the package build

Fixes #28