machinekit/machinekit-hal #47: Missing packaged during ./configure of machinekit-hal on vanilla stretch install

luminize, 2017-11-17:
Yesterday I did a vanilla Stretch install on a VM, and decided to start with installing machinekit-hal.
During the `./configure --with-posix` step the script complained about missing packages, for example the glib2.0 package, the gtk package and whatnot.

So I cloned machinekit, ran `debian/configure -p`, `sudo mk-build-deps -ir` and noticed that a lot of packages got installed (which didn't get installed when doing the same steps from the machinekit-hal source). After that step I switched back to machinekit-hal, and build as expected (no ./configure errors anymore)

Do we need to clean up the ./configure dependancies?

ArcEye, 2017-11-17:
I haven't installed on a bare system to use as a RIP.  Certainly not by your method.
I use a script to install packages before any development type builds.

Whilst there is no GUI such as Axis, the machinekit-hal  sources still include latency-test, halscope etc. so require some/all of the graphic libs.

The difference is probably that the machinekit-hal control files have been pared down too far, so
using your method does not install the required libs, because they are not listed as deps.

https://github.com/machinekit/Machinekit-HAL/blob/master/debian/control.in 
lists the build deps for making the package
https://github.com/machinekit/Machinekit-HAL/blob/master/debian/control.rt-preempt.in
lists the package deps

Comparison of those to the machinekit ones, will probably highlight some gaps.

luminize, 2018-08-03:
I'll take this on

ArcEye, 2018-08-03:
Do try it again before delving in.
I have changed quite a lot since then, it may already be fixed ( or a different problem entirely :laughing: )

luminize, 2019-03-08:
Revisiting this issue:
Installing machinekit-hal RIP on a vanilla debian buster _without_ first installing the machinekit RIP.
following the installation steps, having `sudo mk-build-deps -ir` install a bunch of dependencies.

`./configure --with-rt-preempt` will break several times, so I'm looking which ones are missing so the previous step `sudo mk-build-deps -ir` can be corrected.

so far, manually installed: libsystemd-dev

Do I need a different package for:
```
checking for GLIB... no
configure: error: Package requirements (glib-2.0) were not met:

No package 'glib-2.0' found
```

luminize, 2019-03-08:
these packages are _extra_ installed (not by machinekit-hal `sudo mk-build-deps -ir`)
```
The following NEW packages will be installed:
  gir1.2-gtk-2.0 gir1.2-harfbuzz-0.0 icu-devtools libatk1.0-dev libblkid-dev libcairo-script-interpreter2 libcairo2-dev libffi-dev libfontconfig1-dev
  libfreetype6-dev libfribidi-dev libgdk-pixbuf2.0-dev libglib2.0-dev libglib2.0-dev-bin libgraphite2-dev libgtk2.0-dev libharfbuzz-dev
  libharfbuzz-gobject0 libicu-dev libmount-dev libpango1.0-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpng-dev libselinux1-dev
  libsepol1-dev libtk-img libxcb-shm0-dev libxcomposite-dev libxcursor-dev libxft-dev libxi-dev libxinerama-dev libxml2-utils libxrandr-dev
  libxrender-dev libxss-dev pango1.0-tools tcl8.6-dev tk8.6-dev x11proto-composite-dev x11proto-input-dev x11proto-randr-dev x11proto-scrnsaver-dev
  x11proto-xinerama-dev
```
are there packages that we can omit for machinekit-hal?

ArcEye, 2019-03-08:
I am currently revisiting the packaging for `machinekit-hal` and `machinekit-cnc` to ensure that all the `machinekit` deps are pulled by one or the other.

Hope to have a PR later.

Just noticed that cherry-picking has removed the control.stretch.in extras where needed again.
Need to sort those out too until we drop Jessie.

Never seen some of those packages.
For instance libfribidi.deb - WTF!
`FriBiDi is a BiDi algorithm implementation for Hebrew and/or Arabic languages.`

luminize, 2019-03-08:
> On 8 Mar 2019, at 13:10, ArcEye <notifications@github.com> wrote:
> 
> Never seen some of those packages.
> For instance libfribidi.deb - WTF!
> FriBiDi is a BiDi algorithm implementation for Hebrew and/or Arabic languages.

I saw the first time (for machinekit-hal) that there were packages installed for libreoffice too. Unbelievable!

cerna, 2019-03-10:
I have been doing some testing with Buster install and RIP building machinekit-hal and machinekit-cnc. (Not sure if this is not better suited for #179 as I did it with RT-PREEMPT.)

Installing dependencies with `sudo mk-build-deps -ir` on `uname -a: Linux MaDB 4.19.0-2-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.16-1 (2019-01-17) x86_64 GNU/Linux` (package `machinekit-hal`) with `debian/configure -pr` broke on need to additionally install packages `libsystemd-dev` and `libgtk2.0-dev`. After then it installed and was able to run realtime.

Installing RIP build of `machinekit-cnc` was bigger fun. there is the help in `README`:
```
However this is now possible, should the need arise.
First, you must build machinekit-hal, so that the required libs and headers are available.
Then build the machinekit-cnc RIP with
$ ./autogen.sh && ./configure --with-machinekit-hal='path/to/your/machinekit-hal/build' && make -j$(nproc)
```

However when I tried the ` ./configure --with-machinekit-hal='~/Downloads/machinekit-hal'` it couldn't find it. So maybe the help needs to me more moron friendly. (I was able to solve this with introducing additional variable:
```
export MHAL='~/Downloads/machinekit-hal'
./configure --with-machinekit-hal=$MHAL
```
Not sure how right it is, but it works.)
Then to successfully build I also needed to manually install packages `tcl-dev` and `tk-dev`.

Then I discovered, that there is no RIP-environment script for `machinekit-cnc` so one needs to run `./linuxcnc` from scripts (and the realtime needs to be running beforehand). And when running AXIS there is also a need to install additional package `bwidget`.

(On the other hand, not sure how big percentage of users even use RIP.)

ArcEye, 2019-03-10:
There always was a problem in linuxcnc with using `~/`
At one point you could not do so in ini files etc, using the full path should work.

Whilst it is possible to run both hal and cnc from a RIP, that was never the intention, you could just run machinekit if you wanted that.

I have added quite a lot of deps recently to the packaging, so it should be getting nearer correct, but plenty still to do.

cerna, 2019-03-10:
> There always was a problem in linuxcnc with using ~/
At one point you could not do so in ini files etc, using the full path should work.

Huh, absolute path works. Interesting. But the relative in conjunction with variable works also. Strange. Probably some bash magic between heaven and hell.

> Whilst it is possible to run both hal and cnc from a RIP, that was never the intention, you could just run machinekit if you wanted that.

But with machinekit repository on it's way to obsolescence that's not going to be true much longer. And it's true that installing from packages works - I think - without problems, so there is actually no need to build and use RIP. However, if I was inclined to do some development in machinekit-cnc I would not know to evade getting myself into symlink and configuration hell.

I guess that it will change again in the near future though.

ArcEye, 2019-03-11:
The longer term answer may be to have a local install into a disposable dir and have both -hal and -cnc install into it.
A bit like the CMake `build` dir.

At present the priority is to ensure everything in the packages works, because 99.9% of users will be on packages.

cerna, 2021-12-27:
No longer applicable.

closed 2021-12-27 by cerna