machinekit/machinekit-hal #58: mentions of kinematics in Makefile and user comp

luminize, 2018-08-01:
reviewing PR https://github.com/machinekit/machinekit/pull/1385 I needed to check if this PR needed to be taken over into machinekit-hal issue too. I noticed that (I did not expect) that this search https://github.com/machinekit/machinekit-hal/search?q=lineardeltakins&unscoped_q=lineardeltakins yields mentions to lineardeltakins. Should they be taken over to machinekit-cnc?

[image]

ArcEye, 2018-08-01:
This is one of those chicken and egg things.

It imports _hal_, that lives in `machinekit-hal` too.
It also imports _rs274_, _minigl_ and _glnav_ via `vismach` too, and they live in `machinekit-cnc`.

There are valid arguments for both.

The main thing is that `machinekit-cnc` cannot run without `machinekit-hal` and once they are combined, it does not matter which repo a python lib was in.

Since python scripts are runtime (as opposed to python libs, which must have required elements at build time) they can be in either repo.
If _lineardeltakins_ and _tripodkins_ were in `machinekit-cnc`, it would require all the build artifacts to be there too, or copies of them from the `machinekit-hal` repo.

I tried to keep things simple by just copying over the 'proper' C/C++ libs for the machinekit-cnc build
and leaving the python scripts to resolve at runtime.

Some elements are simply going to span the repos, since it is impossible to duplicate files, as dpkg will not allow it.

luminize, 2018-11-12:
Let's leave it in machinekit-hal only. Makes a lot more sense

closed 2018-11-12 by luminize