machinekit/machinekit-hal #111: version=git not installed at configure time

ArcEye, 2018-08-03:
[image] **Issue by [cdsteinkuehler](https://github.com/cdsteinkuehler)**
_Sun Jul  6 01:14:52 2014_
_Originally opened as https://github.com/machinekit/machinekit/issues/236_

----

I am using the BeagleBone packages on a fresh install of the current latest BeagleBone Debian image (bone-debian-7.4-2014-04-23-2gb.img.xz), and I am having issues getting things running.  While reviewing the logs, I notice the following every time I attempt to launch LinuxCNC:

Jul  6 01:09:03 beaglebone msgd:0: MSGD:0 mlock(global) failed: 12 'Cannot allocate memory'
Jul  6 01:09:03 beaglebone msgd:0: startup instance=inst0 pid=6003 flavor=xenomai rtlevel=1 usrlevel=1 halsize=512000 shm=Posix gcc=4.6.3 version=git not installed at configure time
Jul  6 01:09:03 beaglebone msgd:0: configured: Jul  1 2014 12:27:54 sha=git not installed or executable
Jul  6 01:09:03 beaglebone msgd:0: built:      Jul  1 2014 12:29:13 sha=git not installed or executable

I am working on the "cannot allocate memory" issue, but it looks like the version= and sha= are a goof in the way packages are getting built.

ArcEye, 2018-08-03:
See https://github.com/machinekit/machinekit-cnc/issues/37 to comment edit etc.

cerna, 2020-07-03:
This is - I think - passé now.

The current build flow requires the `git` executable as build dependency, so on (Buster) you cannot run the `dpkg-buildpackage` without it. Building from tarball is also not supported in build flow and as such anybody who wants to do it has to implement his own flow.

I am going to close this issue as *solved by passage of time*.

closed 2020-07-03 by cerna