machinekit/machinekit-hal #113: UUID in machinekit.ini is always the same

ArcEye, 2018-08-03:
[image] **Issue by [machinekoder](https://github.com/machinekoder)**
_Fri Mar 13 13:00:36 2015_
_Originally opened as https://github.com/machinekit/machinekit/issues/531_

----

The Machinekit Debian packages feature a default machinekit.ini file. There is an UUID values in this configuration file that is relevant to identify different Machinekit instances on the network. This UUID should be autogenerated after package install to ensure a different identification for every machine.

ArcEye, 2018-08-03:
See https://github.com/machinekit/machinekit-cnc/issues/39 to comment, edit etc.

cerna, 2020-07-02:
This should be resolved in machinekit/machinekit#1447 and as such in Machinekit-HAL codebase.

closed 2020-07-02 by cerna