machinekit/machinekit-hal #114: Python 3 support

ArcEye, 2018-08-03:
[image] **Issue by [RunningLight](https://github.com/RunningLight)**
_Thu Apr 16 22:16:44 2015_
_Originally opened as https://github.com/machinekit/machinekit/issues/563_

----

This isn't a "sky is falling" issue for now, but we have about 300 python files in our source repo and many are inconsistent with Python 3. Some conflicts can be resolved with minor syntactic changes, e.g., to modify the argument lists of old-style print statements; some require actual thought.

Why do I mention this, especially since I still write Python the way I learned it before the turn of the century? Because I was just reading a posting by a Debian team member (re)confirming their commitment to eliminate Python 2 from Debian by 2020.  Five years isn't such a long time that we shouldn't be thinking about this problem now. 

At a minimum, I believe we should commit to support both Python 2 and 3 in any new modules we write and to begin to port the modules we already have. We might even build pylint or something like it into our code testing.

To quote from https://docs.python.org/3/howto/pyporting.html

> A key point about supporting Python 2 & 3 simultaneously is that you can start today! Even if your dependencies are not supporting Python 3 yet that does not mean you can’t modernize your code now to support Python 3. Most changes required to support Python 3 lead to cleaner code using newer practices even in Python 2.
> 
> Another key point is that modernizing your Python 2 code to also support Python 3 is largely automated for you. While you might have to make some API decisions thanks to Python 3 clarifying text data versus binary data, the lower-level work is now mostly done for you and thus can at least benefit from the automated changes immediately.

Several automation tools which aid in porting and testing are mentioned in the above-referenced page.

ArcEye, 2018-08-03:
Changes were made to make the invocation of python, specify `python2`, therefore allowing distros which ship with default of python3 to build without issue.

Python 3 being non backwardly compatible with python 2 was a spectacular own goal which python.org have yet to recover from.

A lot of elements in python 3 have been back ported to python 2.7, but until all major libs are re-written in python specific form, this situation will continue.

ArcEye, 2018-10-08:
Not sure that the code we have in python will be the main problem.

This is the list of dependent python libraries as best as I can determine.
How many of these libraries are available in python 3 or can be used with it?
```
libboost-python-dev
cython (>= 0.19) (appears so)
dh-python
python-tk
python-zmq (>= 14.0.1)
python-setuptools,
python-simplejson
python-imaging or python-pil
python-imaging-tk or python-pil.imagetk
python-gnome2
python-glade2
python-numpy 
python-gtksourceview2
python-vte
python-xlib
python-gtkglext1
python-configobj,
python-protobuf (>= 2.4.1)
python-avahi
python-pyftpdlib
python-pydot
python-gst0.10 or python-gst-1.0 
python-netifaces
```

loganpowell, 2019-01-10:
Just cross-referencing this with the folks in LinuxCNC, who are also having this conversation:

https://github.com/LinuxCNC/linuxcnc/issues/403#issuecomment-452896732

Also, if you're having problems with Cython, apparently there's a new kid on the block, which doesn't require the special syntax of Cython code: https://github.com/Nuitka/Nuitka

ArcEye, 2019-01-17:
Thanks for the link.

We suffer from the linuxcnc python / gtk legacy and eventually it will all start to unravel. ;)

machinekoder, 2019-03-22:
If we could skip the UI libs, it would be a lot less work. I have looked into making the Axis and co code PEP8 compatible once and noticed that there is quite a lot of old-style Python code in there. With old-style, I mean syntax that isn't available in Python 3 anymore and already deprecated in newer Python 2 versions.

An auto-converter might be a good start. I encourage to do a complete QA cleanup of the Python code at the same time. Implementing automatic flake8 tests and auto-formatting with black. I've already applied those tools to everything Machinetalk related and the `lib/python/machinekit` libs.

Somewhat harder might be the CPython stuff and also the embedded Python support in HAL and interp. We will have to take a look at that.

I don't see why Cython should make problems. It's frequently used project in the Python3 world as well.

ebo, 2020-05-04:
has there been any forward motion on this in the last year and a half?  I mean not only is python2 EOL'ed, many OS's are no longer supporting it.  This is already an issue, and will eventually be a real problem.

Also, with the advent of numba, it might make replacing the cython code a lot easier.

ebo, 2020-05-04:
Also, let me know if anyone has already started a branch/fork for this and I will look at adding to it...

zultron, 2020-05-04:
I don't think there's been any real, recent effort here towards python3 compatibility.  I did start to look at this a week or so back and wrote a few commits [1] before dropping it to return to the LCNC-EMC on MK-HAL port; see #267.  I'm expecting to have a working branch in the next week or so.

Given that, python3 compatibility will be done for HAL in this project, and separately for EMC over in the LCNC project.

On the LCNC side, they have an issue open [2] that cites several good attempts.  @rene-dev already has a branch working (congratulations, wow!).

On the MK-HAL side, we can certainly port a lot of that work over.  My naive and woefully incomplete attempt is probably mostly already done in the LCNC effort, but it still may have changes only applicable to MK, like the changes to `comp`/`instcomp`, protobuf and packaging.

[1]:  https://github.com/zultron/machinekit-hal/commits/2020-04-24-python3
[2]: https://github.com/LinuxCNC/linuxcnc/issues/403

zultron, 2020-05-04:
Also, for Debian folks, see comments over at #268 related to python3 and the port to Bullseye where py3 has been dropped (I remember @ebo runs Gentoo or another non-Debian-derivative distro).

- https://github.com/machinekit/machinekit-hal/issues/268#issuecomment-618985239
- https://github.com/machinekit/machinekit-hal/issues/268#issuecomment-619079983
- https://github.com/machinekit/machinekit-hal/issues/268#issuecomment-619154041

rene-dev, 2020-05-04:
linuxcnc master now supports python3, all tests pass, and axis works.

zultron, 2020-05-04:
> linuxcnc master now supports python3, all tests pass, and axis works.

@rene-dev You're awesome!  And the other folks that contributed to that effort!

Are you targeting the 2.8 release, or saving it for 2.9?

rene-dev, 2020-05-04:
That’s the target for 2.9(2.8 is close to release), because no other ui components work due to gtk2 problems with python3. Im currently porting everything to gtk3, and the qt people need to get qt working with python3, which should be less of an issue.

ebo, 2020-05-05:
Since my intent is to build both a Gentoo overlay, and actual up to date plug and pray images for RPi 3//4 (using gentoo and buster based), I've redone the old linuxcnc and machinekit ebuilds.  I've also started new ones for linuxcnc-2.8.* and gave up on machinekit-hal at the moment.  I can pick that back up later once it is actually working.  It has been a couple of days since I got the ebuilds marginally working, so I cannot give you a current status because the base OS.  Unfortunately I do not have equipment to test while in isolation.  @zultron, please ping me when you do release the python3 related changes and I will try to break away enough time to give it a serious poke.  Until then I'm in the middle of rebuilding the traction battery in my Prius and other non-trivial household projects + plus passing my master gardeners cert...

zultron, 2020-05-11:
The changes to support building LCNC-EMC against MK-HAL are now in master, so if you'd like to try this, be my guest.

I wrote some quick notes pointing to the modified LCNC branch and how to build it [here][1].

To take advantage of what @rene-dev has done, you'll have to merge the LCNC `master` branch into that branch.  You'll have to muddle through a few conflicts, unfortunately.

Then, as I said above, take a look at [my previous attempt] and LinuxCNC/linuxcnc#403 to start the HAL-side work.  I bet you could get 80% done in a day or two.

[1]: https://github.com/machinekit/machinekit-hal/issues/267#issuecomment-626515589
[2]: https://github.com/zultron/machinekit-hal/commits/2020-04-24-python3

ebo, 2020-05-11:
I will look into this ASAP -- but as I am working 40+hrs/week, it will 
only be in the evenings...

As a note, before reading this I was thinking of asking if there had 
been any progress.  I must be psychic ;-)

Anyway, I will let you know how it goes.  My intent is to get it first 
building on my Gentoo laptop (running a preempt kernel), and then on a 
RPi (both under Gentoo and Buster).

   More later,

   EBo --

On May 11 2020 1:18 AM, John Morris wrote:
> The changes to support building LCNC-EMC against MK-HAL are now in
> master, so if you'd like to try this, be my guest.
>
> I wrote some quick notes pointing to the modified LCNC branch and how
> to build [...]

ebo, 2020-05-13:
I was asked to move a note from https://github.com/machinekit/machinekit-hal/issues/267#issuecomment-627676909 regarding build issues I am having with building machinekit-hal.  I have been building off of 2020-04-24-python3.  Please inform me if there is a better one to use.

Since my ebuild is "live" meaning that it checks the repository and updates if any changes have been made, I am getting a different set of error messages at the moment.  In the future I will plan on providing full build logs.  I also need to go to work, so I will have to work on this later.

ebo, 2020-05-14:
ok... I just made a clean fork, and started working on the 2020-04-24-python3 branch.

the commit https://github.com/ebo/machinekit-hal/commit/5e483b0669ac8c1d83517ece66f54ff8e9672130 fixes a couple of issues with autoconfig.  You may or may not consider this enough for a proper PR, but it is a start. 

Also, using my portage ebuild script it currently generates a build configuration of "./configure --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --sysconfdir=/etc --localstatedir=/var/lib --docdir=/usr/share/doc/machinekit-hal-9999 --htmldir=/usr/share/doc/machinekit-hal-9999/html --libdir=/usr/lib64 --enable-drivers --with-x --with-python=/usr/bin/python3.7 --with-boost-python=boost_python-3.7 --with-libmodbus"

Do you see any issues with the config so far?  I can easily prune it, but I would like to know if anyone sees an issue.

zultron, 2020-05-14:
@ebo  I'm sorry, let me clarify my above https://github.com/machinekit/machinekit-hal/issues/114#issuecomment-626518484 .  Please start with latest `machinekit-hal` repo's `master` branch, since major changes have been added just in the last week.

After that, take at look at LinuxCNC/linuxcnc#403, where they discuss doing the same thing at the LCNC project, and just finished.  Some of the most challenging changes will be in recent PRs.  My [2020-04-24-python3 branch][1] may contain useful things to pick, but it's really just a few hours head start that still hasn't been rebased onto the latest `master` branch.

[1]: https://github.com/zultron/machinekit-hal/commits/2020-04-24-python3

ebo, 2020-05-14:
Using the above setup I get the following error:

Compiling ++ hal/lib/halmodule.cc
hal/lib/halmodule.cc: In function ‘void pyhal_delete(PyObject*)’:
hal/lib/halmodule.cc:156:11: error: ‘halobject’ {aka ‘struct halobject’} has no member named ‘ob_type’
  156 |     self->ob_type->tp_free(self);
      |           ^~~~~~~
hal/lib/halmodule.cc: In function ‘int pyhal_write_common(halitem*, PyObject*)’:
hal/lib/halmodule.cc:160:18: error: ‘PyInt_Check’ was not declared in this scope; did you mean ‘PySet_Check’?
  160 |     int is_int = PyInt_Check(value);
      |                  ^~~~~~~~~~~
      |                  PySet_Check
hal/lib/halmodule.cc:161:28: error: ‘PyInt_AsLong’ was not declared in this scope; did you mean ‘PyLong_AsLong’?
  161 |     long intval = is_int ? PyInt_AsLong(value) : -1;
      |                            ^~~~~~~~~~~~
      |                            PyLong_AsLong
hal/lib/halmodule.cc: In function ‘PyObject* pyhal_read_common(halitem*)’:
hal/lib/halmodule.cc:280:34: error: ‘PyInt_FromLong’ was not declared in this scope; did you mean ‘PyLong_FromLong’?
  280 |             case HAL_S32: return PyInt_FromLong(*(item->u->pin.s32));
      |                                  ^~~~~~~~~~~~~~
      |                                  PyLong_FromLong
hal/lib/halmodule.cc:291:34: error: ‘PyInt_FromLong’ was not declared in this scope; did you mean ‘PyLong_FromLong’?
  291 |             case HAL_S32: return PyInt_FromLong(item->u->param.s32);
      |                                  ^~~~~~~~~~~~~~
      |                                  PyLong_FromLong
hal/lib/halmodule.cc: In function ‘PyObject* pyhal_repr(PyObject*)’:
hal/lib/halmodule.cc:442:12: error: ‘PyString_FromFormat’ was not declared in this scope; did you mean ‘PyBytes_FromFormat’?
  442 |     return PyString_FromFormat("<hal component %s(%d) with %d pins and params>",
      |            ^~~~~~~~~~~~~~~~~~~
      |            PyBytes_FromFormat
hal/lib/halmodule.cc: In function ‘PyObject* pyhal_getattro(PyObject*, PyObject*)’:
hal/lib/halmodule.cc:455:46: error: ‘PyString_AsString’ was not declared in this scope
  455 |     return pyhal_read_common(find_item(self, PyString_AsString(attro)));
      |                                              ^~~~~~~~~~~~~~~~~
...

I need to call it a night soon.  Any insights welcomed.

ebo, 2020-05-14:
Ahhh... sorry about that.  I will pull run live from master, and take a look at LCNC#403.

That said, the fixes to the configure.ac might be useful.  I seem to remember a similar issue before, so maybe this will help.

zultron, 2020-05-14:
> ok... I just made a clean fork, and started working on the 2020-04-24-python3 branch.

Messages crossing in flight!  See above:  I recommend starting with `master` branch to avoid later merge headaches.

> the commit [ebo@5e483b0](https://github.com/ebo/machinekit-hal/commit/5e483b0669ac8c1d83517ece66f54ff8e9672130) fixes a couple of issues with autoconfig. You may or may not consider this enough for a proper PR, but it is a start.

Those changes look fine to me.

> Also, using my portage ebuild script it currently generates a build configuration of "./configure --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man [...]

That looks reasonable.  While developing, why not start with an "RIP" build?

zultron, 2020-05-14:
> Using the above setup I get the following error:
>
> Compiling ++ hal/lib/halmodule.cc
> hal/lib/halmodule.cc: In function ‘void pyhal_delete(PyObject*)’:
> hal/lib/halmodule.cc:156:11: error: ‘halobject’ [...]

This is where I recommend you start looking at the changes that went into LCNC.  This is the most challenging part of the port for me, at least, but fortunately @strufkin and maybe others have already done this for us.

ebo, 2020-05-14:
On May 13 2020 8:26 PM, John Morris wrote:
> Messages crossing in flight!  See above:  I recommend starting with
> `master` branch to avoid later merge headaches.
>
>
> Those changes look fine to me.

Take a look at my master branch.  I had to make changes to the same two 
lines...

> That looks reasonable.  While developing, why not start with an "RIP"
> build?

I though all of this was more stable than it is.  Normally I can start 
writing a gentoo ebuild out of the box and get it to do the right thing 
immediately and build the execs using standard Gentoo'isms.  Neither 
machinekit and linuxcnc are really ready for prime time again...

rene-dev, 2020-08-11:
FYI, Im currently working on this, and will keep you updated.

ebo, 2020-08-11:
On Aug 11 2020 3:50 AM, Rene Hopf wrote:
> FYI, Im currently working on this, and will keep you updated.

thanks.  I spent a little time and ended up giving up as it will be a 
bigger effort than I had time for AND I was not sure once it got working 
if I would end up with a fully working machine (which is what I really 
need -- ie. rebuild/replace the controllers for at least 2 machines if 
not 6).

cerna, 2020-08-23:
All `Avahi*` packages for Debian Stretch and Debian Buster uploaded to the machinekit/machinekit repository.

Looking at the Ubuntu, the `py3c-dev` package is in all versions, `python3-yapps` is missing from Ubuntu Bionic 18.04 LTS and the `Avahi*` are missing both from Ubuntu Bionic 18.04 LTS and Ubuntu Focal 20.04 LTS.

So, I am going to take the version from Groovy and build them.

(The dependency's repository is attacking 350 packages. It would be a good idea to create a static backup and upload it to Mega or something else which is long-term stable.)

rene-dev, 2020-08-23:
> This is one reason we can't `update-alternatives --install /usr/bin/python python /usr/bin/python3 1`. Another reason, I tried that on Stretch, and immediately some Debian package pre/post scripts started breaking.

I also strongly advise against using update-alternatives with python, it can break the system, and its really the job of the build system to pick the right versions.

zultron, 2020-08-23:
> I also strongly advise against using update-alternatives with python, it can break the system, and its really the job of the build system to pick the right versions.

Point taken!  It broke Stretch quite badly.  As I said in #299, my [rebased branch][1] fixes all the shebangs to `python3`.

I haven't tried to bake in @cerna's `py3c-dev` packages yet.  The packaging does make use of the rest of the new Yapps and Avahi packages.

[1]: https://github.com/zultron/machinekit-hal/tree/pr-299-rebase/

ebo, 2020-08-23:
On Aug 23 2020 1:49 PM, John Morris wrote:
> Point taken!  It broke Stretch quite badly.  As I said in #299, my
> [rebased branch][1] fixes all the shebangs to `python3`.
>
> I haven't tried to bake in @cerna's `py3c-dev` packages yet.  The [...]

on a totally unrelated not I received a post on converting python code 
to an Abstract Syntax Tree and back again.  Stackoverflow has an 
interesting reply to this:

https://docs.python.org/2/library/2to3.html

and it got me to thinking about this is SUCH a pain, I wonder if anyone 
wrote a converter...  Well check this out -- python's 2to3:

     https://docs.python.org/2/library/2to3.html

I had not known about this...  Never used it, so do not know how much 
3'isms it will deal with, but it is likely a start with hundreds of 
python files...

zultron, 2020-08-23:
> python's 2to3

I did know about `2to3`, and yet for some reason I still went with emacs macros.  That might've turned the 30 minute job into one minute.  It was about a hundred files.  Sometimes I choose the known pain, even if it's likely going to be worse.

ebo, 2020-08-23:
On Aug 23 2020 2:34 PM, John Morris wrote:
> ...
> Sometimes I choose the known pain,
> even if it's likely going to be worse.

Fair enough.

cerna, 2020-08-24:
Missing packages for Ubuntu Bionic 18.04 LTS and Ubuntu Focal 20.04 LTS uploaded to the [Machinekit/Machinekit](https://cloudsmith.io/~machinekit/repos/machinekit/) dependency's repository.

rene-dev, 2020-08-31:
closed by https://github.com/machinekit/machinekit-hal/pull/303

closed 2020-08-31 by rene-dev