machinekit/machinekit-hal #119: information when machinekit fails is wrong/inadequate

ArcEye, 2018-08-03:
[image] **Issue by [luminize](https://github.com/luminize)**
_Wed Apr  4 06:53:44 2018_
_Originally opened as https://github.com/machinekit/machinekit/issues/1366_

----

When something fails the following is shown:

```
See /var/log/linuxcnc.log for more information.
Shutting down and cleaning up Machinekit...
Cleanup done
Machinekit terminated with an error.  You can find more information in the log:
   /home/machinekit/linuxcnc_debug.txt
and
   /home/machinekit/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
```

Personally I think this is wrong info and should contain better instructions, like something as "start with export DEBUG=5" and "look at /var/log/linuxcnc.log before posting". Maybe some basic instructions in helping people narrowing down.

I don't remember ever finding anything in linuxcnc_debux.txt or linuxcnc_print.txt

Opinions?

ArcEye, 2018-08-03:
See https://github.com/machinekit/machinekit-cnc/issues/44 for full details, this is a header only.

luminize, 2018-11-12:
The machinekit-hal package does not contain the "machinekit" script at scripts/linuxcnc.in so this issue has no meaning here.
Following these PR's https://github.com/machinekit/machinekit-cnc/pull/54 and https://github.com/machinekit/machinekit/pull/1416 this issue can be closed

closed 2018-11-12 by luminize