machinekit/machinekit-hal #120: Build is littered with compiler warnings

ArcEye, 2018-08-03:
[image] **Issue by [zultron](https://github.com/zultron)**
_Fri Apr  6 23:20:12 2018_
_Originally opened as https://github.com/machinekit/machinekit/issues/1368_

----

The build is littered with compiler warnings.  When compiling my updates, this makes it really hard to tell if I've introduced a new warning.  #327 had them all fixed a couple of years back, but that was never merged, and it's unclear how many of those patches still address outstanding warnings, or how many would even apply.

ArcEye, 2018-08-03:
See https://github.com/machinekit/machinekit-cnc/issues/45 for full details, this is a copy header only

zultron, 2018-12-11:
Can this issue be closed now that the corresponding machinekit/machinekit-cnc#45 is fixed?

closed 2018-12-11 by ArcEye