machinekit/machinekit-hal #68: build: hostmot2 dependencies broken

ArcEye, 2018-08-03:
[image] **Issue by [mhaberler](https://github.com/mhaberler)**
_Thu Jan 15 19:21:17 2015_
_Originally opened as https://github.com/machinekit/machinekit/issues/441_

----

Michal reported this, I reproduced:

micges@debian:~/PROJECTS/mk/src$ touch hal/drivers/mesa-hostmot2/hostmot2.h
micges@debian:~/PROJECTS/mk/src$ make
Reading 276/277 dependency files
Done reading dependencies

Reading 0/0 realtime dependency files
Done reading realtime dependencies
make: Wejście do katalogu `/home/micges/PROJECTS/mk/src'
Making modules for flavor xenomai
Reading 284/285 dependency files
Done reading dependencies

Reading 0/236 realtime dependency files
Done reading realtime dependencies
make[1]: Wejście do katalogu `/home/micges/PROJECTS/mk/src'
make[1]: Opuszczenie katalogu`/home/micges/PROJECTS/mk/src'
make: Opuszczenie katalogu `/home/micges/PROJECTS/mk/src'
micges@debian:~/PROJECTS/mk/src$

ArcEye, 2018-08-03:
[image] **Comment by [cdsteinkuehler](https://github.com/cdsteinkuehler)**
_Thu Jan 15 21:22:29 2015_

----

On 1/15/2015 1:21 PM, Michael Haberler wrote:

> Michal reported this, I reproduced:
>
> micges@debian:~/PROJECTS/mk/src$ touch hal/drivers/mesa-hostmot2/hostmot2.h
> micges@debian:~/PROJECTS/mk/src$ make
> Reading 276/277 dependency files
> Done reading [...]

I had a similar problem when updating src/emc/motion/motion_debug.h, it
didn't force anything to recompile, just the .c files I changed.

I figured it was something I was doing wrong, but perhaps the depend
system has some major recent breakage?

## 

Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2018-08-03:
[image] **Comment by [zultron](https://github.com/zultron)**
_Thu Jan 15 22:04:21 2015_

----

I'm pretty sure this has been going on for a while, quite possibly since my UB build system updates.

I'll try to figure it out tonight.

ArcEye, 2018-08-03:
[image] **Comment by [micges](https://github.com/micges)**
_Thu Oct 15 22:29:59 2015_

----

Any solutions for this yet?

cerna, 2021-12-27:
No longer applicable.

closed 2021-12-27 by cerna