machinekit/machinekit-hal #96: possible aarch64 support

ArcEye, 2018-08-03:
[image] **Issue by [bjharper](https://github.com/bjharper)**
_Sun Sep  3 03:34:00 2017_
_Originally opened as https://github.com/machinekit/machinekit/issues/1269_

----

I have had some success compiling and running LinuxCNC on aarch64 (Ubuntu 16.04) using Realtime preempt 4.11 kernel but I am struggling a little with machine kit.  I seems to always end up with an error looking for sys/io.h
```
Compiling machinetalk/support/sizes.c
In file included from rtapi/rtapi.h:1016:0,
                 from hal/lib/hal.h:126,
                 from ./emc/motion/motion.h:80,
                 from machinetalk/support/sizes.c:2:
rtapi/rtapi_io.h:39:43: fatal error: sys/io.h: No such file or directory
compilation terminated.
Makefile:420: recipe for target 'objects/machinetalk/support/sizes.o' failed
```

Basic steps is to install all dependencies then
$ debian/configure -pr
$ sudo mk-build-deps -ir
$ cd src
$ ./autogen.sh
$ ./configure --with-rt-preempt  --with-posix
$ make -j2

Kernel im working with at the moment is development Realtime PREEEMT branch "'Linux odroid64 4.11.12-rt10 #6 SMP PREEMPT RT Sat Aug 26 17:27:03 AEST 2017 aarch64 aarch64 aarch64 GNU/Linux"

Im working on a HAL driver for the Odroid C2 and its GPIO pins so I dont know if I really need the io.h support in rtapi (no PCI interface ill be hitting GPIO registers directly hopefully) and if I need outb / inb functions then it looks like I could probably emulate them myself.

Looks like I will need to setup an aarch64 specific configuration in the machinekit source or maybe install for further dependencies for aarch64?  Im compiling machinekit on an aarch64 platform also, no cross-compiling.

ArcEye, 2018-08-03:
[image] **Comment by [sirop](https://github.com/sirop)**
_Sun Sep  3 04:31:24 2017_

----

@bjharper 
Can't you use either Debian Jessie or Stretch instead?
See https://forum.odroid.com/viewtopic.php?f=138&t=27449 .

ArcEye, 2018-08-03:
[image] **Comment by [bjharper](https://github.com/bjharper)**
_Sun Sep  3 04:50:35 2017_

----

Thanks I will try Debian and see how it goes.

ArcEye, 2018-08-03:
[image] **Comment by [sirop](https://github.com/sirop)**
_Sun Sep  3 05:16:02 2017_

----

Could you please publish your instructions for building Linux odroid64 4.11.12-rt10,
as then I'd try it on my site too.

ArcEye, 2018-08-03:
[image] **Comment by [ArcEye](https://github.com/ArcEye)**
_Sun Sep  3 06:00:21 2017_

----

> rtapi/rtapi_io.h:39:43: fatal error: sys/io.h: No such file or directory

You will probably find it is in a subdirectory within /usr/include that relates to the arch type.
Ubuntu especially kept moving things around, I just used to symlink the contents to the main dir and all worked fine.

ArcEye, 2018-08-03:
[image] **Comment by [bjharper](https://github.com/bjharper)**
_Sun Sep  3 09:59:03 2017_

----

Linux odroid64 4.11.12-rt10 has some limitations for the moment (the main one being no hdmi output),  I was testing LinuxCNC over remote X11 sessions but want to use MachineKit for the extra features such as web UI (particularly as I have no HDMI) and more customised devices such as what has been done with Beaglebone and Pi.  HDMI support for C2 should be soon enough in mainline Linux but 4.11 is enough to get me started for now.

The kernel build is straightforward on an aarch64 platform but the installation is a bit more complex.
- clone the git from https://git.kernel.org/pub/scm/linux/kernel/git/rt/linux-rt-devel.git/log/?h=linux-4.11.y-rt
- run a defconfig
- modify the kernel config for Realtime
CONFIG_PREEMPT_RT_FULL
CONFIG_HZ_1000
CONFIG_EMBEDDED
CONFIG_SCHED_SMT (not sure if this is the smartest choice but seems to work well in testing)

The kernel itself can be a little troublesome to install as it must be made into a compatible uImage for uboot and booted with a different configuration from uboot but I will provide some instruction on the build as soon as I can document it a little clearer.

I did a search through /usr/include but no io.h on Ubuntu.  I notice in my kernel source there is an io.h but most likely not the user library im after for aarch64?  I will try again on Debian to see if I can work it out or it has different libraries.

Thanks!

ArcEye, 2018-08-03:
[image] **Comment by [sirop](https://github.com/sirop)**
_Sun Sep  3 10:18:36 2017_

----

@bjharper 
Thanks so far. 

It would be interesting to compare the latency of Odroid C2 with that of Pi3 https://github.com/koppi/mk/blob/master/Machinekit-RT-Preempt-RPI.md#latency-evaluation .

ArcEye, 2018-08-03:
[image] **Comment by [bjharper](https://github.com/bjharper)**
_Sun Sep  3 11:09:34 2017_

----

`odroid@odroid64:\~$ uname -a
Linux odroid64 4.11.12-rt10 #6 SMP PREEMPT RT Sat Aug 26 17:27:03 AEST 2017 aarch64 aarch64 aarch64 GNU/Linux
odroid@odroid64:~$ sudo cyclictest -t1 -p 80 -n -i 10000 -l 10000
\# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 0.93 0.42 0.16 2/218 27058          

T: 0 (27052) P:80 I:10000 C:  10000 Min:      7 Act:   10 Avg:   10 Max:      23`

that was done with a `odroid@odroid64:~$ cat /dev/zero > /dev/null` on the odroid and a `$ sudo ping -i 0.01 10.1.1.12` from my host to the odroid.   Impressive I guess but I can bust the max out if I really try.  (ie 100 x glxgears running over ssh & X11 or heavy disk IO got it up to 200 ms)

ArcEye, 2018-08-03:
[image] **Comment by [sirop](https://github.com/sirop)**
_Sun Sep  3 11:18:31 2017_

----

Yes, very  impressive latency results.

Some time ago with an older kernel 3.XX there was a problem with the network driver on C2.
Have you noticed anything like that with the newer kernel?

Anyway: waiting for  kernel installation instructions.

ArcEye, 2018-08-03:
[image] **Comment by [bjharper](https://github.com/bjharper)**
_Sun Sep  3 11:58:36 2017_

----

> Some time ago with an older kernel 3.XX there was a problem with the network driver on C2.

I actually had issues with the USB drivers in 3.10.y kernel, disabling usb got the numbers down nicely but I was not pushing the network to tell if it could be a problem.  In 4.11 the latency is a worse with a 100Mbit connection, 1GBit gives better results.  I did some ftrace latency tests and could see ethernet getting in the way if I maxed it out and the same with emmc card for disk.

I will do install instructions for stretch as soon as I can.

ArcEye, 2018-08-03:
[image] **Comment by [ArcEye](https://github.com/ArcEye)**
_Sun Sep  3 12:58:58 2017_

----

>rtapi/rtapi_io.h:39:43: fatal error: sys/io.h: No such file or directory

>You will probably find it is in a subdirectory within /usr/include that relates to the arch type.
Ubuntu especially kept moving things around, I just used to symlink the contents to the main dir and all worked fine.

On an armhf system, (Debian Stretch) the path is `/usr/include/arm-linux-gnueabihf/sys/io.h`
Yours will be very similar

ArcEye, 2018-08-03:
[image] **Comment by [bjharper](https://github.com/bjharper)**
_Mon Sep  4 13:07:02 2017_

----

I've wrote a quick README on how I installed on meveric's Debian Stretch and then the 4.11-y-rt kernel.  Please let me know if there are any questions, errors or something does not work.

http://www.builtonline.net/c2rt-v4.11.12-rt11/README

I havent attempted compiling machinekit as yet on Stretch, but I have checked Debian for io.h.  No sign of it (does not exist in /usr/include/aarch64-linux-gnu/sys/io.h or /usr/include/aarch64-linux-gnu/asm/io.h or anywhere).  I will check the kernel config to see if it should be installed as part of the headers as maybe this is where I am going wrong.

ArcEye, 2018-08-03:
[image] **Comment by [ArcEye](https://github.com/ArcEye)**
_Mon Sep  4 13:37:15 2017_

----

> I havent attempted compiling machinekit as yet on Stretch, but I have checked Debian for io.h. No sign of it (does not exist in /usr/include/aarch64-linux-gnu/sys/io.h or /usr/include/aarch64-linux-gnu/asm/io.h or anywhere). 

In that case it is probably the same situation I had when I compiled machinekit on the Pine64 some time back.

There was no io.h for 64 bit ARM and I had to create my own.
The ARM architecture does not support some of the calls (ioperm & iopl), the asm code is different and assumptions about ports and base addresses don't work.

This is the file I wrote which works with Pine64
What may well not be right for your board is the base address PCI_IOBASE.
I had to compute this after determining the start memory address that insmod uses to load modules, as explained in the comments.

```
* 
A largely dummy io.h to statisfy the requirements of Machinekit  
on aarch64
The functions still require a valid base address, which appears to be
#define PCI_IOBASE              ((void __iomem *)(MODULES_VADDR - SZ_2M))
MODULES_VADDR being the address at which insmod begins loading kernel modules
*/

#ifndef	_SYS_IO_H
#define	_SYS_IO_H

#include <features.h>

// as per arch/.../memory.txt
#define PCI_IOBASE   0xfee00000UL
// ((void __iomem *)(MODULES_VADDR - SZ_2M))
//((void __iomem *)PCI_IO_START)

__BEGIN_DECLS

/* If TURN_ON is TRUE, request for permission to do direct i/o on the
   port numbers in the range [FROM,FROM+NUM-1].  Otherwise, turn I/O
   permission off for that range.  This call requires root privileges.

   Portability note: not all Linux platforms support this call.  Most
   platforms based on the PC I/O architecture probably will, however.
   E.g., Linux/Alpha for Alpha PCs supports this.  */
//extern int ioperm (unsigned long int __from, unsigned long int __num,
//                   int __turn_on) __THROW;

/* Set the I/O privilege level to LEVEL.  If LEVEL>3, permission to
   access any I/O port is granted.  This call requires root
   privileges. */

// Does not appear this call is supported
// dummy function to return 0 (success and keep pci_write happy

//int iopl(int __level) { return 0; }

#if defined __GNUC__ && __GNUC__ >= 2

// Does not appear this call is supported
// dummy function to return 0 (success and keep pci_write happy

static __inline int iopl(int __level) { return 0; }

static __inline unsigned char 
inb (unsigned short int __port)
{
    unsigned char val;
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
    
    asm volatile("ldrb %w0, [%1]" : "=r" (val) : "r" (addr));
    return val;
}

static __inline unsigned short int
inw (unsigned short int __port)
{
    unsigned short val;
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
  
    asm volatile("ldrh %w0, [%1]" : "=r" (val) : "r" (addr));
    return val;
}

static __inline unsigned int
inl (unsigned short int __port)
{
    unsigned int val;
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
  
    asm volatile("ldrh %w0, [%1]" : "=r" (val) : "r" (addr));
    
    return val;
}

///////////////////////////////////////////////////////////////////

static __inline void
outb (unsigned char __value, unsigned short int __port)
{
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
    asm volatile("strb %w0, [%1]" : : "r" (__value), "r" (addr));
}

static __inline void
outw (unsigned short int __value, unsigned short int __port)
{
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
    asm volatile("strh %w0, [%1]" : : "r" (__value), "r" (addr));
}

static __inline void
outl (unsigned int __value, unsigned short int __port)
{
    unsigned long addr = ((unsigned long) __port + PCI_IOBASE);
    asm volatile("strh %w0, [%1]" : : "r" (__value), "r" (addr));
}

#endif	/* GNU C */

__END_DECLS
#endif /* _SYS_IO_H */
```

ArcEye, 2018-08-03:
[image] **Comment by [sirop](https://github.com/sirop)**
_Mon Sep  4 14:40:08 2017_

----

What about https://kernel.googlesource.com/pub/scm/linux/kernel/git/rt/linux-rt-devel/+/v4.11.12-rt11/include/asm-generic/io.h ?

ArcEye, 2018-08-03:
[image] **Comment by [bjharper](https://github.com/bjharper)**
_Wed Sep  6 02:01:47 2017_

----

Thanks @ArcEye im looking at the code and looking into a dummy address to use for the Odroid C2 and PCI_IOBASE

@sirop I think https://kernel.googlesource.com/pub/scm/linux/kernel/git/rt/linux-rt-devel/+/v4.11.12-rt11/include/asm-generic/io.h is kernel libraries but we need userspace which is not being exported to /usr/include for aarch64 on Odroid C2

ArcEye, 2018-08-03:
[image] **Comment by [bjharper](https://github.com/bjharper)**
_Thu Sep  7 13:26:33 2017_

----

Its now compiling doing what you have suggested @ArcEye For PCI_IOBASE I used what was reported at startup for the PCI virtual address allocation  PCI I/O : 0xffff7dfffee00000 - 0xffff7dffffe00000   (    16 MB)

Im not sure this is right, need to look at it further (insmod was using a different address)

I had some troubles building various versions of zeromq and czmq but got a working config on stretch.

The GUI is working over remote X and emcweb is working when enabled in sim mode so I can now starting playing with some config and looking at the HAL driver.

Thanks!

ArcEye, 2018-08-03:
[image] **Comment by [ArcEye](https://github.com/ArcEye)**
_Thu Sep  7 13:41:39 2017_

----

Glad you got there.

Regards czmq and zeromq, I made some packages, see https://github.com/machinekit/machinekit/issues/1228#issuecomment-316364736

They work fine, I am building on Stretch amd64, i386 and armhf using them.

ArcEye, 2018-08-03:
[image] **Comment by [bjharper](https://github.com/bjharper)**
_Thu Sep  7 14:06:04 2017_

----

I did actually see them but im using aarch64 and didnt think you would have a port there?

ArcEye, 2018-08-03:
[image] **Comment by [ArcEye](https://github.com/ArcEye)**
_Thu Sep  7 14:12:18 2017_

----

Of course you are, my brain is trying to do too many things at once!

I may have later, but it is someway down the list.  I just built from sources as you did and then
forgot all about it.

If you want to produce a package, because you are likely to do this repeatedly or want to circulate it,
the easiest way is to use `checkinstall`.

Build the library then from its root run `checkinstall install=no make install`
Answer the q's and change whatever and it will output a usable .deb

That is what I did initially with the other Stretch packages until I got a build system sorted.

ArcEye, 2018-08-03:
[image] **Comment by [bjharper](https://github.com/bjharper)**
_Mon Sep 11 08:38:14 2017_

----

I uploaded the zseromq and czmq packages I created here http://www.builtonline.net/c2rt-v4.11.12-rt11/  Sorry im not the best with debian packaging so they are /usr/local but should work ok.  Ill try to provide some detail on the machinekit build process as there were some dependencies required for stretch minimal image.  The 2 packages zeromq and czmq where the only ones I needed to compile from source and then the sys/io.h

I have been working on a basic gpio hal driver for the C2 starting as a base with the Beaglebone hal_bb_gpio.c to control GPIO and also incorporating an option to build machinekit for the C2.  So far so good everything is compiling ok (I think) optimised for A-53 cores with neon,  floating point and compiler specific options enabled.

I now need to create a hal config which raises a question, am I better to have pin config in a devicetree for the board or in a .hal config file in userspace or both?  I need to either make the hal driver search the devicetree or check a hal config file for pin configuration.  Im using hal_gpio_demo.hal file (Raspberry) as a test hal config and starting point for debug on the Odroid C2.

ArcEye, 2018-08-03:
[image] **Comment by [machinekoder](https://github.com/machinekoder)**
_Sun Oct  1 09:44:04 2017_

----

@bjharper I described how to create a GPIO HAL driver [here](https://machinekoder.com/machinekit-on-the-c-h-i-p-9-computer/). For the pin config, I suggest using the device tree overlay to stay consistent with other GPIO drivers. The solution used for the BBB and CHIP passing the GPIO pins via an HAL argument works very well and does not need any checking of the GPIO configuration from the HAL driver.

ArcEye, 2018-08-03:
[image] **Comment by [machinekoder](https://github.com/machinekoder)**
_Tue Dec  5 12:57:40 2017_

----

@bjharper What is the status of your project? I have a spare ODroid C2 I', thinking about putting to use with Machinekit.

the-snowwhite, 2019-08-30:
@zultron 
What is the correct way to build the mk-cross-builder/dockerfile ?

I'm trying to look into modifying the docker cross build image to add arm64 mk packages:
I'm using this as a base:
[dovetailautomata/mk-cross-builder/dockerfile]https://hub.docker.com/r/dovetailautomata/mk-cross-builder)
I'm able to build a docker file.
However when running the scripts/build_docker I get stuck into:

    missing package  protobuf 

Reverting back to just building the plain unmodified script as is (settings I have attempted to figure out):

    docker build -t mk-cross-builder2:armhf_9 --build-arg BASE_IMAGE=debian:stretch --build-arg DISTRO_VER=9 --build-arg DEBIAN_ARCH=armhf --build-arg DISTRO_CODENAME=stretch --build-arg SYS_ROOT=armhf .

this command

    scripts/build_docker -t armhf_9 -c deb

leads to:

    docker build -t mk-cross-builder2:armhf_9 --build-arg BASE_IMAGE=debian:stretch --build-arg 
    DISTRO_VER=9 --build-arg DEBIAN_ARCH=armhf --build-arg DISTRO_CODENAME=stretch -- 
    build-arg SYS_ROOT=armhf .

zultron, 2019-08-30:
@the-snowwhite `aarch64` support, great!  I'm going to be porting to the new BBAI soon; sounds like you're knocking out a major piece.

I don't know where the `missing package  protobuf` error message would come from without more context.

Building `aarch64` support takes changes in both the Docker image and Machinekit.  I'm assuming you have already discovered the [GH repo][1] associated with the Docker Hub images you linked to above.  Since I don't know how far you've gotten modifying those sources to add `aarch64` support, here's how I'd start.

Following the `armhf` support examples, I'd grep around for `armhf`, `arm-linux-gnueabihf`, `ARM_HOST_MULTIARCH`, `armelf_linux_eabi`, and add their `aarch64`, `aarch64-linux-gnu`, `ARM64_HOST_MULTIARCH`, `aarch64elf` counterparts.

Then build the container.  Your above `docker build` command line looks reasonable, but you'll need to substitute `s/armhf/aarch64/`.

A quick look indicates the changes aren't too many.  It's always a big challenge to add a new distro, but I'm crossing my fingers that adding a new architecture won't be such a big deal.  If you get stuck, commit your changes and point me to them, and I'll take a look.

[1]: https://github.com/Dovetail-Automata/mk-cross-builder

the-snowwhite, 2019-08-30:
@zultron 
Thanks
I found it best to take a step bach and see if I can build an umodified stretch armhf dockerimage, and then get that to build some (mk-hal)debs, so far without sucess so:
1. step at a time:

First I'm using kde neon bionic 18.04 on my workstation.

    Docker version 19.03.1, build 74b1e89

I run following commands:

    cd ../mk-cross-builder
    docker build -t mk-cross-builder:armhf_9 --build-arg BASE_IMAGE=debian:stretch --build-arg DISTRO_VER=9 --build-arg DEBIAN_ARCH=armhf --build-arg DISTRO_CODENAME=stretch --build-arg SYS_ROOT=armhf .

which gives [this build log](https://gist.github.com/the-snowwhite/6e9e083d7d06067af6238ea0d6bf37fa)

what I notice is that all the constructs like this:

    RUN test $DISTRO_VER -gt 8 \
        ||{ \
            VER=4.9.4-2017.01 && \
            ARCH=arm-linux-gnueabihf && \
            DIR=gcc-linaro-${VER}-x86_64_${ARCH} && \
            TXZ=${DIR}.tar.xz && \
            URI=http://releases.linaro.org/components/toolchain/binaries/latest-4/${ARCH} && \
            WORKD=/tmp/linaro && \
            mkdir -p ${WORKD} && cd ${WORKD} && \
            echo "Downloading gcc-linaro-${VER}" && \
            wget --progress=dot:mega ${URI}/${TXZ} && \
            wget -q ${URI}/${TXZ}.asc && \
            echo "Validating checksum of compiler download" && \
            md5sum -c ${TXZ}.asc && \
            echo "Extracting compiler to /opt/" && \
            tar xCf /opt ${WORKD}/${TXZ} && \
            ln -snf ${DIR} /opt/gcc-linaro-hf && \
            rm -rf ${WORKD} && \
            ln -s ../../bin/ccache /usr/lib/ccache/arm-linux-gnueabihf-gcc && \
            ln -s ../../bin/ccache /usr/lib/ccache/arm-linux-gnueabihf-g++ ; \
        }

Don't work as expected unless they are changed to use if .. then like this:

    RUN if test $DISTRO_VER -gt 8; \
        then \
            VER=4.9.4-2017.01 && \
            ARCH=arm-linux-gnueabihf && \
            DIR=gcc-linaro-${VER}-x86_64_${ARCH} && \
            TXZ=${DIR}.tar.xz && \
            URI=http://releases.linaro.org/components/toolchain/binaries/latest-4/${ARCH} && \
            WORKD=/tmp/linaro && \
            mkdir -p ${WORKD} && cd ${WORKD} && \
            echo "Downloading gcc-linaro-${VER}" && \
            wget --progress=dot:mega ${URI}/${TXZ} && \
            wget -q ${URI}/${TXZ}.asc && \
            echo "Validating checksum of compiler download" && \
            md5sum -c ${TXZ}.asc && \
            echo "Extracting compiler to /opt/" && \
            tar xCf /opt ${WORKD}/${TXZ} && \
            ln -snf ${DIR} /opt/gcc-linaro-hf && \
            rm -rf ${WORKD} && \
            ln -s ../../bin/ccache /usr/lib/ccache/arm-linux-gnueabihf-gcc && \
            ln -s ../../bin/ccache /usr/lib/ccache/arm-linux-gnueabihf-g++ ; \
        fi

Further on running:

    cd ../machinekit-hal
    export NO_PULL=1
    scripts/build_docker -i mk-cross-builder -t armhf_9 -c deb

ends up with a 

    checking for CZMQ... no
    configure: error: Package requirements (libczmq > 4.0) were not met:

    No package 'libczmq' found

[build failure log here](https://gist.github.com/the-snowwhite/fc508a37d5f5b71a82cf91f51174069f
)

So first I have to find out If I need to change distro for the docker build or if its correct to place in if .. then substitutes ?

cerna, 2019-08-30:
```
checking for CZMQ... no
configure: error: Package requirements (libczmq > 4.0) were not met:

No package 'libczmq' found
 ```

Isn't this caused by the fact that ZeroMQ libraries are downloaded from Machinekit Debian repository and there is not any aarch64 package? Try [official repository](https://build.opensuse.org/project/show/network:messaging:zeromq:git-draft) from [here](https://zeromq.org/download/).

zultron, 2019-08-30:
These constructs ARE working as intended:
```
RUN test $DISTRO_VER -gt 8 \
    ||{ \
    [...] \
    }
```
This confusing notation means "only run this block if `$DISTRO_VER` equals 8".

It would be good to convert these to more the more readable form you suggest, but the logic needs to be done right:
```
RUN if test $DISTRO_VER -eq 8; then \
        [...] \
    fi
```

I'll try out this build myself.  Seems like your build command should work, so I wonder if bit-rot has set in.

the-snowwhite, 2019-08-30:
@cerna 
This is from a armhf docker build and the libczmq packages  [are installed from the official Stretch repo by the docker build script](https://gist.github.com/the-snowwhite/6e9e083d7d06067af6238ea0d6bf37fa#file-mk-cross-builder-armhf_9-default_docker_build-log-L3132)

@zultron 
>This confusing notation means "only run this block if $DISTRO_VER equals 8". 

No wonder I messed up yesterday attempting to rewrite that logic,
I can't wrap my head around:

     (greater than or equal (-gt) ) (or (||) ) = only if equal ...

the-snowwhite, 2019-08-30:
[Here the script is installing native (arm64) packages instead of forigin arch (armhf, arm64)Dockerfile#L298](https://github.com/Dovetail-Automata/mk-cross-builder/blob/master/Dockerfile#L298)

zultron, 2019-08-30:
Actually, "(greater than or equal (-gt) ) (or (||) ) x = do x only if less than or equal".  I think at the time I was experimenting with getting the syntax more brief, where we now see I simultaneously succeeded and failed.  :P

Anyway, I just ran through the `armhf_9` Docker image and MK package build, no problems:

    $ cd mk-cross-builder
    $ ./hooks/build zultron/mk-cross-builder:armhf_9
    $ cd ../machinekit-hal
    $ scripts/build_docker -t armhf_9 -c deb -i zultron/mk-cross-builder

So if the image you built is suspect, you should simply pull the one we use in production:

    $ docker pull dovetailautomata/mk-cross-builder:armhf_9
    $ scripts/build_docker -t armhf_9 -c deb

Can you confirm that these work?

the-snowwhite, 2019-08-30:
Oh yes the one in production works flawlessly here... however I can't just pull a arm64/aarch64_9/10
before I'm able to build at leat one revision....:-)

zultron, 2019-08-30:
I got through a whole image build with [this branch][1].  MK-hal doesn't build yet after the changes in [this branch][2], but it's super late here and I have to pass out.  I don't remember how I fooled `dpkg-buildpackage` into finding build deps for `armhf`, a question for tomorrow.

[1]:  https://github.com/zultron/mk-cross-builder/tree/arm64
[2]: https://github.com/zultron/machinekit-hal/tree/arm64

the-snowwhite, 2019-08-30:
@zultron 
OK I noticed some of your inspired work ..great
meanwhile I found out I got the SYS_ROOT variable wrong and missed spme variables in my former build attempts,
the correct docker buildline is:

    docker build -t mk-cross-builder:armhf_9 --build-arg BASE_IMAGE=debian:stretch --build-arg DISTRO_VER=9 --build-arg DEBIAN_ARCH=armhf --build-arg DISTRO_CODENAME=stretch --build-arg SYS_ROOT="/sysroot" --build-arg HOST_MULTIARCH=arm-linux-gnueabihf --build-arg LDEMULATION="armelf_linux_eabi" .

And I now got .deb packages freshly compiled, from my home built mk_cross_builder docker container.
Can't wait to follow up on the aarkh64 Ultra96 work ... :-1:

zultron, 2019-08-31:
Your `docker build` command is building for `armhf`, not `arm64` architecture.  I must have missed that you weren't trying to produce `arm64` binaries, and in my confusion didn't clearly point out that this command would have gotten your Docker image built without the headache, sorry:

    ./hooks/build mk-cross-builder:armhf_9

As far as building for `arm64`, I found my typo on the MK side that broke the build, and this time made it as far as finding the problem with missing `inb()`/`outb()` on `aarch64`.  This has been seen before [in the MK project][1].

[1]:  https://github.com/machinekit/machinekit/issues/1269

the-snowwhite, 2019-08-31:
OK
About the missing io.h:
In my aarch rip build attempt I got past that using @ArcEye 's   [solution](https://github.com/machinekit/machinekit/issues/1269#issuecomment-326965404) umodified as
I did not know what to do about the 

    #define PCI_IOBASE   0xfee00000UL

Entry (Ultra96 has no pci implementation), or mow to find out the base module loading address on the Ultra96.

This solution worked well enough to make me able to compile finish and run the simulator

the-snowwhite, 2019-09-01:
I forked your debbuilder repo and added to your [branch](https://github.com/the-snowwhite/mk-cross-builder/tree/arm64) a the @ArcEye  fix for the sys/io.h and another fix for the next

    ld: cannot open linker script file ldscripts/aarch64elf.xr: No such file or directory

I found  [here](https://www.labbott.name/blog/2018/07/10/the-cabbage-patch-for-linker-scripts/) next I then ran into this error:

    touch build-stamp
    fakeroot debian/rules binary-arch
    ## have to test after build or Makefile.inc will not exist ##
    cp debian/machinekit-hal-posix.install.in debian/machinekit-hal-posix.install
    cp debian/machinekit-hal-rt-preempt.install.in debian/machinekit-hal-rt-preempt.install
    cp debian/machinekit-hal-xenomai.install.in debian/machinekit-hal-xenomai.install
    if (grep ^USERMODE_PCI=yes src/Makefile.inc -q); then \
        echo "usr/include/linuxcnc/userpci/*.h" >> debian/machinekit-hal-posix.install; \
        echo "usr/include/linuxcnc/userpci/*.h" >> debian/machinekit-hal-rt-preempt.install; \
        echo "usr/include/linuxcnc/userpci/*.h" >> debian/machinekit-hal-xenomai.install; \
    fi
    ## only want this for armhf builds ##
    if [ ! -f debian/platform_pc ] ; then \
        echo "usr/lib/linuxcnc/prubin/*" >> debian/machinekit-hal-posix.install; \
        echo "usr/lib/linuxcnc/prubin/*" >> debian/machinekit-hal-rt-preempt.install; \
        echo "usr/lib/linuxcnc/prubin/*" >> debian/machinekit-hal-xenomai.install; \
    fi
    dh_testdir
    dh_testroot
    dh_prep
    mkdir -p debian/tmp/usr/lib/python2.7/dist-packages/gladevcp
    dh_installdirs
    mkdir -p debian/tmp
    (cd debian/extras && cp -a * ../tmp)
    (cd src; export DESTDIR=`pwd`/../debian/tmp; /usr/bin/make V= install)
    make[1]: Entering directory '/home/mibr/Developer/ext-repos_git/machinekit-hal/src'

    install -d -m 0755 -o root  /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/lib/linuxcnc/posix \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/etc/linuxcnc /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/bin \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/include/linuxcnc \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/share/doc/linuxcnc /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/share/linuxcnc/ncfiles \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/libexec/linuxcnc \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/lib/tcltk/linuxcnc/scripts \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/share/man/man1 \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/share/man/man3 \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/share/man/man9 \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/etc/rsyslog.d \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/etc/security/limits.d \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/etc/udev/rules.d \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/share/linuxcnc/examples/sample-configs \
            /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp../include/userpci
    install -d -m 0755 -o root      /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/include/linuxcnc/userpci
    install -m 4755 -o root ../libexec/rtapi_app_* /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/libexec/linuxcnc
    install -d -m 0755 -o root /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/lib/python2.7/dist-packages/machinekit
    install -d -m 0755 -o root /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/lib/python2.7/dist-packages/machinekit/nosetests
    install -m 4755 -o root ../libexec/pci_write /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/libexec/linuxcnc
    install -d -m 0755 -o root /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/lib/python2.7/dist-packages/machinetalk
    install -m 4755 -o root ../libexec/pci_read /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/libexec/linuxcnc
    install -d -m 0755 -o root /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/lib/python2.7/dist-packages/machinetalk/protobuf
    install -d -m 0755 -o root /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/lib/python2.7/dist-packages/fdm
    install -m 0755 -o root ../libexec/inivar /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/libexec/linuxcnc
    install -d -m 0755 -o root /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/lib/python2.7/dist-packages/drivers
    install -m 0755 -o root ../libexec/rtapi_msgd /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/libexec/linuxcnc
    install -m 0755 -o root ../bin/hal_input /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/bin
    install -m 0755 -o root ../libexec/flavor /home/mibr/Developer/ext-repos_git/machinekit-hal/src/../debian/tmp/usr/libexec/linuxcnc
[... cut ...]
    dh_install: machinekit-hal-posix missing files: usr/lib/linuxcnc/prubin/*
    dh_install: Cannot find (any matches for) "usr/lib/linuxcnc/prubin/*" (tried in "debian/tmp" and "debian/tmp")
    dh_install: machinekit-hal-rt-preempt missing files: usr/lib/linuxcnc/prubin/*
    dh_install: missing files, aborting
    debian/rules:144: recipe for target 'install' failed
    make: *** [install] Error 2
    dpkg-buildpackage: error: fakeroot debian/rules binary-arch gave error exit status 2

Not supposed to compile for prubin on aarch64 ? (an armhf depedency....)

luminize, 2019-09-01:
> ## only want this for armhf builds too ##
> if [ -f debian/platform_pc ] ; then \
>     rm -f debian/platform_pc; \
> else \
>     mkdir -p debian/tmp/usr/lib/linuxcnc/prubin; \
>     cp rtlib/prubin/* [...]
I think prubin is beaglebone (PRU) specific.

the-snowwhite, 2019-09-01:
@luminize 
Got that with this patch to @zultron 's current machinekit-hal arm64 branch.
(I have some pending commits in my own machinekit-hal fork, so I paste the patch here)

---

    From 375fbbeeddb275e27fdb1ce64e36260a27c7d896 Mon Sep 17 00:00:00 2001
    From: Michael Brown <producer@holotronic.dk>
    Date: Sun, 1 Sep 2019 20:59:23 +0200
    Subject: [PATCH 2/2] arm64: remove prubin from packaging in aarch6

    ---
    debian/rules.in | 2 +-
    1 file changed, 1 insertion(+), 1 deletion(-)

    diff --git a/debian/rules.in b/debian/rules.in
    index 234bd023b..839f40cf7 100755
    --- a/debian/rules.in
    +++ b/debian/rules.in
    @@ -153,7 +153,7 @@ install: build
        fi
    
        ## only want this for armhf builds ##
    -	if [ ! -f debian/platform_pc ] ; then \
    +	if [ ! -f debian/platform_pc -a ! "$(DEB_HOST_ARCH)" = "arm64" ] ; then \
            echo "usr/lib/linuxcnc/prubin/*" >> debian/machinekit-hal-posix.install; \
            echo "usr/lib/linuxcnc/prubin/*" >> debian/machinekit-hal-rt-preempt.install; \
            echo "usr/lib/linuxcnc/prubin/*" >> debian/machinekit-hal-xenomai.install; \
    -- 
    2.20.1

So now I have installable arm64 machinekit-hal debs onto machinekit-cnc ....

the-snowwhite, 2019-09-01:
Sigh ... running halrun (with re-preempt package) gives these messages:

    Warning - /usr/libexec/linuxcnc/rtapi_app_posix not setuid
    'sudo make setuid' missing?
    msgd:0 stopped
    rtapi:0 stopped
    Warning - /usr/libexec/linuxcnc/rtapi_app_posix not setuid
    'sudo make setuid' missing?
    rtapi_msgd command:  /usr/libexec/linuxcnc/rtapi_msgd --instance=0 --rtmsglevel=1 --usrmsglevel=1 --halsize=524288
    rtapi_app command:  /usr/libexec/linuxcnc/rtapi_app_posix --instance=0
    /usr/bin/realtime: line 237: /usr/libexec/linuxcnc/rtapi_app_posix: No such file or directory
    rtapi_app startup failed; aborting
    halcmd: cant connect to rtapi_app: -1 (uri= uuid=dd080a3c-60cc-45c1-b0ab-fdebf329df9d): rtapi_rpc(): reply timeout

    halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/linuxcnc.log
    E: 19-09-01 23:24:48 dangling 'DEALER' socket created at hal/utils/halcmd_rtapiapp.cc:284
    halcmd: cant connect to rtapi_app: -1 (uri= uuid=dd080a3c-60cc-45c1-b0ab-fdebf329df9d): rtapi_rpc(): reply timeout

    halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/linuxcnc.log
    E: 19-09-01 23:24:54 dangling 'DEALER' socket created at hal/utils/halcmd_rtapiapp.cc:284
    halcmd: cant connect to rtapi_app: -1 (uri= uuid=dd080a3c-60cc-45c1-b0ab-fdebf329df9d): rtapi_rpc(): reply timeout

    halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/linuxcnc.log
    E: 19-09-01 23:24:59 dangling 'DEALER' socket created at hal/utils/halcmd_rtapiapp.cc:284
    Warning - /usr/libexec/linuxcnc/rtapi_app_posix not setuid
    'sudo make setuid' missing?
    ERROR:  msgd:0 failed to exit on its own; sending SIGTERM

Halrun does work with the posix package installed however hm2_socfpga.ol driver is then missing.

While Xilinx petalinux has no official support for rt-preempt I was able to apply the rt patch on the kernel however... same result.

    machinekit@mksocfpga-nano-soc:~$ uname -a
    Linux mksocfpga-nano-soc 4.14.0-rt1 #1 SMP Sun Sep 1 20:59:42 UTC 2019 aarch64 GNU/Linux

the-snowwhite, 2019-09-01:
machinekit@mksocfpga-nano-soc:~$ cat /var/log/linuxcnc.log

    Sep  1 23:24:43 mksocfpga-nano-soc msgd:0: startup pid=4741 flavor=posix rtlevel=1 usrlevel=1 halsize=524288 shm=Posix cc=gcc 7.4.1 20181213 [linaro-7.4-2019.02 revision 56ec6f6b99cc167ff0c2f8e1a2eed33b1edc85d4]  version=v0.2~arm64~a7ec1fc3d
    Sep  1 23:24:43 mksocfpga-nano-soc msgd:0: ØMQ=4.2.1 czmq=4.0.2 protobuf=3.0.0 atomics=gcc intrinsics    libwebsockets=2.0.3
    Sep  1 23:24:43 mksocfpga-nano-soc msgd:0: configured: sha=a7ec1fc3d
    Sep  1 23:24:43 mksocfpga-nano-soc msgd:0: built:      Sep  1 2019 18:38:36 sha=a7ec1fc3d
    Sep  1 23:24:43 mksocfpga-nano-soc msgd:0: register_stuff: actual hostname as announced by avahi='mksocfpga-nano-soc.local'
    Sep  1 23:24:43 mksocfpga-nano-soc msgd:0: zeroconf: registering: 'Log service on mksocfpga-nano-soc.local pid 4741'
    Sep  1 23:24:44 mksocfpga-nano-soc msgd:0: zeroconf: registered 'Log service on mksocfpga-nano-soc.local pid 4741' _machinekit._tcp 49152 TXT "uuid=dd080a3c-60cc-45c1-b0ab-fdebf329df9d" "instance=ea36a960-ccfe-11e9-b804-000ec6e1e19a" "service=log" "dsn=tcp://mksocfpga-nano-soc.local:49152"
    Sep  1 23:25:05 mksocfpga-nano-soc msgd:0: msgd:0: Terminated - shutting down
    Sep  1 23:25:05 mksocfpga-nano-soc msgd:0: zeroconf: unregistering 'Log service on mksocfpga-nano-soc.local pid 4741'
    Sep  1 23:25:05 mksocfpga-nano-soc msgd:0: log buffer hwm: 0% (0 msgs, 0 bytes out of 524288)
    Sep  1 23:25:05 mksocfpga-nano-soc msgd:0: normal shutdown - global segment detached

the-snowwhite, 2019-09-01:
Just to confirm it's not a kernel issue as halrun loads fine in my mk-rip image, so I guess some files are missing...in the deb packaging...

the-snowwhite, 2019-09-01:
I was just going to sleep when my thoughts came to this construct:

    -	if [ ! -f debian/platform_pc ] ; then \
    +	if [ ! -f debian/platform_pc -a ! "$(DEB_HOST_ARCH)" = "arm64" ] ; then \

the platform_pc variable must be set somewhere to differenitiate (originially) between the beaglebone and the pc (non soc) platform's and has then been carried over to mksocfpga (armhf + arm64).

The differentiation between armhf and arm64 and pc can instead be implemented as:

    if [ "$(DEB_HOST_ARCH)" = "armhf" ]; then

or

    if [ "$(DEB_HOST_ARCH)" = "arm64" ]; then

???

zultron, 2019-09-03:
Yes, the `platform_pc` test need to be replaced with testing `debian/rules` arch flags, as you suggest (I already have in my unified modules branch).  We'll need `arm64` PRU support for the BBAI, but to suit your purposes and not break the status quo, I think your second suggestion makes the best sense:

    if [ "$(DEB_HOST_ARCH)" = "armhf" ]; then

zultron, 2019-09-04:
I'm getting close to supporting `arm64` in the Docker images and MK sources.  See [this branch][1].  I think I have a proper fix for the `inb()`/`outb()` issue in there.

@ArcEye, no rush, can we have your custom `libck-dev` packages for Buster `arm64`-architecture in the MK package repo?

[1]: https://github.com/zultron/machinekit-hal/tree/arm64

the-snowwhite, 2019-09-11:
OK
I just had a long fight getting the Ultra96 sd-image properly setup with:
rt prempt kernel
Correct devicetree settings
Bitfile generated by Vivado
For now its just a messy hack that now gets detected by my earlier made Machinekit aarch64 debs.
@zultron 
I'm now ready once again to catch up on your work [here:](https://github.com/zultron/machinekit-hal/tree/arm64)

the-snowwhite, 2019-09-11:
This is my latest linuxcnc log:

    `    machinekit@mksocfpga-ultra96-soc:~$ cat /var/log/linuxcnc.log

    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: startup pid=4692 flavor=rt-preempt rtlevel=5 usrlevel=5 halsize=524288 shm=Posix cc=gcc 6.3.1 20170404  version=v0.2~arm64~456745530
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: ØMQ=4.2.1 czmq=4.0.2 protobuf=3.0.0 atomics=concurrencykit 0.5.2   libwebsockets=2.0.3
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: configured: sha=456745530
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: built:      Sep  2 2019 23:37:42 sha=456745530
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: register_stuff: actual hostname as announced by avahi='mksocfpga-ultra96-soc.local'
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: zeroconf: registering: 'Log service on mksocfpga-ultra96-soc.local pid 4692'
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 2:rtapi_app:4697:user rtapi:0: cannot create core dumps - /proc/sys/fs/suid_dumpable contains 0
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 2:rtapi_app:4697:user you might have to run 'echo 1 > /proc/sys/fs/suid_dumpable' as root to enable rtapi_app core dumps
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user rtapi.so default iparms: ''
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user RTAPI:0  rt-preempt v0.2~arm64~456745530 init
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 3:rtapi_app:4697:user libcgroup initialization failed: (50001) Cgroup is not mounted
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user rtapi: loaded from rtapi.so
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user hal_lib.so default iparms: ''
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user rtapi_app_main:195 HAL: initializing RT hal_lib support
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user halg_xinitfv:90 HAL: initializing component 'hal_lib' type=4 arg1=0 arg2=0/0x0
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user hal_heap_addmem:58 HAL: extending arena by 262144 bytes
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user halg_export_xfunctfv:85 HAL: exporting function 'newinst' type 2 fp=0 owner=66
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user halg_export_xfunctfv:85 HAL: exporting function 'delinst' type 2 fp=0 owner=66
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user halg_xinitfv:271 HAL: singleton component 'hal_lib' id=66 initialized
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user rtapi_app_main:199 HAL: RT hal_lib support initialized rc=66
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user hal_lib: loaded from hal_lib.so
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user accepting commands at ipc:///tmp/0.rtapi.dd080a3c-60cc-45c1-b0ab-fdebf329df9d
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 3:rtapi_app:4697:user rtapi_app:0 ready flavor=rt-preempt gcc=6.3.1 20170404 git=v0.2~arm64~456745530
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user pid=4697 flavor=rt-preempt gcc=6.3.1 20170404 git=v0.2~arm64~456745530
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user pid=4697 flavor=rt-preempt gcc=6.3.1 20170404 git=v0.2~arm64~456745530
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: ulapi:4698:user _ulapi_init(): ulapi rt-preempt v0.2~arm64~456745530 loaded
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: ulapi:4698:user halg_xinitfv:271 HAL: singleton component 'hal_lib4698' id=70 initialized
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: hal_lib:4698:user --halcmd ping
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: hal_lib:4698:user halg_exit:293 HAL: removing component 72 'halcmd4698'
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: hal_lib:4698:user ulapi_hal_lib_cleanup:235 HAL: lib_module_id=70
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: hal_lib:4698:user halg_exit:293 HAL: removing component 70 'hal_lib4698'
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: hal_lib:4698:user halg_exit:315 HAL: hal_errorcount()=0
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: hal_lib:4698:user halg_exit:316 HAL: _halerrno=0
    Sep 11 13:15:42 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user pid=4697 flavor=rt-preempt gcc=6.3.1 20170404 git=v0.2~arm64~456745530
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: ulapi:4703:user _ulapi_init(): ulapi rt-preempt v0.2~arm64~456745530 loaded
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: ulapi:4703:user halg_xinitfv:271 HAL: singleton component 'hal_lib4703' id=74 initialized
    Sep 11 13:15:42 mksocfpga-ultra96-soc msgd:0: hal_lib:4703:user --halcmd -kf
    Sep 11 13:15:43 mksocfpga-ultra96-soc msgd:0: zeroconf: registered 'Log service on mksocfpga-ultra96-soc.local pid 4692' _machinekit._tcp 0 TXT "uuid=dd080a3c-60cc-45c1-b0ab-fdebf329df9d" "instance=7df71dc8-d485-11e9-9a87-000ec6e1e19a" "service=log" "dsn=ipc:///tmp/0.log.dd080a3c-60cc-45c1-b0ab-fdebf329df9d"
    Sep 11 13:15:47 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user hostmot2.so default iparms: ''
    Sep 11 13:15:47 mksocfpga-ultra96-soc rtapi:0: 4:rtapi_app:4697:user halg_xinitfv:90 HAL: initializing component 'hostmot2' type=1 arg1=0 arg2=0/0x0
[... cut ...]
    Sep 11 13:15:47 mksocfpga-ultra96-soc msgd:0: hal_lib:4697:rt hm2_soc_ol: loading Mesa AnyIO HostMot2 socfpga overlay driver version 0.9
    Sep 11 13:15:47 mksocfpga-ultra96-soc msgd:0: rtapi_app exit detected - scheduled shutdown
    Sep 11 13:15:49 mksocfpga-ultra96-soc msgd:0: msgd shutting down
    Sep 11 13:15:49 mksocfpga-ultra96-soc msgd:0: zeroconf: unregistering 'Log service on mksocfpga-ultra96-soc.local pid 4692'
    Sep 11 13:15:49 mksocfpga-ultra96-soc msgd:0: log buffer hwm: 0% (5 msgs, 406 bytes out of 524288)
    Sep 11 13:15:49 mksocfpga-ultra96-soc msgd:0: normal shutdown - global segment detached
`

the-snowwhite, 2019-09-11:
@zultron 
Seems like something got off on the wrong foot in first commit in [your patch ](https://github.com/zultron/machinekit-hal/commit/39570f768a0f2c886337d8e40e1c2be476b95ff7#diff-3c97568e02244484e9302d85471d8c30L613)
Definitely want to change the Platform to MKSOCFPGA instead of BEAGLEBONE (no BB style pru),
I'm currently trying to work thru the inb()/outb() issue's as something (probably PCI_PLATFORM) is disabling compilation of all other mesa cores than hm2_soc_ol ...

zultron, 2019-09-12:
@the-snowwhite, glad you're looking at that branch.

> Seems like something got off on the wrong foot in first commit in [your patch [...]

Are you talking about the `TARGET_PLATFORM_*` Makefile variable and `config.h` macro?

That hunk sets `TARGET_PLATFORM_BEAGLEBONE` on `aarch64` in order to support the PRUs on the BBAI.  Following the `armhf` architecture build rule, `TARGET_PLATFORM_BEAGLEBONE` gets set by default on all ARM architectures.

Maybe the problem you're seeing is your platform needs the same treatment.  I'm not familiar with the various ARM boards, so I selfishly added only my own target platform, but sounds like we should add it for [`TARGET_PLATFORM_SOCFPGA`][1] also.

[1]: https://github.com/zultron/machinekit-hal/blob/39570f768a0f2c886337d8e40e1c2be476b95ff7/src/configure.ac#L581

> I'm currently trying to work thru the inb()/outb() issue's as something (probably PCI_PLATFORM) is disabling compilation of all other mesa cores than hm2_soc_ol ...

After reviewing the code, I fixed the `inb()`/`outb()` issues in fdfa9732 simply by dropping them from ARM builds.  The commit log explains; TLDR, the simply aren't used in any drivers we normally use on ARM architectures.

> This is my latest linuxcnc log:
> [...]
> Sep 11 13:15:47 mksocfpga-ultra96-soc rtapi:0: 1:rtapi_app:4697:[user signal 7 - 'Bus error' received, dumping core (current dir=/home/machinekit)

Do you think the `inb()`/`outb()` issue is related to this?  Is this a run from my branch or yours?  What does the gdb backtrace on the core file show?

zultron, 2019-09-13:
> There seems to be some confusion as to whats aarch64 and whats armhf
> I initially thought the BBB X15(AI) was 64-bit however:
> the:
> [BBAI](https://beagleboard.org/ai) is [Cortex-A15](https://en.wikipedia.org/wiki/ARM_Cortex-A15) --> armv7,armhf toolchain
> The [Ultra96 ](https://www.96boards.org/product/ultra96)/ [Xilinx Zynq UltraScale+ MPSoC ZU3EG A484](https://www.xilinx.com/products/silicon-devices/soc/zynq-ultrascale-mpsoc.html) is [Cortex A53](https://en.wikipedia.org/wiki/ARM_Cortex-A53) --> ARMv8-A aarch64/arm64 toolchain.
>
> This stil leaves TARGET_PLATFORM_BEAGLEBONE stuck to the 32-bit armhf and not aarch64 ?
> (the aarch64 toolchain thrown lots of errors about prubin.xxx unless these parts are disselected)

Ah, thanks for clearing that up.  My confusion; I've also been playing with a RPi4.  I reverted that change.

I also added `TARGET_PLATFORM_SOCFPGA` to the `aarch64` build; is this desired?  It will build the `hm2_soc_ol` and `hostmot2` modules, and the `mksocmemio` executable.

> @zultron Do I understand correctly that you would prefer the aarch64 work to be introduced via your [older arm64 branch](https://github.com/zultron/machinekit-hal/tree/arm64) instead of on top of the [current official machinekit-hal repo](https://github.com/machinekit/machinekit-hal) ?

No, that was an accident.  I rebased onto the latest master and force-pushed the branch; take a look.

> Ok so I worked thru applying the @zultron inb()/outb() fixes [(plus some crude hacks for adding missing USERMODE_PCI dependable linuxcnc/userpc hostmot2.c requirements)](https://github.com/the-snowwhite/machinekit-hal/commit/84b61435ec439c8fdf31874d5b986b85a2d5f322)

[...]

> Some of the commits on my [arm64-work extension](https://github.com/the-snowwhite/machinekit-hal/tree/arm64-work) branch need to be refined (as I had to break some rules due to needing some obscured [USERMODE_PCI flagged](https://github.com/the-snowwhite/machinekit-hal/commit/84b61435ec439c8fdf31874d5b986b85a2d5f322) dependencies.

I'm so sorry for anyone who has to touch this build system.  My heart goes out to you.  I cleaned up the `USERMODE_PCI` and packaging stuff in my commits e5cd27ac and fc648a3b.

Enabling `USERMODE_PCI` will also reintroduce the requirements for `inb()`/`outb()`, though, right?  Sorry for my ignorant question:  Why do you need `USERMODE_PCI` for the hm2 driver?  Is the FPGA a PCI device on the Ultra96?

I also picked over your `memcpy.S` commit.

So take a look at my [branch].  I don't have time to test right now, but I'm hoping it contains everything your branch does.

We still need to figure out how to address the missing `inb()`/`outb()`, if mksocfpga does indeed require PCI support.  We may have to implement an equivalent for ARM architectures.

[1]: https://github.com/zultron/machinekit-hal/commits/arm64

zultron, 2019-09-13:
> Enabling `USERMODE_PCI` will also reintroduce the requirements for `inb()`/`outb()`, though, right? Sorry for my ignorant question: Why do you need `USERMODE_PCI` for the hm2 driver? Is the FPGA a PCI device on the Ultra96?
> 
[...]
> We still need to figure out how to address the missing `inb()`/`outb()`, if mksocfpga does indeed require PCI support. We may have to implement an equivalent for ARM architectures.

[Here][1] is the related failure in my Travis CI instance.

So the question that still stands, is `USERMODE_PCI` required for Ultra96 support, and if so, any ideas for implementing `inb()`/`outb()` replacements?

[1]: https://travis-ci.org/zultron/machinekit-hal/jobs/584494470#L1354-L1363

the-snowwhite, 2019-09-13:
I would rather introduce a new global variable: HAS_FPGA (or HAS_SOC_FPGA) than revert to using USERMODE_PCI. (this will reintroduce too much unneeded stuff)

The usermode_pci requirement only (AFAIK) blocks 3 HAS_FPGA needed files these ones here: [rtapi/userpci/device.o		  \
￼	    rtapi/userpci/firmware.o		  \
￼	    rtapi/userpci/string.o](https://github.com/the-snowwhite/machinekit-hal/commit/84b61435ec439c8fdf31874d5b986b85a2d5f322#diff-4b21d60060f2cf7b0573027482319e74)
In the (top) src/Makefile their folder is included before the (local) BUILD_HOSTMOT2 variable is introduced and the BUILD_HOSTMOT2 variable also doesn't reach into the folder of the last 3 needed files for the device-tree probing stuff.
In the (top) src/Makefile it is perhaps possible to move [these 2 constructs](https://github.com/the-snowwhite/machinekit-hal/commit/84b61435ec439c8fdf31874d5b986b85a2d5f322#diff-f57f2991a6aa25fe45d8036c51bf8b4dR746) down to [below the point](https://github.com/the-snowwhite/machinekit-hal/commit/84b61435ec439c8fdf31874d5b986b85a2d5f322#diff-f57f2991a6aa25fe45d8036c51bf8b4dR1159) where  the BUILD_HOSTMOT2 variable is introduced and set.
This only leaves the problem with getting a variable in place (HAS_FPGA ?) that like the (Global ?) USERMODE_PCI can reach into [this Submakefile](https://github.com/machinekit/machinekit-hal/blob/84b61435ec439c8fdf31874d5b986b85a2d5f322/src/rtapi/userpci/Submakefile) with a construct like:

    ifeq ($(USERMODE_PCI):$(HAS_FPGA),yes)

The HAS_FPGA variable could also be used to get out of tying the FPGA stuff to cpu architectures..

the-snowwhite, 2019-09-13:
Having said all that...
Images come to my mind that (low cost) arm(64) socs are beginning to have pci support, so this
is probably needing to be fixed.. upstream (rather now than later ?).
Perhaps someone has an Idea of how inb()/outb() and (the missing) memcpy are handled nowadays
with armsocs that have pci support ?

the-snowwhite, 2019-09-13:
@zultron 
On testing your latest branch only 1 commit is missing:

---

[pasted diff for this one](https://github.com/the-snowwhite/machinekit-hal/commit/456745530aeb2fdf400f78927b240bf23f526139)

	diff --git a/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c b/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c
	index bad9b90fa..b2fafacee 100644
	--- a/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c
	+++ b/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c
	@@ -603,7 +603,7 @@ static int instantiate(const int argc, char* const *argv)
			nread = read(fd, blob, st.st_size);
			if (nread != st.st_size) {
				LL_ERR("reading '%s': expected %zu got %u - %s\n", brd->descriptor,
	-                (unsigned) st.st_size, nread, strerror(errno));
	+                (size_t) st.st_size, (unsigned) nread, strerror(errno));
				return -EINVAL;
			}
			close(fd);

This fix also compiles on the current (non arm64) dovetailautomata armhf  docker

-----

As for the missing inb()/outb() I've settled on keeping the USERMODE_PCI flag and including the @ArcEye provided sys/io.h file [(only) in the arm64 build images like in this commit:](https://github.com/the-snowwhite/mk-cross-builder/commit/506406ed4292071c653ad76c67cfc1fad1493825)

the-snowwhite, 2019-09-14:
OK
@zultron 
I took a (actually some) shot(s) at building machinekit-cnc arm64 debs and finally arrived at something installable and able to successfully run axis gui sim (I/O testing will follow tomorrow as its really late here now).
[Actual commits are in this branch](https://github.com/the-snowwhite/machinekit-cnc/tree/arm64).
:-)

the-snowwhite, 2019-09-15:
@zultron 
My (c code) workaround for the missing memcpy.S is only able to read pins not write.
Digging further I found out the the memcpy function comes via:
libglibc
and is included in the aarch64 glib package:

    machinekit@mksocfpga-ultra96-soc:~$ nm -D /lib/aarch64-linux-gnu/libglib-2.0.so.0 | grep memcpy
                 U memcpy
                 U __memcpy_chk

These last lines of my aarch64 package build are perhaps an indication of a libglibc link error:

	dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/machinekit-hal-posix/usr/libexec/linuxcnc/rtapi_msgd was not linked against libatomic.so.1 (it uses none of the library's symbols)
	dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/machinekit-hal-posix/usr/bin/hy_vfd debian/machinekit-hal-posix/usr/bin/gs2_vfd were not linked against libglib-2.0.so.0 (they use none of the library's symbols)
	dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/machinekit-hal-rt-preempt/usr/bin/gs2_vfd debian/machinekit-hal-rt-preempt/usr/bin/hy_vfd were not linked against libglib-2.0.so.0 (they use none of the library's symbols)
	dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/machinekit-hal-rt-preempt/usr/libexec/linuxcnc/rtapi_msgd was not linked against libatomic.so.1 (it uses none of the library's symbols)
	dh_gencontrol
	dpkg-gencontrol: warning: package machinekit-hal-posix: unused substitution variable ${shlibs:Depends}
	dpkg-gencontrol: warning: package machinekit-hal-posix: unused substitution variable ${shlibs:Depends}
	dpkg-gencontrol: warning: package machinekit-hal-rt-preempt: unused substitution variable ${shlibs:Depends}
	dpkg-gencontrol: warning: package machinekit-hal-rt-preempt: unused substitution variable ${shlibs:Depends}

the-snowwhite, 2019-09-15:
I'm stuck at the memcpy issue
I just created a new aar64 rip build with the updated arm64 branches here: [Builder](https://github.com/the-snowwhite/mk-cross-builder/tree/arm64) [machinekit-hal](https://github.com/the-snowwhite/machinekit-hal/tree/arm64) [machinekit-cnc](https://github.com/the-snowwhite/machinekit-cnc/tree/arm64)

---

Here is the gdb backtrace from the rip-build:

	root@ultra96:~# gdb /home/linaro/machinekit-hal/libexec/rtapi_app_rt-preempt /home/linaro/core
	Reading symbols from /home/linaro/machinekit-hal/libexec/rtapi_app_rt-preempt...done.

	warning: core file may not match specified executable file.
	[New LWP 4136]
	[New LWP 4141]
	[New LWP 4140]
	[Thread debugging using libthread_db enabled]
	Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
	Core was generated by `rtapi:0                                                                       '.
	Program terminated with signal SIGBUS, Bus error.
	#0  0x0000007fba85ec68 in memcpy () from /lib/aarch64-linux-gnu/libc.so.6
	[Current thread is 1 (Thread 0x7fba403000 (LWP 4136))]
	(gdb) backtrace
	#0  0x0000007fba85ec68 in memcpy () from /lib/aarch64-linux-gnu/libc.so.6
	#1  0x0000007fb8f7712c in hm2_soc_read (this=0x7fba134ee0, addr=260, buffer=0x7fc47af960, size=8) at hal/drivers/mesa-hostmot2/hm2_soc_ol.c:179
	#2  0x0000007fb8f9faf4 in hm2_register (llio=0x7fba134ee0, config_string=0x7fba214e40 <error: Cannot access memory at address 0x7fba214e40>)
		at hal/drivers/mesa-hostmot2/hostmot2.c:1475
	#3  0x0000007fb8f77c58 in hm2_soc_register (brd=0x7fba134e88, fwid=0x0, fwid_len=0, inst_id=82) at hal/drivers/mesa-hostmot2/hm2_soc_ol.c:477
	#4  0x0000007fb8f78230 in instantiate (argc=4, argv=0x55b917a020) at hal/drivers/mesa-hostmot2/hm2_soc_ol.c:605
	#5  0x0000007fba1718d0 in create_instance (fa=0x7fc47afb98) at hal/lib/hal_comp.c:507
	#6  0x0000007fba1532ac in hal_call_usrfunct (name=0x5585061060 "newinst", argc=4, argv=0x55b917a020, ureturn=0x7fc47afc94) at hal/lib/hal_funct.c:192
	#7  0x0000005585049e1c in do_newinst_cmd (comp="hm2_soc_ol", instname=<error reading variable: Cannot access memory at address 0x0>, args=..., pbreply=..., 
		instance=<optimized out>) at rtapi/rt-preempt/rtapi_app.cc:511
	#8  0x000000558504debc in rtapi_request (loop=<optimized out>, socket=0x55b9169e00, arg=<optimized out>) at rtapi/rt-preempt/rtapi_app.cc:928
	#9  0x0000007fbac728e0 in zloop_start () from /usr/lib/aarch64-linux-gnu/libczmq.so.4
	#10 0x000000558504d440 in mainloop (argc=<optimized out>, argv=<optimized out>) at rtapi/rt-preempt/rtapi_app.cc:1414
	#11 0x0000005585047fd4 in main (argc=3, argv=0x7fc47b0e98) at rtapi/rt-preempt/rtapi_app.cc:1769
	(gdb) 

---
culprit is:

---

hostmot2.c:1475
[        if (!llio->read(llio, HM2_ADDR_CONFIGNAME, name, 8)) {
            HM2_ERR("error reading HM2 Config Name\n");
            r = -EIO;
            goto fail0;
        }
](https://github.com/the-snowwhite/machinekit-hal/blob/arm64/src/hal/drivers/mesa-hostmot2/hostmot2.c#L1475)

---

hm2_soc_ol.c:179
[memcpy(buffer, brd->base + addr, size);](https://github.com/the-snowwhite/machinekit-hal/blob/arm64/src/hal/drivers/mesa-hostmot2/hm2_soc_ol.c#L179) 

---
If the memcpy routine instead is defined as this  c functiom
`void myMemCpy(void *dest, void *src, size_t n) 
{ 
   // Typecast src and dest addresses to (char *) 
   char *csrc = (char *)src; 
   char *cdest = (char *)dest; 
  
   // Copy contents of src[] to dest[] 
   for (int i=0; i<n; i++) 
       cdest[i] = csrc[i]; 
} `

Reads perform well however no writes work

the-snowwhite, 2019-09-15:
OK but odd

memcpy in aarch64 libglibc.so has a different/broken read functionality:
This is tested both in my arm64 rip build and aarch64 package build ultra96 images
in hm2_soc_ol.c:
replacing only the read function with the c function like this:

	static void myMemCpy(void *dest, void *src, size_t n) 
	{ 
	// Typecast src and dest addresses to (char *) 
	char *csrc = (char *)src; 
	char *cdest = (char *)dest; 
	
	// Copy contents of src[] to dest[] 
	for (int i=0; i<n; i++) 
		cdest[i] = csrc[i]; 
	} 

	//
	// these are the "low-level I/O" functions exported up
	//
	static int hm2_soc_read(hm2_lowlevel_io_t *this, u32 addr, void *buffer, int size) {
		hm2_soc_t *brd = this->private;

	#ifdef WARN_ON_UNALIGNED_ACCESS
		/* hm2_read_idrom performs a 16-bit read, which seems to be OK, so let's allow it */
		if ( ((addr & 0x1) != 0) || (((size & 0x03) != 0)  && (size != 2))) {
		LL_ERR( "hm2_soc_read: Unaligned Access: %08x %04x\n", addr,size);
		}
	#endif
		myMemCpy(buffer, brd->base + addr, size);
		return 1;  // success
	}

makes me able to set and read / write pins, but not much more as memcpy is used to read in many places, I can attempt reinstalling /lib/aarch64-linux-gnu/libglib-2.0.so.0

the-snowwhite, 2019-09-16:
Well I hope someone has some more solid suggestions as to why memcpy is having memory violation problems on the aarch64 platform, and perhaps can point out what I'm unable to see....

zultron, 2019-09-16:
Sorry, @the-snowwhite, I don't have any hardware to look at your `memcpy()` problem on.

    #0  memcpy () at ../sysdeps/aarch64/memcpy.S:102
    102     ../sysdeps/aarch64/memcpy.S: No such file or directory.

I think this error really means that `memcpy()` is returning `ENOENT`, and not that the `memcpy.S` file is missing.  That's surely related to the bus error.  I've never encountered one of those, but wikipedia says it's similar to a segmentation violation except that it refers to an invalid address.

I'd start by reverting the `myMemCpy()` patch and looking at the operands to `memcpy()` to see if they're valid or not, `brd->base + addr` first.

If the hm2-related unit tests don't run correctly, that could be an easy place to start a rapid debugging cycle of run-test/tweak-source/build/repeat iterations.

Good luck!

Let me ask one more time, why does `hm2_soc_ol.c` need  `#include "rtapi_io.h"`?  There aren't actually any `inb()` and `outb()` function calls there.  Can you just remove the `#include` and drop the more lengthy `inb()`/`outb()` replacement?

the-snowwhite, 2019-09-16:
@zultron 
hm2_soc_ol.c does NOT need #include "rtapi_io.h" (I removed it)
The reason for the inb()/outb() replacement is for all the PCI related stuff that gets dragged in by the USERMODE_PCI flag of which the Socfpga platform strictly only needs the stuff in the:  [src/rtapi/userpci/](https://github.com/the-snowwhite/machinekit-hal/blob/master/src/rtapi/userpci/firmware.c) folder.
However up the line I see the the next arm64 user may have a board with a pci slot and wants to add a "normal" mesa card, Then the PCI_BASE_ADDRESS can be adjusted and everyone is happy ?

---
memcpy:
First I'm able to get the MyMemCpy function to do I/O work if I set it to copy the bytes backwards (Count Top to down), problem is I get a 240 ns (low) glitch every 1000 ms(baseperiod) If I set a pin to high.
I have been searching for the actual string.h prototype used, to see if the operands to memcpy are valid and found [it here I think](https://packages.debian.org/stretch/libc6-dev-arm64-cross):

The definition reads:

    __BEGIN_NAMESPACE_STD
    /* Copy N bytes of SRC to DEST.  */
    extern void *memcpy (void *__restrict __dest, const void *__restrict __src,
		     size_t __n) __THROW __nonnull ((1, 2));

I have no clue as to what void *__restrict does.

the-snowwhite, 2019-09-16:
Excerpt from the [arm porting guide](https://community.arm.com/cfs-file/__key/communityserver-blogs-components-weblogfiles/00-00-00-21-42/Porting-to-ARM-64_2D00_bit-whitepaper.pdf)

    64-bit data models
    There are various standard data models in use today. They differ mainly in the size
    defined for integers, longs and pointers.

[image]

    64-bit Linux implementations use LP64 and this is supported by the defined A64-LP64
    Procedure Call Standard A64-LP64. Other PCS variants may be defined which support
    LLP64 (used by 64-bit Windows) and ILP32 (a variant which can ease porting 32-bit
    software which doesn’t need 64-bit addressing and can also improve performance of
    applications which use a lot of pointers but do not require 64-bit addressing).
    When compiling for ARMv7-A, you will most likely have used a data model equivalent
    to what is shown as ILP32 in the table.
    Note that the ARM compiler for ARMv8 accepts __int64 as a synonym for “long long”.

[61 more replies]