machinekit/machinekit-hal #122: Machinekit docs are listing oldest Debian version first

machinekoder, 2018-08-04:
The docs page: http://www.machinekit.io/docs/getting-started/installing-packages/

Lists the Wheezy installs first. Wheezy is EOL, and in general, a new user might want to use Stretch so I suggest changing the order to list the newest version first.

ArcEye, 2018-08-04:
Agreed.  

Wheezy is not only EOL as far as Debian is concerned, we have not produced packages for it since 20th March 2018.

There probably also needs to be a note to explain it is EOL, package support is frozen and no new versions will be produced.

I will move this to machinekit-docs however, was added to machinekit-hal for some reason.

ArcEye, 2018-08-04:
Issue moved to machinekit/machinekit-docs/issues/#285

closed 2018-08-04 by ArcEye