machinekit/machinekit-hal #146: Passing instance parameters as an array of names to an instantiatiable component

luminize, 2018-09-20:
I'm writing a component (in c) that should take a configuration string like "pins=bbfu" which will create pins of types bool, bool, float and u32, and an optional string like "names=foo,bar,baz,bung" that depicts the name of these pins.

However I do not seem to be able to pass these as I were able with non-instantiatiable components. There I would define like:

ArcEye, 2018-09-20:
See the watch component I just added.
https://github.com/machinekit/machinekit/blob/master/src/hal/user_icomps/watch.c

The code shows how to get the args from argv and the sim hal file shows the instantiation line.
https://github.com/machinekit/machinekit/blob/master/configs/sim/axis/watch-demo-sim/custom_postgui.hal#L10
eg
```
newinst watch w1 --- pin_name=scounter.count preset_name=pyvcp.spinbox.0.param_pin preset_type=0 target_value=71.0 forever=1`
```

ArcEye, 2018-09-20:
OK I see you have filled in the second half of your question, it was missing when I answered.

The only safe way to pass strings to an instantiable component, is through the argc/argv mechanism.

This is because the only string storage allocated is for the kernel parameters relating to the original instance.
Subsequent instances reuse the string allocation address and if the new string is too long, it will cause a segfault on overrunning the buffer.

Kernel modules were only ever designed to be loaded once and it is the legacy kernel parameters which are used.  Even the int parameters need to be voided after they are read, because they will be reused and if not set by the next instance will remain at the value set by the last one.

See in the code, that pincount is set to -1 after being stored
https://github.com/machinekit/machinekit/blob/master/src/hal/user_icomps/watch.c#L265
instcomp will create code which does the same.

Using `extra_inst_setup()` is the cleanest way to do your arg fetching, albiet if just one or two, you could do it inside `instantiate()` at the same point.

luminize, 2018-09-20:
right, so argv[3] has my `"samples=bbfu"` string, this I parse and copy the "bbfu" to the global char `char samples[MAX_PINS]` variable. Then afterwards I need to clear the global `samples[0]='\0';`

ArcEye, 2018-09-20:
@mhaberler 's example
https://github.com/machinekit/machinekit/blob/master/src/hal/icomp-example/icomp.c#L124
shows a use of the get_opts() library functions to parse the args.

If you have a limited number of args and know exactly what they are, you may think my for() loop iterating through them is simpler.
Multiple feline fur removing techniques exist :stuck_out_tongue_winking_eye:

ArcEye, 2018-09-20:
> right, so argv[3] has my "samples=bbfu" string, this I parse and copy the "bbfu" to the global char char samples[MAX_PINS] variable. Then afterwards I need to clear the global samples[0]='\0';

Whatever method you use, you cannot assume that a particular string will be at a particular index.
If someone uses `names=` before `samples=`, it will break.
Iterate through all the args with no pre-conception of what they contain.

I would have to see your code.
Generally you would void a string by writing `\0` to the first char before it is used, then you can test if it was set and use it if it was.

luminize, 2018-09-20:
ah, yes, another cat-skinning technique :)
yes, I'm doing just that but I wrote too quickly, like in your watch component, parse all args, see if there's a match, and if so take action.

Here's a different question about allocating memory:
I have a struct in the instance data, containing arrays of data:
```
struct inst_data {
    ....
    struct sample   *pin_sample;      
    ....
}

}
struct sample {
    __u64       timestamp;		        // timestamp of measurement
    int         sample_setup[6];        // array depicting number of type of pins
    hal_bit_t   *arr_bit[0];		    // arrays of sample values
    hal_float_t *arr_flt[0];
    hal_u32_t   *arr_u32[0];
    hal_s32_t   *arr_s32[0];
};
```
do I need to know the exact size of the array before instantiating HAL memory? iow first parse the amount of the types (got that from the `samples=` parameter, and add that size to the size of the instance data struct? Or do I need to set the arrays to a fixed length?

example:
```
    pin_arrays_size = *pin_setup[i] * sizeof(hal_bit_t *);
    pin_arrays_size += *pin_setup[i] * sizeof(hal_float_t *);
    pin_arrays_size += *pin_setup[i] * sizeof(hal_u32_t *);
    pin_arrays_size += *pin_setup[i] * sizeof(hal_s32_t *);

    // instantiate component
    if ((inst_id = hal_inst_create(name, comp_id,
        sizeof(struct inst_data) + pin_arrays_size,
        (void **)&ip)) < 0)
	return -1;
```

ArcEye, 2018-09-20:
NB. The `struct sample` needs to be defined before `inst_data`.
You could forward declare, but there is little point.

`inst_data` must allocate space for the maximum number of items an array can hold.
You cannot use dynamic runtime allocation here, it is a fixed definition structure.

This is what the pincount variable and MAXCOUNT does.
Its maximum possible size is used for the amount of storage allocated, so that whatever value pincount is set to by subsequent instantiations, it can accommodate them.

This code snippet from debounce.c (generated by instcomp) shows the actual memory allocation is hardcoded to the MAXCOUNT
```
#define MAXCOUNT 16
#define DEFAULTCOUNT 8

static int pincount = 8;
RTAPI_IP_INT(pincount, "");

struct inst_data
    {
    hal_bit_t *_in[16];
    hal_bit_t *_out[16];
    hal_s32_t *_state[16];
    hal_s32_t *delay;
    int local_argc;
    char **local_argv;
    int local_pincount;
    };

static int maxpins __attribute__((unused)) = 16;
```
but that the pin allocation uses the `pincount` param passed, to create a number of pins which is within that maximum
```
    z = pincount;
    if(z > maxpins)
       z = maxpins;
    for(j=0; j < z; j++)
        {
        r = hal_pin_bit_newf(HAL_IN, &(ip->_in[j]), owner_id,
            "%s.%01d.in", name, j);
        if(r != 0) return r;
        }
```

luminize, 2018-09-20:
right, thanks for the detailed explanation!

luminize, 2018-09-20:
one more Q:
It looks like arguments are only picked up when passing also the three dashes. Is that intentional? Without the three dashes iterating argv[] notifies with:
```
Sep 20 17:05:26 TP51 rtapi:0: passing args for 'ir' failed: 'samples=bbfsu'
Sep 20 17:05:26 TP51 rtapi:0: 1:rtapi_app:29258:user Unknown parameter `samples=bbfsu'
Sep 20 17:05:26 TP51 rtapi:0: 1:rtapi_app:29258:user passing args for 'ir' failed: 'samples=bbfsu'
```

and when passing the 3 dashes I can iterate without problem:
```
Sep 20 17:05:53 TP51 rtapi:0: 4:rtapi_app:29258:user do_newinst_cmd: instargs='--- samples=bbfSsuSU'
```

ArcEye, 2018-09-20:
It is very intentional.

Every arg after the component name and before the `---` delimiter is passed to the component as RTAPI_IP_XX params.
These are kernel parameters and do not appear in the component argv

Those args after the `---` delimiter are pushed into the argc/argv mechanism.

So for the line
```
newinst watch w1 pincount=1 --- pin_name=scounter.count preset_name=pyvcp.spinbox.0.param_pin preset_type=0 target_value=71.0 forever=1
```

newinst is the original executable, receiving all space delimited args in its argv

It creates an instance of watch, called `w1` with any args up to the delimiter.
In this case `pincount=1`

The w1 component gets passed the original argv[1] onwards, less the kernel parameter and the delimiter
argv[0] is `watch`
argv[1] is `w1`
argv[2] is `pin_name=scounter.count`
etc.

This was the only reliable way of doing it.

I even wrote a routine in hal_lib to do a local strdup() of argv
https://github.com/machinekit/machinekit/blob/master/src/hal/lib/hal_memory.c#L219

This makes a copy of the args within the char **argv in the inst_data struct,
so that they are preserved with no chance of being overwritten by the next instantiation.
instcomp generates this automatically and frees it before exit.

This probably does not matter if the global ones are only used at instantiation, but is used later in the function() that could misfire otherwise and local_argv[x] should be used.

again from debounce.c
```
struct inst_data
    {
    hal_bit_t *_in[16];
    hal_bit_t *_out[16];
    hal_s32_t *_state[16];
    hal_s32_t *delay;
    int local_argc;
    char **local_argv;
    int local_pincount;
    };
.....
static int export_halobj(........)
{
...
ip->local_argv = halg_dupargv(1, argc, argv);
ip->local_argc = argc;
.....
}
```

luminize, 2018-09-21:
Got it. There's a lot that's going on. Thanks!

closed 2018-09-21 by luminize