machinekit/machinekit-hal #150: Jessie support timeline and CI issues

zultron, 2018-10-11:
How long will the Machinekit project continue supporting Jessie?

There may be an issue updating the cross-build Docker images we use in Travis.  @dkhughes alerted me over in Dovetail-Automata/mk-cross-builder#3 that the cross-builder Docker images have started failing for Jessie because the Emdebian repo has fallen out of sync.  I noticed a [recent post][1] on the debian-embedded list implying that Emdebian has been abandoned and is broken.

This may be a non-issue since the current images may continue to work without updates until Jessie eventually becomes history, but any information about the project's plans would help in case I decide to update the images (which I may do soon to support cpusets and Buster).

[1]: https://lists.debian.org/debian-embedded/2018/09/msg00000.html

ArcEye, 2018-10-11:
I don't think there are any plans per se.
We normally support until EOL, which is what we did with Wheezy.
2020 in the case of Jessie.

We don't make any new packages, advise people to switch to newer distro but retain last set of packages for usage of those who can't or won't change.

I suspect a lot of BBB users are now on Stretch, because they are on later rt-preempt kernels.
However there will be a silent mass of users still on whatever they started with, because it works so why change :smile:

zultron, 2018-12-05:
Thanks, @ArcEye.  Based on your and @dkhughes's feedback, and cribbing from some of @dkhughes's work, I [fixed up][1] `mk-cross-builder` to use Linaro's cross-build toolchain instead of the defunct Emdebian's.

I don't want to push images straight to `dovetailautomata/mk-cross-builder` on the Docker hub in case there's a problem, since the images there can't be reproduced.  For now, I'm testing out of [`zultron/mk-cross-builder`][2], where my test [cpusets PR][3] is [almost passing][4].  (I'm getting hangs on `tests/rtapi.0` on Jessie only, in case you've seen that one.)

[1]: https://github.com/zultron/mk-cross-builder/commit/5d4503e
[2]: https://hub.docker.com/r/zultron/mk-cross-builder/builds/
[3]: https://github.com/zultron/machinekit/pull/8
[4]: https://travis-ci.org/zultron/machinekit/builds/463723679

ArcEye, 2018-12-05:
@zultron Let me know when you think they are ready for testing.

I have my own Jenkins server, which is where I first tested the current machinekit builders before deploying them.
I can set up separate jobs to use your new dockers across the range of builds and give them a good work out before you finally deploy.

Alternately, name them something slightly different in your docker hub
eg. `dovetailautomata/mk-linaro-cross-builder` and I can switch to them with a 20 second cut and 
paste in Jenkins and just swap back if there is a problem.

zultron, 2018-12-05:
Thanks, @ArcEye.  I'll ask you to point builds at `zultron/mk-cross-builder` at some point.

For now, I'm still having trouble getting the `amd64_8` tests to pass, even on `master` with the original `dovetailautomata/mk-cross-builder:amd64_8` container image!  This is true in Travis and also running `scripts/build_docker -t amd64_8 -c test` on my PC.

It's easy to isolate in a single test; after the RIP build, `scripts/build_docker -t amd64_8 -c shell bash -c ". scripts/rip-environment; runtests -n tests/remap/remap-io"` hangs with the following in the `stderr` file:
```
+ linuxcnc -r test-ngc.ini
rtapi_msgd command:  [...]/libexec/rtapi_msgd --instance=0 --rtmsglevel=1 --usrmsglevel=1 --halsize=524288
rtapi_app command:  [...]/libexec/rtapi_app_posix --instance=0
[...]/scripts/linuxcnc: line 545:   377 Segmentation fault      (core dumped) $HALCMD stop
[...]/scripts/linuxcnc: line 545:   380 Segmentation fault      (core dumped) $HALCMD unload all
[...]/scripts/realtime: line 308:   470 Segmentation fault      (core dumped) halcmd shutdown
```

FYI, no access to Jenkins from China.  If it's hosted on Michael's server, it's blocked, same as the MK repos.

Are the Travis builds now completely disabled for the Machinekit repos?

ArcEye, 2018-12-05:
> Are the Travis builds now completely disabled for the Machinekit repos?

Yes they are and all sorts of IPs, especially from CN, are blocked on the server :)

I will look at the `zultron/` Docker page tomorrow and pull the dockers to test locally.
I think that one does not usually produce problems, it was hm2-idrom I had to remove from the build tests because it always failed in a docker.

zultron, 2018-12-05:
Ah ha, looks like the CI system doesn't run the tests for Jessie.  We don't know when that broke, but it wasn't related to this work, anyway.

zultron, 2018-12-05:
So the new containers are [looking good][1].  I'm going to take a crack at snuffing out those last compiler warnings, and submit a few PRs soon.  Thanks, and stay tuned!

[1]: https://jenkins.machinekit.io/job/machinekit-PR-builder/135/

ArcEye, 2018-12-06:
@zultron 
I have tested all 6 containers for deb builds locally thus far

armhf_8 consistently fails with this error
```
Compiling realtime xenomai rtapi/rtapi_shmem.c
Compiling realtime xenomai rtapi/rtapi_time.c
Compiling realtime xenomai rtapi/rtapi_exception.c
Compiling realtime xenomai rtapi/rtapi_heap.c
Compiling realtime xenomai rtapi/xenomai.c
Compiling realtime xenomai rtapi/rtapi_main.c
rtapi/xenomai.c: In function ‘_rtapi_wait_hook’:
rtapi/xenomai.c:382:6: warning: ‘return’ with no value, in function returning non-void [-Wreturn-type]
      return;
      ^
Compiling realtime xenomai rtapi/rtapi_compat.c
Compiling realtime xenomai machinetalk/support/nanopb.c
Compiling realtime xenomai machinetalk/nanopb/pb_decode.c
Compiling realtime xenomai machinetalk/nanopb/pb_encode.c
Compiling realtime xenomai machinetalk/nanopb/pb_common.c
Compiling realtime xenomai rtapi/rtapi_pci.c
Compiling realtime xenomai hal/lib/hal_lib.c
rtapi/rtapi_pci.c:34:21: fatal error: libudev.h: No such file or directory
 #include <libudev.h>
                     ^
compilation terminated.
make[2]: *** [objects/xenomai/rtapi/rtapi_pci.o] Error 1
Makefile:1672: recipe for target 'objects/xenomai/rtapi/rtapi_pci.o' failed
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/root/src/machinekit/src'
Makefile:126: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/root/src/machinekit/src'
debian/rules:82: recipe for target 'build-stamp' failed
make: *** [build-stamp] Error 2
dpkg-buildpackage: error: debian/rules build-arch gave error exit status 2
```

However armhf_9 completes and builds

Looks like missing `libudev-dev:armhf` in jessie ?

zultron, 2018-12-06:
This is because of a difference in the Emdebian compiler:  for some reason, even though the `--sysroot=/sysroot` argument wasn't consistently applied to RT compiles, it still was able to locate headers in `/sysroot/usr/include`.  Commit [a377a9a5][1] fixes this inconsistent application of `$CPPFLAGS` so that it works with the Linaro compiler.

The compile will break on `master`, but will work on my `cpusets` branch.  See yesterday's [Jenkins job][2] where I forced the container image selection to the `zultron/mk-cross-builder` tags:  the build passed on all six images.

[1]: https://github.com/zultron/machinekit/pull/8/commits/a377a9a5
[2]: https://jenkins.machinekit.io/job/machinekit-PR-builder/135/

zultron, 2019-04-01:
I think this question is answered:  Jessie support will continue on for long enough that it's worth going through the pain to replace Emdebian with Linaro for `armhf` builds in the Jessie Docker images.  This work has already been done and the new Docker images are in production.  Closing issue.

closed 2019-04-01 by zultron