machinekit/machinekit-hal #164: Jessie `amd64` tests fail in Docker

zultron, 2018-12-10:
Recent tests in the rebuilt containers (see e.g. #163) fail on the `amd64_8` tag.  The CI hasn't been testing Jessie recently, so this problem was introduced at some indeterminate time in the past.

The error is deterministic, in the sense that it can be reproduced reliably, but non-deterministic in the sense that it fails in different tests, depending on the ordering of the tests.  When running the entire test suite, the failure always occurs at  the `rtapi:0` test.

To reproduce, run `./build_docker -t amd64_8 -c test`.  The `rtapi:0` test will fail.  Running other combinations of tests will show that `halcmd` is failing with a segfault.

ArcEye, 2018-12-10:
I am running the whole suite of tests currently on a jessie amd64 partition (ie. not inside a docker)

Finding the tests fail at `remap/remap.io`  with halcmd segfault.

```
/home/mick/src/machinekit/scripts/realtime: line 308: 27998 Segmentation fault      halcmd shutdown
```
Line 308 is the start of the unload() function, so not very helpful.

They had failed on `hm2_idrom` earlier but continued OK.
(I removed `hm2_idrom` from the Stretch tests in the Jenkins job, because it never succeeded inside a docker, but did on the command line.)

ArcEye, 2019-01-17:
This was where I saw you say it was a jessie thing @zultron !!!
Knew I had not made it up (well was pretty sure)

I will close this as covered by the other issues.

zultron, 2019-01-18:
Actually, this is an unrelated problem, could you please reopen?

The issue in #173 is different:  affects Stretch (at least), is nondeterministic, and it fails only in the `threads.0` test, while other tests pass.

This issue, by contrast, affects only Jessie, is deterministic, and hangs during the `rtapi.0` test (following tests never run), although this is a symptom of an earlier failure, and different tests can fail (and `rtapi.0` can pass) when manually running different subsets of tests with different ordering.

I haven't had time to dig into this one yet, but it should just be a matter of isolating a procedure to reproduce the bug (easy, since it's deterministic), and debugging.

zultron, 2019-07-13:
#228 fixes this one.  Turns out to be a problem I introduced all by myself.  ;(  I introduced a new bug caused by `ps -o` differences in Jessie that introduced a new RTAPI exit detection problem (RTAPI hasn't yet exited but `realtime stop` believes it has).  Of course this was when attempting to fix an old bug where RTAPI exit detection problem (RTAPI exits but `realtime stop` believes it still runs).

So, the Jessie-only aspect is explained.  The `rtapi.0` failures reported here, and the `hm2-idrom` failures that got me looking at this problem again, are explained because both tests (and no others?) run `realtime stop; realtime start` in quick succession, exactly what triggers the bug.

The symptom is as @ArcEye describes above:

    [...]/scripts/realtime: line 308: 27998 Segmentation fault      halcmd shutdown

Hopefully the one-liner fix puts this issue to bed for good.  (Famous last words!)

PR #228 (2019-07-13): realtime script:  Fix RTAPI unload detection in Jessie
In Stretch and Buster, output formats may be combined with commas:

    $ ps -p 1 -o pid=,s=
        1 S

On Jessie, this does not work:

    $ ps -p 1 -o pid=,s=
      ,s=
        1

But separating them as separate options does:

    $ ps -p 1 -o pid= -o s=
        1 S

This patch fixes e.g. `rtapi:0` exit detection in the `realtime`
script, a problem introduced by YT in 61446ef4.

Fixes #164.
commit 2e353352 (2019-07-13): realtime script:  Fix RTAPI unload detection in Jessie

In Stretch and Buster, output formats may be combined with commas:

    $ ps -p 1 -o pid=,s=
        1 S

On Jessie, this does not work:

    $ ps -p 1 -o pid=,s=
      ,s=
        1

But separating them as separate options does:

    $ ps -p 1 -o pid= -o s=
        1 S

This patch fixes e.g. `rtapi:0` exit detection in the `realtime`
script, a problem introduced by YT in 61446ef4.

Fixes #164.
closed 2019-07-13 by ArcEye