machinekit/machinekit-hal #165: `machinekit-THREADS` packages can't be coinstalled

zultron, 2018-12-11:
PR machinekit/machinekit#1230 squashed the `machinekit-dev` package
into the main and thread packages in order to solve failing
out-of-tree component builds on `armhf`, a problem arising from
splitting up slow, emulated package builds to complete within the
50-minute Travis CI time limit; see discussion at
machinekit/machinekit#1060.

After this change, the thread packages could no longer be
simultaneously installed on a single system, since files were
duplicated among the packages and caused conflicts.  This hasn't been
a problem for small-scale use cases, with Machinekit installed on a
single system and only requiring a single threads package.  This does
turn out to be a problem when building portable OS images intended to
run in multiple threading environments, and Machinekit's run-time
interchangable thread styles was designed to work for such use cases.

Since the original problem was solved in duplicate by reducing `armhf`
build times (cross-compiled replaced emulated builds) and by moving
off of Travis CI, the split builds have already been recombined.  The
out-of-tree component builds should now work properly, clearing the
way to revive the old `machinekit-dev` package, and restoring the
ability to install multiple thread packages in parallel once again.

ArcEye, 2018-12-11:
I can see the theoretical requirement, wasn't aware anyone was trying to build a generic install image.

The issue has become a little smaller since the contraction of kernel flavours, down to basically rt-preempt only.

The `-dev` package could of course be reinstated.
The original issue was that the dev package was built on a posix env and produced posix modules irrespective of what kind of kernel you were running

The time to do it would be when the component flavour issue is completely resolved.
I have been gradually stripping out all the un-used hard thread stuff from the Makefiles etc. but the big remaining bit is the extremely arcane FOR_ALL_FLAVOURS build.

The components produced are binary identical to those for other kernels on the same architecture.
ie. you can run a rt-preempt kernel with posix components, the only size difference between is the length of the strings `posix` and `rt-preempt`.

Reducing to one build would cut down build times and reduce waste in installation.

If you fancy bringing your gnu make skills into a collaboration, that would be a good project.

zultron, 2018-12-11:
MK still distributes `rt-preempt` and `posix` packages for Stretch, the same packages I'm having issue with.  I'm also hitting the usual snags building out-of-tree components for both flavors; the immediate fix will likely be to simply copy the module from the `posix` to the `rt-preempt` RT lib directory.  ;(

I've wanted to combine the builds for some time now.  My most recent bleatings were in #108.  The idea may be similar to what you're thinking, to fold together the currently per-thread flavor RT components, which as you say are binary identical.  This would solve not only the named issue here, but also fix the above-stated problem, building out-of-tree comps for multiple thread flavors.

On top of this, that issue also proposes to combine the `rtapi_app` programs for all flavors, selecting the thread flavor at run-time from within `rtapi_app`, and removing all the external script machinery.  That would be very nice, but I think it's not necessary to cure the big pain.

I'm going to submit a PR soon reinstating the `-dev` package, since it's an easy target and the work is already done.  However, I'd also like to collaborate on merging the component flavor builds.  I could certainly find time in the next few weeks to work on that.

ArcEye, 2018-12-11:
The problem with changing to single flavour agnostic components, is indeed the rtapi_app_\<flavour\>
and ulapi_\<flavour\> being tied into the same build element.
Just building one set of components and changing the path it is expected to be on is the easy bit.

The idea of doing a runtime resolution of flavour sounds a good one.

> I'd also like to collaborate on merging the component flavor builds. I could certainly find time in the next few weeks to work on that.

Excellent - I will review the changes I made previously for single component builds.  
I almost nailed it - but fell foul of flavour expectations I hadn't noticed elsewhere.

zultron, 2018-12-11:
Do you mean the problem building `rtapi_app_<flavour>` and `ulapi_<flavour>` is in the `Makefile`, or somewhere else?  If the former, it shouldn't be a big deal to fix.

Do you already have a branch somewhere that starts on these changes?  It would sure be nice to have a head start!

Thanks, @ArcEye

ArcEye, 2018-12-11:
> Do you mean the problem building rtapi_app_<flavour> and ulapi_<flavour> is in the Makefile, or somewhere else? If the former, it shouldn't be a big deal to fix.

The same mechanism within the Makefile and Submakefiles, that builds the components per flavour, also builds the relevant rtapi / ulapi binaries AFAIR.

That `FOR_ALL_FLAVOURS` bit is really what needs to be replaced with a linear rule which is more transparent as to what it does.
I simplified it considerably when I took out hard kernel threads, but the obfuscated scripting is far from obvious in its quite considerable scope.

> Do you already have a branch somewhere that starts on these changes? It would sure be nice to have a head start!

I do somewhere, but need to strip out the relevant parts.
Will look at it over next few days

zultron, 2020-05-03:
This was finally solved with the merge of the "single module directory" work.  The separate threads packages now contain very few files, none of which conflict anymore.  Closing this issue.

closed 2020-05-03 by zultron