machinekit/machinekit-hal #179: Put machinekit on life support, continue with machinekit-hal and machinekit-cnc

luminize, 2019-02-27:
following this comment 
https://github.com/machinekit/machinekit/pull/1462#issuecomment-467475471
from @ArcEye I think we should move discussion out of the PR in question.

@cdsteinkuehler @machinekoder @zultron your input would be appreciated.

Proposal:
machinekit/machinekit on "life support" packages from wheeze up to and including buster
machinekit/machinekit no new PR's
machinekit/machinekit-hal and machinekit/machinekit-cnc as alternative and main repo's

(please add additional points if needed)

ArcEye, 2019-02-27:
@cerna
>  I have to say that I also don't see the reasoning behind creating a new set of 2-hal and 2-cnc packages and deprecating the current set of -hal and -cnc. Maybe if @ArcEye illuminated his reasons? 

Already answered @luminize, as no-one uses the `-hal` and `-cnc` split repos I created nearly 2 years ago, the work can go into them.

> (Because I thought that non-versioning is a Machinekit thing, given that version number is still 0.1-something, in contrast to LinuxCNC versions 2.7 or 2.8.)

There is no dogma about versions, it is continous integration which means the git SHA is the version.
In fact `-hal` and `-cnc` are already version `0.2` in the version file, purely for pragmatic reasons, so that any reported problems would more easily be obviously from the split repos, since no-one ever provides sufficient info.

In so far as Gentoo qualifies as a distro, rather than a bleeding edge set of source files, packaging split for the 2 repos will be welcome, even though we do not officially support it.

It is a bellweather in a way for problems in the future.
It will not be too long before gtk2, gnome2, python2, gladevcp2 etc. not only lose support but actually cease to function because packages are no longer available built for the current distros.
This will happen first in the likes of Gentoo and is the reason I run Debian sid on my desktop, to warn me when things start to fall apart, as they have already done to an extent.

ArcEye, 2019-02-27:
What needs to happen now in practical terms is

- Update machinekit-cnc to the current top commit in machinekit

- Update machinekit-hal to the current top commit in machinekit

  ( not completely straight forward as some commits span repos and some require manual
  conflict fixing )

- Test those repos both as machinekit-hal RIP builds and combined machinekit-hal / machinekit-cnc
   package installs.

- Hold all PRs in the interim

It is the third point where assistance will be required.

The packaging needs to be adequate, but there is no point spending too much time on it, because the flavour agnostic modules work brings about another change in packages.

ArcEye, 2019-02-27:
@cerna
> Is there a way how to deprecate Debian package so the user will see the note informing him of the change when doing apt upgrade or apt list --upgradable?

In due course we might need something, albeit nothing I looked at briefly caters for 2 packages which now clash and cannot be installed together, having one recommend the other package, when it itself provides the same files.

For now it will be very convenient to just have 'no updates available' and be able to get on with testing without any general awareness of the existence of the split repos.

ArcEye, 2019-03-01:
All the packaging artifacts, some of which were corrupted by machinekit changes, should be about right for now.

In due course, when the `single_module_directory` work is merged, the packaging will need to change again, but for now it should be complete and fairly solid.

So testing of the `machinekit-hal` packages and testing `machinekit-hal` in combination with `machinekit-cnc` is required on as many different platforms as possible.

Tests need to include a RIP build of `machinekit-hal` and its use to load and run components also.
(There is no RIP build option for machinekit-cnc , unless you follow a convoluted build process used in the package builder, to obtain missing libraries contained in machinekit-hal first)

Any reports welcome.

cerna, 2020-03-06:
The `single_module_directory` work was merged into this repository couple of days ago. And the [main Machinekit](https://github.com/machinekit/machinekit) repository has been officially [Archived](https://help.github.com/en/github/creating-cloning-and-archiving-repositories/archiving-a-github-repository) on GitHUB (made read only).

So I think this issue is solved.

closed 2020-03-06 by cerna