machinekit/machinekit-hal #195: CI Docker images:  Transfer ownership to GH "machinekit" organization?

zultron, 2019-03-15:
The Machinekit Jenkins CI system is currently using Docker images from the [`dovetailautomata/mk-cross-builder` Docker Hub repo][1].  These are autobuilds of the [`Dovetail-Automata/mk-cross-builder` GitHub repo][2].  Those DH and GH organization IDs are mine.  After recent updates, these two repos might be best transferred to the `machinekit` organization.

PR Dovetail-Automata/mk-cross-builder#5 was a big milestone.  It adds support for the new-style automated builds where the `Dockerfile` is parameterized in the Docker Hub build, and it drops the ugly system of templating many `Dockerfile.<tag>` files and checking those into git.  It builds in support for Buster, based on initial work by @ArcEye.  The result looks more mature and more easy to maintain, and the `README` is updated with a view to help others start using it quickly.

I propose we transfer these two repos to the `machinekit` organization.  This isn't a big emergency, since today, after even the worst disaster Machinekit project admins could restore functionality in mere hours.  However it makes good sense to place this important piece of infrastructure in a location more visible to the community, and to manage the GH repo the usual way, ensuring that more eyeballs pass over changes and reducing dependence on any single person.

This should be an easy task, doable in minutes.  A Machinekit GitHub organization admin would need to set up a Docker Hub organization and link it to the project's GitHub organization.  Then I could simply initiate the transfer of the repos in coordination with PRs to the `machinekit`, `machinekit-hal` etc. repos that update the Docker image tags.

[1]: https://hub.docker.com/r/dovetailautomata/mk-cross-builder/tags
[2]: https://github.com/Dovetail-Automata/mk-cross-builder

ArcEye, 2019-03-15:
Seems good sense to me.

You need to be an owner of both repos to transfer across.
If you want do it the easy way, just make me an owner of the repo and I can transfer it (Or join the team😉)

zultron, 2019-06-18:
Alternatively, because the Docker repos are now tightly enough integrated with both the MK source code and CI configuration (and now that they're pretty well cleaned up and proven), perhaps `mk-cross-builder` repo code should be directly merged into the MK-hal repo, and Docker build stages even layered across the MK-cnc repo.

See https://github.com/machinekit/machinekit-hal/issues/221#issuecomment-503265788

ArcEye, 2019-06-20:
It is certainly a good thought for after the `single-module-dir` work is merged and the repo is stable.

(Probably will take some more thought when the second phase, of `linuxcnc-cnc` is done as to what exactly happens with -cnc builds)

Right now might just be adding complexity.

PR #269 (2020-03-21): Eat MK cross builder
This pull request fixes #195 in later decision to directly integrate Debian buildsystem into Machinekit-HAL.

The content consist of the functional base of https://github.com/Dovetail-Automata/mk-cross-builder repository without additional helper tools. Changes originating from my account (@cerna) are minor fixes to make the images build.
closed 2020-03-21 by luminize