machinekit/machinekit-hal #201: Machinekit-hal-xenomai Debian Jessie package has unmet dependencies

cerna, 2019-05-02:
When trying to install Machinekit-HAL on Debian 8 Jessie system (both `AMD64` and `i386`) with Xenomai kernel from Machinekit repository, there are unmet dependencies:

(Mentioned in [discussion](https://groups.google.com/forum/#!topic/machinekit/brFHAoWycC0) and in  machinekit/machinekit#1462 .)
```
The following packages have unmet dependencies:
 machinekit-hal-xenomai : Depends: libboost-python1.62.0 but it is not installable or
                                   libboost-python1.62.0 but it is not installable
                          Depends: libboost-system1.62.0 but it is not installable or
                                   libboost-system1.67.0 but it is not installable
                          Depends: libboost-thread1.62.0 but it is not installable or
                                   libboost-thread1.62.0 but it is not installable
                          Depends: libprotobuf10 but it is not installable or
                                   libprotobuf17 but it is not installable
                          Depends: libstdc++6 (>= 5.2) but 4.9.2-10+deb8u2 is to be installed
                          Depends: libtinfo5 (>= 6) but 5.9+20140913-1+deb8u3 is to be installed
E: Unable to correct problems, you have held broken packages.
```
As the official Debian repository has only libboost*1.55, protobuf9 and libstdc++6  4.9.2 and libtinfo 5.9.

I see 3 possible solutions:

1. Lower the requested versions in a dependency specification
2. Add needed packages to Machinekit Jessie repository
3. Add documentation telling users to compile and install the aforementioned packages themselves

ArcEye, 2019-05-05:
https://github.com/machinekit/machinekit-hal/blame/master/debian/control.xenomai.in

This commit came over from one for machinekit, where the debian structure differs.

Because the 'fix' for Buster was not limited to Buster builds, it breaks Jessie.
This should be in a `control-buster.xenomai.in` file or similar and `configure` altered to use it for Buster builds.

However this leads to another problem.
What should be happening is that `libboost-*-dev` is specified as a `build-dep`, not an install dep and the correct version for that distro will be used.

The packages build, because the dockers have all the required packages, but that is no use if they contain unattainable dependencies, which were not the ones the package was built against.

Needs looking at as a whole.  These deps are common to all packages, the xenomai deps should just be `xenomai-runtime` really.

zultron, 2019-05-06:
Whoops!  Thanks for figuring out my mistake, @ArcEye.

ArcEye, 2019-05-06:
Some of the mistakes before that were mine.  You just built on them :laughing:

ArcEye, 2019-05-06:
I have overhauled the package build for machinekit-hal and used the debhelper macros to insert the correct package dependencies, outside the specified ones.
ie. `${misc:Depends}, ${shlibs:Depends}, $(python:Depends}`

This seems to arrive at the correct versioning for the distro build, libstdc++6 version for Jessie is (>=4.9) for instance.

@cerna 
Are you able to test the install for Jessie?
A full set of Jessie amd64  packages (posix, rt-preempt and xenomai) are here for a couple of days
http://www.mgware.co.uk/temp/

cerna, 2019-05-07:
(Not saying that I am not doing some stupid mistake.)

@ArcEye ,
for the `machinekit-hal-xenomai` and the `machinekit-hal-rt-preempt` there seems to be a dependency conflict. Both packages depends on the `machinekit-hal-posix`, however trying to install `-xanomai` after successful installation of `-posix` throws error:

```
de@Jessie64SATA:~/Desktop/mk$ ls
machinekit-hal-posix_0.2.1557153686.git99a0286-1~jessie_amd64.deb
machinekit-hal-rt-preempt_0.2.1557153686.git99a0286-1~jessie_amd64.deb
machinekit-hal-xenomai_0.2.1557153686.git99a0286-1~jessie_amd64.deb
de@Jessie64SATA:~/Desktop/mk$ sudo dpkg -i machinekit-hal-posix_0.2.1557153686.git99a0286-1~jessie_amd64.deb machinekit-hal-rt-preempt_0.2.1557153686.git99a0286-1~jessie_amd64.deb
(Reading database ... 97011 files and directories currently installed.)
Preparing to unpack machinekit-hal-posix_0.2.1557153686.git99a0286-1~jessie_amd64.deb ...
Unpacking machinekit-hal-posix (0.2.1557153686.git99a0286-1~jessie) ...
Selecting previously unselected package machinekit-hal-rt-preempt.
Preparing to unpack machinekit-hal-rt-preempt_0.2.1557153686.git99a0286-1~jessie_amd64.deb ...
Unpacking machinekit-hal-rt-preempt (0.2.1557153686.git99a0286-1~jessie) ...
dpkg: error processing archive machinekit-hal-rt-preempt_0.2.1557153686.git99a0286-1~jessie_amd64.deb (--install):
 trying to overwrite '/etc/modprobe.d/linuxcnc.conf', which is also in package machinekit-hal-posix 0.2.1557153686.git99a0286-1~jessie
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Setting up machinekit-hal-posix (0.2.1557153686.git99a0286-1~jessie) ...
Valid MKUUID in mkuuid 68061cbc-0084-49fc-89c3-00897ed274e1 - using that
Old value in /etc/linuxcnc/machinekit.ini replaced with new MKUUID 68061cbc-0084-49fc-89c3-00897ed274e1
Errors were encountered while processing:
machinekit-hal-rt-preempt_0.2.1557153686.git99a0286-1~jessie_amd64.deb
```

Otherwise, the `machinekit-hal-posix` installs and runs without problems.

ArcEye, 2019-05-07:
OK, there is something screwy there.

I can see that the machinekit-hal-posix dep has been inserted by the macros, in xenomai and rt-preempt, but unsure why that is.
It is not in the machinekit-hal-posix control file, it should not be in any of them.

Thanks for testing, I will look at it again later and try to find where this is coming from.

ArcEye, 2019-05-08:
@cerna
Think I have sorted it now.
Could not get the macros to work properly, but extracted the package versions etc per distro and use those in the control file, seems to work OK.

A full set of the new Jessie amd64 packages (posix, rt-preempt and xenomai) are here again.
http://www.mgware.co.uk/temp/
if you could do a test install please.

ArcEye, 2019-05-09:
Hopefully this commit will fix the problems you had and prevent others.
https://github.com/machinekit/machinekit-hal/pull/203

cerna, 2019-05-09:
@ArcEye ,

Yes, packages seems to install OK. They are not mutually exclusive (i.e. trying to install machinekit-hal-xenomai over machinekit-hal-posix will throw on `trying to overwrite '/etc/modprobe.d/linuxcnc.conf', which is also in package machinekit-hal-xenomai 0.2.1557333844.git9b9dfc7-1~jessie` and not on `conflicts` property).

Interesting thing is that `OUT OF MEMORY epoll.cpp` error. The same system, the exact same packages and on Xenomai it is present and on RT-PREEMPT it is not.

ArcEye, 2019-05-09:
I can add a `Conflicts:` line for now.

When the work to remove the 'for each flavour' builds comes in, amongst other things we should be able to do away with a lot of the flavour stuff and have a main machinekit-hal package, with a few flavour dependent bits in a `machinekit-hal-<flavour>` package.

> Interesting thing is that OUT OF MEMORY epoll.cpp error. The same system, the exact same packages and on Xenomai it is present and on RT-PREEMPT it is not.

Strange, AFAIR that error seemed likely to be related to czmq, but maybe not.

Thanks for testing

cerna, 2019-05-09:
@ArcEye ,
hasn't [the repository](http://deb.machinekit.io/debian) just been killed by this change? Seems everything was updated today and many packages are missing.

ArcEye, 2019-05-09:
On 09/05/19 16:28, cerna wrote:
    
      @ArcEye ,
        hasn't the
          repository just been killed by this change? Seems
        everything was updated today and many packages are missing.
      —
          it on GitHub, or mute
          the thread.
      [
{
"@context": <a class="moz-txt-link-rfc2396E" href="http://schema.org">"http://schema.org"</a>,
"@type": "EmailMessage",
"@type": "ViewAction",
"target": <a class="moz-txt-link-rfc2396E" href="https://github.com/machinekit/machinekit-hal/issues/201#issuecomment-490951825">"https://github.com/machinekit/machinekit-hal/issues/201#issuecomment-490951825"</a>,
"url": <a class="moz-txt-link-rfc2396E" href="https://github.com/machinekit/machinekit-hal/issues/201#issuecomment-490951825">"https://github.com/machinekit/machinekit-hal/issues/201#issuecomment-490951825"</a>,
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": <a class="moz-txt-link-rfc2396E" href="https://github.com">"https://github.com"</a>
}
}
]
    
    No, I am rebuilding the whole repository, because the definitions
    for Buster have disappeared and the database file is corrupt.
    This became apparant after a mksocfpga commit.
    
    Nothing to do with the machinekit-hal commits
    
    I don't know where you wrote this, it does not appear in the issue.

cerna, 2019-05-09:
I come across this when there were only `machinekit-hal` and `machinekit` packages in Machinekit Debian repository and prematurely reached the conclusion that something went very wrong. After I started chasing reason why my installation cannot download package when it could yesterday, I discovered that packages are slowly popping up in the repository, so I reached the conclusion that something triggered a whole rebuild and retracted my comment as it was unnecessarily alarmist.

> I can add a Conflicts: line for now.

Thank you.

> Strange, AFAIR that error seemed likely to be related to czmq, but maybe not.

It is. I am going to test it against the official [ZeroMQ repository](http://download.opensuse.org/repositories/network:/messaging:/zeromq:/release-stable/).

I don't think it should have anything to do with Xenomai's smaller stack size, as it is not RT and not crashing completely.

cerna, 2019-05-10:
How long does the complete rebuild take? Because it looks like some packages (`xenomai-runtime` for `amd64` for example) were not build yet.

ArcEye, 2019-05-10:
It was manual and it finished yesterday.

AFAICT there was never an amd64 xenomai-runtime package.
Or if there was it was deleted a long time ago.

All the old packages are in http://deb.machinekit.io/archives/ and there is not a single amd64 package for wheezy or jessie in xenomai-runtime.
My own repo which has the xenomai packages copied across from machinekit some years ago does not have amd64 either

ArcEye, 2019-05-10:
https://packagecloud.io/dovetail-automata/machinekit?page=2

The above is where the packages originally came from and there is no amd64 Jessie package of xenomai-runtime there either.

No one used xenomai on anything but armhf AFAIK.

ArcEye, 2019-05-10:
@cerna 
I had 5 minutes, so I built you jessie amd64 package versions of the xenomai-2.6.4 code inside a docker.

I have uploaded them to the same place as previously.

If you confirm they work I will add them to the machinekit repo.

cerna, 2019-05-10:
@ArcEye 
These packages work.
However, I discovered I was being a moron and inserted typo into Debian repository address and that was the reason I could not install that package. (And obviously I thought that it was in the Machinekit repository.) 

> No one used xenomai on anything but armhf AFAIK.

I like the co-kernel idea. Moreover, the [xenomai.c](https://github.com/zultron/machinekit/blob/mk-single-module-dir-zultron-rebase/src/rtapi/flavor/xenomai.c) file doesn't look like it could not be talked into working with modern Xenomai with current kernel. 

So basically starting point.

closed 2019-05-10 by cerna

ArcEye, 2019-05-11:
> @ArcEye
These packages work.

Good, they are now in the main repo