machinekit/machinekit-hal #206: Mkwrapper binary missing in both -cnc and -hal repos

the-snowwhite, 2019-05-15:
Im currently having this issue:
[Mklauncher Problem migrating to MachineKit -hal -cnc packages](https://groups.google.com/forum/#!topic/machinekit/Nqn-L1A1N3M)

I found out it is due to the mkwrapper binary missing due to this PR:
https://github.com/machinekit/machinekit-hal/pull/18

I was able to compile new debs and get my setup to work via this patch:
https://github.com/the-snowwhite/machinekit-hal/tree/mklauncher-work

I have no idea of where the mklauncher binary belongs .. its a gui ?

the-snowwhite, 2019-05-15:
I built with:

    scripts/build_docker -t armhf_9 -c deb

luminize, 2019-05-15:
@the-snowwhite it's a python script
If you make a PR of your patch i'll merge it asap.

the-snowwhite, 2019-05-15:
ok

the-snowwhite, 2019-05-15:
https://github.com/machinekit/machinekit-hal/pull/207

ArcEye, 2019-05-16:
@the-snowwhite 
Well done for finding and fixing.

This was always going to be a problem until proper testing was done.
Obviously no-one has tried using the combined packages to run remotely via Cetus and whatever.
I don't use the python / QtVcp stuff, so I certainly have not.

So long as it is just added to one package that should be fine.
dpkg will refuse to install a package that contains an identical file to a previously installed package in it.

the-snowwhite, 2019-05-16:
ok thanks

closed 2019-05-16 by the-snowwhite