machinekit/machinekit-hal #214: buster yapps2 error

luminize, 2019-06-05:
Just tried to upgrade my RIP build on buster (just feeling my way around) to the latest of machinekit-hal, and I had to install yapps2 package.
When compiling there's the following error:
```
Parsing python hal/utils/comp.g
Traceback (most recent call last):
  File "/usr/bin/yapps2", line 11, in <module>
    load_entry_point('Yapps2==2.2.1', 'console_scripts', 'yapps2')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
ImportError: bad magic number in 'yapps': b'\x03\xf3\r\n'
```

luminize, 2019-06-05:
edit: forgot a bit of terminal output
```
Compiling hal/user_comps/gs2_vfd.c
make: *** [hal/utils/Submakefile:126: objects/hal/utils/comp.py] Error 1
make: *** Waiting for unfinished jobs....
```

ArcEye, 2019-06-05:
I just built machinekit-hal RIP in Buster and get no errors.

Run `dpkg --get-selections | grep yapp`

I get 
```
root@INTEL-i7:/# dpkg --get-selections | grep yapp
python-yapps					install
python3-yapps				install
yapps2						install
```

ArcEye, 2019-06-05:
Based upon this, I would guess you have old compiled python files from the last version in your system still

https://stackoverflow.com/questions/52477683/importerror-bad-magic-number-in-time-b-x03-xf3-r-n-in-django

luminize, 2019-06-06:
ah, `python-yapps` package was missing. Compiling went OK, problem solved

closed 2019-06-06 by luminize