machinekit/machinekit-hal #221: concurrencykit is not checked for during configure

mbuesch, 2019-06-17:
src/rtapi/rtapi_atomics.h and some other places have support for concurrencykit.org:

```
 26 #include "config.h" // HAVE_CK
 31 #ifdef HAVE_CK
```

However there is no configure rule for HAVE_CK.

```
$ git grep HAVE_CK
src/hal/icomp-example/pin.hh:#ifdef HAVE_CK
src/hal/lib/hal_accessor.h:#include "config.h"  // HAVE_CK
src/hal/lib/hal_accessor.h:#ifdef HAVE_CK
src/regressions/check-tests/check_tests.c:#ifdef HAVE_CK
src/regressions/check-tests/check_tests.c:#ifdef HAVE_CK
src/rtapi/rtapi.h:#ifdef HAVE_CK
src/rtapi/rtapi_atomics.h:#include "config.h" // HAVE_CK
src/rtapi/rtapi_atomics.h:#ifdef HAVE_CK
src/rtapi/rtapi_atomics.h:#if !defined(HAVE_CK) || !defined(CK_F_PR_LOAD_64)
src/rtapi/rtapi_atomics.h:#if !defined(HAVE_CK) || !defined(CK_F_PR_STORE_64)
src/rtapi/rtapi_atomics.h:#if !defined(HAVE_CK) || !defined(CK_F_PR_INC_64)
src/rtapi/rtapi_atomics.h:#if !defined(HAVE_CK) || !defined(CK_F_PR_CAS_64)
src/rtapi/rtapi_atomics.h:#if !defined(HAVE_CK)
src/rtapi/rtapi_atomics.h:#endif  // !defined(HAVE_CK)
src/rtapi/rtapi_heap_private.h:#ifdef HAVE_CK
src/rtapi/rtapi_int.h:#ifdef HAVE_CK
src/rtapi/rtapi_msgd.cc:#ifdef HAVE_CK
```

The default without ck will be expensive full memory barriers for all of rmb/wmb.

machinekit-cnc seems to have a similar situation.

ArcEye, 2019-06-17:
Well spotted.

Machinekit is exactly the same.

At some point it was missed out, should be at Section 2.3.1 of configure.ac

I'll look at it later, I have the configure.ac code elsewhere

ArcEye, 2019-06-18:
@zultron 
This was @mhaberler's last hurrah but never included in machinekit before he left.
I obviously missed the activation amongst the many configure.ac versions floating around when I brought in the multicore work.

Will require libck and libck-dev to be added to the recommended packaging deps and thence to the dockers in due course.

I won't do that yet or it will mess up the package builds, we can discuss when time allows.

zultron, 2019-06-18:
Can somebody actually make a PR for the packaging changes?  The workflow starts with [updating MK dependencies in the `mk-cross-builder` repo][1] by directly copying over files from `/debian`.  Those are committed and pushed, which triggers a Docker Hub automated build.

[1]: https://github.com/Dovetail-Automata/mk-cross-builder#updating-the-machinekit-dependencies

ArcEye, 2019-06-18:
Yeah somebody will when they get time. :laughing: 

machinekit and everything created from it has worked perfectly well, if sub-optimally, without CK usage for the past 4 years so don't suppose a few more days will make any difference

ArcEye, 2019-06-20:
> The workflow starts with updating MK dependencies in the mk-cross-builder repo by directly copying over files from /debian. Those are committed and pushed, which triggers a Docker Hub automated build.

You will need to target the PR https://github.com/machinekit/machinekit-hal/pull/222 directly,
as it cannot be merged with machinekit-hal until the dockers support it.

ArcEye, 2019-06-20:
>  machinekit-cnc seems to have a similar situation.

It has the headers from machinekit-hal copied over to it for defines and function prototypes.
I don't think there is any ck usage outside of machinekit-hal.
It certainly builds fine without `HAVE_CK` or `USE_CK` being resolved.

zultron, 2019-06-21:
Thanks for #222, that's what I needed.

It's odd, or maybe not, that `libck` is built for Stretch and Sid but not Buster.  It really IS odd that `libck` is built for Jessie `armhf` and `amd64`, but not `i386`.

I have some images building on Docker Hub.  I'll reproduce #222 in a test PR with an extra commit to temporarily point the build at these test images, and if that goes well, I'll update the production images.

zultron, 2019-06-25:
PR #222 will be ready for a new CI build and review once updated Docker images are done building.  I think my part is done there.  That was an unexpected lot of work!  Thanks @ArcEye for everything you did to help.

PR #222 (2019-06-28): Add Concurrency Kit checks and define setting to configure.ac
Part of the multicore work which is missing for some reason.

Fixes #221
closed 2019-06-28 by ArcEye