machinekit/machinekit-hal #226: Disable CPU idle states at runtime

zultron, 2019-07-03:
I'm tuning latency on a ThinkPad X201 tablet (Intel Core i7-620LM processor).  The base configuration is disabling any BIOS option apparently related to power-saving and HT (overkill for testing), and setting the `isolcpus` kernel CL arg.  This configuration alone gives sub-100k jitter with `latency-test 25000 1000000`.

Adding either of `intel_idle.max_cstate=1` or `idle=poll` gets jitter below 17k.  Adding both doesn't seem to make any significant incremental improvement.

It turns out there's a way to accomplish the equivalent at run-time using the Linux [PM Quality Of Service Interface][2].  I stole [some trivial C code][3] that does this, and get the same sub-17k jitter as `intel_idle.max_cstate=1` and `idle=poll`.

The advantages I see over the kernel command line setting are availability of power management when RTAPI is not running (I'd like this on my dev laptop), and added to `rtapi_app` as an option, easier latency settings for users.

[1]: http://www.machinekit.io/docs/hal/threads-and-latency/
[2]: https://github.com/torvalds/linux/blob/master/Documentation/power/pm_qos_interface.txt
[3]: https://gist.github.com/zultron/639c00091696f079ba34372cc5e8ed2e

ArcEye, 2019-07-04:
The github entry is Oct 2017.
Is there a definative start point for this code in terms of kernel version numbers?

zultron, 2019-07-04:
You mean the `Documentation/power/pm_qos_interface.txt` file?  That's the most recent change, but the file was [moved there][1] in 2008, and IIRC the feature was there in Linux v. 2.6.something.  Not a new feature.  Anyway, I'd implement it as an option that's off by default, unless folks started clamoring for it.

[1]: https://github.com/torvalds/linux/blob/53471121a8aad3f0b590bfce7c95a1f5f52150f3/Documentation/power/pm_qos_interface.txt

ArcEye, 2019-07-04:
Given the date of the github entry I was expecting you to say >= 4.9 or something like that.

It is incredible this has been in since 2.6 and no-one uses or publicises it for latency.

Go for it, if we can get rt-preempt kernels doing consistent low latency with this and cgroups, it kills rtai completely even for parport setups.

zultron, 2019-07-04:
I'm still learning.  This is all related to "idle states" (also called "cstates" in kernel command line args and in some BIOS settings).  [This Linux Kernel user's guide][0] has a lot of clear information about the subject.

The commonly copy-and-pasted `intel_idle.max_cstate=0`, `processor.max_cstate=0` and `idle=poll` kernel command line args all operate on idle states, depending on the CPU idle driver, as shown in `/sys/devices/system/cpu/cpuidle/current_driver`, one of `intel_idle` or `acpi_idle` or maybe `none`.  Judging by Google results, LCNC folks seem to prefer `idle=poll` over `*.max_cstate`.

- `intel_idle.max_cstate=n` limits the idle state to (0..n) for the `intel_idle` driver (see `/sys/module/intel_idle/parameters/max_cstate`)
- `processor.max_cstate=n` limits the idle state to (1..n) for the `acpi_idle` driver (`...=0` limits to state 1)
- `idle=poll` disables both drivers and all energy saving states

The run-time `/dev/cpu_dma_latency` PM QoS interface apparently limits the idle state to (0..n), AFAICT regardless of the current idle driver.  (The kernel user's guide also mentions per-CPU controls in `/sys/devices/system/cpu/cpu${CPU}/cpuidle`.)

[0]: https://www.kernel.org/doc/html/v5.0/admin-guide/pm/cpuidle.html
[1]: https://www.google.com/search?q=site%3Awiki.linuxcnc.org+max_cstate

zultron, 2019-07-04:
Thanks for the encouragement, @ArcEye.  It seems like if somebody wanted to, they could solve latency issues on a lot of systems purely at run-time by using this feature and the `isolcpus` alternative [`partrt`][1].  I'm now curious how many of those BIOS options are actually necessary on the i7.  The J1900 ended up not needing any changes at all.

[1]: http://www.machinekit.io/docs/hal/threads-and-latency/#dedicated-cpuset-using-code-partrt-code-script

cerna, 2019-07-04:
I don't see any "?" in the original question, so I will take it as a general brainstorming/questioning/interrogation session. If this was not the intent, then, well, don't take it personally.

Do as you please, of course, but I don't think that the application should change system-wide settings automagically on it's own. In my humble opinion, the best way about it is to make it optional settings user can choose when starting the `rtapi_app`. Given it is highly flavor specific it should be in the `module_init_hook` function. Make the `module_init_hook` take string as a parameter (that would be generally useful for range of future work initiatives), string taken as a sub-command from `rtapi_app` command line argument (like `git clone --branch=...`) and use that string for the setup.

I have been thinking about little something which could use of reverse of this technique. I don't see currently in HAL a implementation of general use generics real-time capable health monitoring and reporting service which could inform components about general state of the system (like this idea or not using process hardware isolation) and unforeseen situations (from stats like hard RT time overruns and so on specific to flavor) and let the component decide what to do (refuse to load, do emergency procedure and so on). It just idea so far, but it would delegate the responsibility if the system needs to turn off the "idle states" to the RT HAL component.

P.S. I wouldn't worry about what LinuxCNC community is preferring. If the author of this solution had some reason for it, now it is lost to history and it could be just personal reason, nowadays most of them is just repeating it over-and-over again like a dogma.

zultron, 2019-07-05:
@cerna Correct, I opened this issue to discuss this idea for a new feature.

I agree that this feature should be off by default.  As I said, I usually implement features like this as options that must be enabled.  This way the feature is less likely to break things for other users.

I don't agree that this is a flavor-specific feature, though.  It's useful for any RT thread system, RT_PREEMPT, Xenomai and even the inapplicable RTAI.  The only exception is non-RT POSIX threads.

I'll think about your other idea next time I'm digging around in RTAPI.  Sounds interesting.

cdsteinkuehler, 2019-07-07:
C States are evil.

At Day-Job, we ran into issues with C-State transitions with hardware.
 The CPU would shut down for hundreds of milliseconds while the CPU
clock PLL re-locked to a new frequency.  We were bus mastering data to
main system memory which shouldn't be affected by such things, but
with the CPU off-line, even the PCIe hardware paths got stalled
waiting for the cache sniffing logic to come back online.

IMHO, anything we can do to disable C States at runtime is a good
thing, and I'd even say it should be the default (with a disable
available for the random corner case where it makes things worse, like
the battery powered window blind controller or something).  I think
"good latency out of the box" trumps "my computer uses a little more
power when running Machinekit" in terms of "complaints from the user
base".  :)

-- 
Charles Steinkuehler
charles@steinkuehler.net

cerna, 2019-07-10:
@zultron 

>I don't agree that this is a flavor-specific feature, though. It's useful for any RT thread system, RT_PREEMPT, Xenomai and even the inapplicable RTAI. The only exception is non-RT POSIX threads.

Well, OK, if you think so. I am not going to be dense. I have tried it on Xenomai and EVL Core both (run your code in console and default Machinekit's `latency-test` in another) and didn't see any change on my i7-8550u, but then the major killer of low latency is the graphic and I have kernel compiled with recommended settings from Xenomai.

> I'll think about your other idea next time I'm digging around in RTAPI. Sounds interesting.

I am taking notes through my dive into `rtapi` code and will post RFC when I have my idea fully settled and implementation though up. It is partly based on Michael Haberler's idea from #82 and partly on discovering how poor Machinekit is at health security. (You can have long ended thread but it is still showing as an active thread. Partially loaded configuration will happily run and other such non-safe nasty behaviors.) Will need to identify states such as this to make a sense, though. So, it is definitely useful. Post more if you have it.

PR #261 (2020-03-06): Power Management Quality of Service interface
This implements the Power Management Quality of Service interface as described for example on [RedHat site](https://access.redhat.com/articles/65410) and discussed here in machinekit/machinekit-hal#226.

Implemented only for the RT_PREEMPT flavour as it doesn't make sense for the non-realtime `POSIX` flavour. Xenomai and (EVL Core) based systems should heed the help in [official wiki](https://gitlab.denx.de/Xenomai/xenomai/-/wikis/Installing_Xenomai_3) and have power management disabled by default.

fixes #226
closed 2020-03-06 by luminize