machinekit/machinekit-hal #238: 90-xhc.rules deployed multiple times

machinekoder, 2019-10-07:
`/usr/share/linuxcnc/udev/90-xhc.rules` seems to be pulled into the flavor packages, which causes package installation problems:

```
Unpacking machinekit-rt-preempt (0.1.1570435188.gitba33b5a-1~stretch) over (0.1.1567114877.git6c1b9e3-1~stretch) ...                                                                                                                                    
dpkg: error processing archive /var/cache/apt/archives/machinekit-rt-preempt_0.1.1570435188.gitba33b5a-1~stretch_armhf.deb (--unpack):                                                                                                                  
 trying to overwrite '/usr/share/linuxcnc/udev/90-xhc.rules', which is also in package machinekit 0.1.1570435188.gitba33b5a-1~stretch 
```

ArcEye, 2019-10-07:
https://github.com/machinekit/machinekit/blame/master/debian/machinekit-posix.install.in

Suspect @alkabal put the same stuff in the machinekit commit as the machinekit-hal without realising the differences between them.

ArcEye, 2019-10-07:
#1490 #1491 & #1492 refer
Easier to just fix it that talk about it.

closed 2019-10-07 by ArcEye

alkabal, 2019-10-08:
Hi

I’m really really sorry about that. 

Please I apologize.

Br