machinekit/machinekit-hal #245: Docker builds broken

ArcEye, 2019-10-27:
Recent changes have broken docker builds, by activating a build var, probably ${xenomai-kernels},
which has the effect of making configure look for insmod and associated tools (configure.ac L:2405)

Whilst local builds work fine because insmod is a basic part of any running system, the dockers do not contain it and fail.

Commenting out L:2405 - 2424 results in a functioning docker build again

ArcEye, 2019-10-27:
Confusingly the recent Jenkins builds for Buster seem to have completed.

However running scripts/build-debian -t amd64_10 -c deb from within a clone of machinekit-hal on a local machine, will get an error in configure, 'insmod not found'

cerna, 2020-07-01:
Given that the build scripts and `insmod` code with which this issue was originally open no longer exist in Machinekit-HAL codebase, I am going to close this.

closed 2020-07-01 by cerna