machinekit/machinekit-hal #271: Machinekit-HAL and Machinekit-HAL-dev packages cannot be co-installed

cerna, 2020-04-10:
User @cmcquinn discovered a bug that `machinekit-hal` and `machinekit-hal-dev` packages cannot be co-installed on one system. This is a problem, because `machinekit-hal-dev` depends on `machinekit-hal`. More info can be had on [Machinekit forum](https://groups.google.com/forum/#!topic/machinekit/CtfWZp7l1F8).

@zultron, do you have an idea how to solve this?

(This issue is created for objective of logging and tracking this bug.)

cerna, 2020-04-11:
I have been playing with `dh-exec`, respective `dh-exec-install-move` script. But so far without success. (And the `dh-exec-install-move` doesn't seem to be part of 0.23 release currently in Debian - even though it is four years old.)

cerna, 2020-04-11:
I was able to reach functioning setup in commit https://github.com/cerna/machinekit-hal/commit/80df5da274a1c3134127a1173e0e1644f8c11a83. Problem is, it needs `dh-exec` version 0.25, which currently is not in any official Debian repository, so you will have to build the package yourself from [dh-exec](https://github.com/algernon/dh-exec) repository.

This has a problem that `dh-exec` is compat level **12**, so no `jessie` (there is `stretch-backports` with `debhelper` version **12**).

My solution is probably convoluted, but _...oh well_.

cerna, 2020-04-12:
In the #packaging Debian IRC channel, I got the advice to just list all files manually and that I should not under any circumstances put the python script `comp` into `-dev` package. I have no idea why, but it goes against some Debian policy.

Given that I have no deep understanding of the subject, I am going to heed that advice.

So, that leaves the `Makefile.*` and that will be quick fix.

cmcquinn, 2020-04-19:
@cerna, the packages are now co-installable, but interestingly the install is not working on multistrap. It works fine on a native install (see [here](https://github.com/cmcquinn/machinekit-cnc/runs/598901524?check_suite_focus=true)), but still seems to get ignored in a multistrap config (see [here](https://github.com/cmcquinn/machinekit-cnc/runs/598901546?check_suite_focus=true)). The packages are present in the list of packages to be installed (see [here](https://github.com/cmcquinn/machinekit-cnc/runs/598901546?check_suite_focus=true#step:4:433)), but they don't ever seem to get extracted.

Any thoughts?

Also, I hope I'm putting this in the right place. I figured I would move our discussion from the forum here to GitHub.

cerna, 2020-04-19:
@cmcquinn,

>Also, I hope I'm putting this in the right place. I figured I would move our discussion from the forum here to GitHub.

Sure, on GitHub - at least - it will be delivered only to people interested in development. So, it's fine.

>Any thoughts?

Well, I have been tinkering with it today and one thing which seems odd to me is that if the `cmcquinn/mk-hal-cross-builder` image in Docker HUB registry is created from the Machinekit-HAL _git_ repository, then Dockerfile in `machinekit-cnc` is trying to create `multistrap` in the same exact folder as already exists from the `machinekit-hal` one. (Frankly, I have no idea if you can call `multistrap` twice on same folder.)

Also, I assume that you know this, but the `Debian packaging` scripts in `machinekit-cnc` need some love (rewrite) and currently are not working.

cerna, 2020-04-19:
OK, I was able to create `i386_9` image with 
```
I have no name!@07934d1666a7:/# apt list --installed | grep machinekit-hal

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

machinekit-hal/now 0.3.1586676891.git3e033ff-1~stretch i386 [installed,local]
machinekit-hal-dev/now 0.3.1586676891.git3e033ff-1~stretch i386 [installed,local]
machinekit-hal-posix/now 0.3.1586676891.git3e033ff-1~stretch i386 [installed,local]
machinekit-hal-rt-preempt/now 0.3.1586676891.git3e033ff-1~stretch i386 [installed,local]
```
installed. (Locally.)

(And I am going to reopen this issue.)

cmcquinn, 2020-04-22:
How did you manage to get the machinekit-hal packages installed?

I have reworked the `cmcquinn/mk-cnc-cross-builder` Dockerfile so that it does not depend on `cmcquinn/mk-hal-cross-builder`. This way, `multistrap` will create a clean system in the `mk-cnc-cross-builder` container.

Also, I tried adding `machinekit-hal` and `machinekit-hal-dev` to the `Build-Depends` section in [debian/control.in](https://github.com/cmcquinn/machinekit-cnc/blob/package-ci/debian/control.in). So far, it works on the `amd64` arch, as the Docker container build failed unless I included the path to the machinekit-hal packages in the `apt-get` command line (see [job output](https://github.com/cmcquinn/machinekit-cnc/runs/610206657?check_suite_focus=true#step:4:2542) and [fix](https://github.com/cmcquinn/machinekit-cnc/commit/d4d5d7634357b5886d4bdda95e29006ba9c739cb)). Running in the `cmcquinn/mk-cnc-cross-builder:amd64_10` Docker container:

```
root@c650ebecb339:/# apt list --installed | grep machinekit

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

machinekit-cnc-build-deps/now 1.0 amd64 [installed,local]
machinekit-hal-dbgsym/now 0.3.1587250689.gitf462cc9-1~buster amd64 [installed,local]
machinekit-hal-dev/now 0.3.1587250689.gitf462cc9-1~buster amd64 [installed,local]
machinekit-hal-posix/now 0.3.1587250689.gitf462cc9-1~buster amd64 [installed,local]
machinekit-hal-rt-preempt/now 0.3.1587250689.gitf462cc9-1~buster amd64 [installed,local]
machinekit-hal/now 0.3.1587250689.gitf462cc9-1~buster amd64 [installed,local]
```

Still no joy getting the packages to install in non-`amd64` configurations.

cerna, 2020-04-23:
@cmcquinn,
>How did you manage to get the machinekit-hal packages installed?

I was able to come up with two way:

- Delete the `/sysroot` folder with everything in it during build before `multistrap` commands in `machinekit-cnc-debian-builder`. That way it will create the new filesystem from grounds-up without old one cluttering around. The negative effect for this is that the onion layers add up. The end result has about 3.5 GB (because it has both filesystem internally).
- Use `bootstrap` step in `machinekit-hal-debian-builder` only for basic Debian filesystem, add every repository as a `aptsource`, then `chroot` into it and configure it with `/var/lib/dpkg/info/dash.preinst install` and `dpkg --configure -a` as described in [help](https://wiki.debian.org/Multistrap). Install everything by just `apt-get install -y program_package`. This works and I originally thought that it would be the best solution, but I am seeing some problems with command `apt-get update` - it is possible that I just didn't set some settings which needs to be set. I - so far - don't know.

>I have reworked the cmcquinn/mk-cnc-cross-builder Dockerfile so that it does not depend on cmcquinn/mk-hal-cross-builder. This way, multistrap will create a clean system in the mk-cnc-cross-builder container.

OK, if you think it is better that way.

> Also, I tried adding machinekit-hal and machinekit-hal-dev to the Build-Depends section in debian/control.in. So far, it works on the amd64 arch, as the Docker container build failed unless I [...]

You will probably need to create local repository for `bootstrap` step to catch on, something like this part is doing:
```
# Generage Packages file for local repo
RUN ( cd /tmp/debs && dpkg-scanpackages -m . > /tmp/debs/Packages )
```

The `apt-get` command (again, probably) will not work without configuring the filesystem first, I think. And you would need to run it `chrooted` into `/sysroot` folder.

zultron, 2020-04-30:
> In the #packaging Debian IRC channel, I got the advice to just list all files manually and that I should not under any circumstances put the python script `comp` into `-dev` package. I have no idea [...]

I think the `comp` and `runtests` scripts belong in the `mk-dev` package.  There's plenty of precedent for this.  The following command shows all executables installed by `*-dev` packages into `/usr/bin` in our Docker image, of which `gdbus-codegen`, `gtester-report`, `gtk-builder-convert`, `pyste` are all python scripts.

```
$ for i in /usr/bin/*; do dpkg-query -S $i 2>&1; done | grep .-dev:
[...cherry-picking examples...]
libfreetype6-dev: /usr/bin/freetype-config
libglib2.0-dev: /usr/bin/gdbus-codegen
libgdk-pixbuf2.0-dev: /usr/bin/gdk-pixbuf-csource
libgdk-pixbuf2.0-dev: /usr/bin/gdk-pixbuf-pixdata
libgdk-pixbuf2.0-dev: /usr/bin/gdk-pixbuf-query-loaders
libglib2.0-dev: /usr/bin/glib-genmarshal
libglib2.0-dev: /usr/bin/glib-gettextize
libglib2.0-dev: /usr/bin/glib-mkenums
libglib2.0-dev: /usr/bin/gobject-query
libglib2.0-dev: /usr/bin/gtester
libglib2.0-dev: /usr/bin/gtester-report
libgtk2.0-dev: /usr/bin/gtk-builder-convert
libicu-dev: /usr/bin/icu-config
libpng-dev:amd64: /usr/bin/libpng-config
libpng-dev:amd64: /usr/bin/libpng16-config
libncurses5-dev:amd64: /usr/bin/ncurses5-config
[...etc....]
```

cerna, 2020-04-30:
>I think the `comp` and `runtests` scripts belong in the `mk-dev` package. There's plenty of precedent for this.

It seems logical. I also think that tools used for development - even executable ones - should be in `-dev` package. Otherwise, the package should be named `-headers`. And more so given that there is practically no chance for any Machinekit project to be accepted into Debian proper repository, I think strictly adhering to Debian policies is not necessary. Hence, the advice is moot point.

zultron, 2020-05-11:
I believe this issue should be resolved since #282.

PR #272 (2020-04-12): Enable co-installation of machinekit-hal and machinekit-hal-dev packages
This pull request fixes #271 by putting the `comp` executable into `machinekit-hal` package instead of in `machinekit-hal-dev` package.

By this you can install `machinekit-hal-dev` on top of `machinekit-hal` without errors.
closed 2020-05-11 by zultron