machinekit/machinekit-hal #292: Dockerfile:  Compound commands separated by semicolons may hide failures

zultron, 2020-08-11:
Commands in `Dockerfile` separated by semicolons can fail unnoticed.  Here's the example I saw:

```
RUN apt-get update;        \
    apt-get install -y     \
        sudo               \
        machinekit-fixuid; \
    apt-get clean;
```

The `machinekit-fixuid` package isn't installed, but the following `apt-get clean` command covers up the failure, so the build continues:

```
---> 13df6bac2922
Step 18/40 : RUN apt-get update;            apt-get install -y             sudo                       machinekit-fixuid;     apt-get clean;
 ---> Running in c957d92ea676
[...]
E: Unable to locate package machinekit-fixuid
Removing intermediate container c957d92ea676
 ---> 578c02271a74
Step 19/40 : COPY scripts/buildsystem/debian/base-entrypoint.sh /opt/bin/base-entrypoint.sh
```

The old Dockerfile separated commands with `&&`; that way, if any intermediate command fails, the whole step fails.

This resulting failure is seen later:
```

Run docker run --tty --rm -u "$(id -u):$(id -g)" \
Machinekit-HAL entrypoint: Changing burnt-in user machinekit UID and/or GID to ones passed by --user run option.
All files under / belonging to old user ID:1000 or group ID:1000 will be updated to new values of user ID:1001 and group ID:116.
/opt/bin/base-entrypoint.sh: line 65: machinekit-fixuid: command not found
```

cerna, 2020-08-11:
@zultron,
I think I implemented it based on moby/moby#1838 or one of similar Docker multiline issues/proposals (it's quite prevalent in Docker) purely because I find this syntax cleaner and more eye-pleasing.

**Of course, I had no idea it could fail this badly.**

If I were to guess, I would say that the fundamental shell running in container is not set up to fail on first encountered error (`set -e`, I think), which causes the hiding of error and the output error code does not signal problem.

So, bottom line: Sorry, didn't know, now I will (until I will forget). The miss of `machinekit-fixuid` was due networking error or is it for some other distribution (and I should put it in Machinekit/Machinekit) repository?

zultron, 2020-08-11:
> I find this syntax cleaner and more eye-pleasing.

I do agree, and had to get over my peeve; the other thing I tried was putting each statement in a separate `RUN`, but that turns out to create too many layers, and doesn't work for `if foo; then bar; baz; fi`.

I have a commit fixing this, so don't worry about it.

> The miss of `machinekit-fixuid` was due networking error or is it for some other distribution (and I should put it in Machinekit/Machinekit) repository?

I'm working on supporting Focal for ROS Noetic.  It would be great if you could get `machinekit-fixuid` into the MK repo, but no rush, since I'm also waiting for @rene-dev's py3 port.  Thanks!

cerna, 2020-08-12:
@zultron,
>[...]It would be great if you could get machinekit-fixuid into the MK repo[...]

I have put the `machinekit-fixuid` packages for Ubuntu 20.04 Focal into [Machinekit/Machinekit](https://cloudsmith.io/~machinekit/repos/machinekit/packages/) repository. It looks like Ubuntu killed the **i386 (i686)** architecture, so packages are available for **amd64**, **armhf** and **arm64**. I did a test based on your branch with dropping of zultron/machinekit-hal@67e74548c8475ef19b67445e438f869ade42614c and it looks to be [working](https://github.com/cerna/machinekit-hal/runs/975995099?check_suite_focus=true#step:9:14).

I prepared the Ubuntu (or other Debian forks) support as I wanted packages/testing for Ubuntu 18.04 Bionic (some [platforms](https://forum.linuxcnc.org/38-general-linuxcnc-questions/36258-jetson-nano-as-linuxcnc-driver) distribute kernel for it), but then I was postponing updating the Dockerfile, because Ubuntu  is doing some unnecessary _hanky panky_ with `ports.ubuntu.com` address. (I would have to learn sed/awk/perl syntax and that would be a night or two.) So, everything should push itself to [Machinekit/Machinekit-HAL](https://cloudsmith.io/~machinekit/repos/machinekit-hal/packages/) repository. Hopefully.

zultron, 2020-08-12:
> I have put the `machinekit-fixuid` packages for Ubuntu 20.04 Focal into [Machinekit/Machinekit](https://cloudsmith.io/~machinekit/repos/machinekit/packages/) repository.

Getting a bit OT for this issue, but I'm seeing the `machinekit-fixuid` packages over in the separate [Machinekit/Machinekit package repo][1], perfect.

[1] https://cloudsmith.io/~machinekit/repos/machinekit/

Thank you, and wow, thanks a whole lot extra for testing!

PR #293 (2020-08-12): Dockerfile:  Change `;` to `&&`
Commands in `Dockerfile` separated by semicolons can fail unnoticed.
Here's the example I saw:

```
RUN apt-get update;        \
    apt-get install -y     \
        sudo               \
        machinekit-fixuid; \
    apt-get clean;
```

The `machinekit-fixuid` package isn't installed, but the following
`apt-get clean` command covers up the failure, so the build continues:

```
---> 13df6bac2922
Step 18/40 : RUN apt-get update;            apt-get install -y             sudo                       machinekit-fixuid;     apt-get clean;
 ---> Running in c957d92ea6 [... cut]
commit babf861e (2020-08-12): Dockerfile:  Change `;` to `&&`

Commands in `Dockerfile` separated by semicolons can fail unnoticed.
Here's the example I saw:

```
RUN apt-get update;        \
    apt-get install -y     \
        sudo               \
        machinekit-fixuid; \
    apt-get clean;
```

The `machinekit-fixuid` package isn't installed, but the following
`apt-get clean` command covers up the failure, so the build continues:

```
---> 13df6bac2922
Step 18/40 : RUN apt-get update;            apt-get install -y             sudo                       machinekit-fixuid;     apt-get clean;
 ---> Running in c957d92ea [... cut]
closed 2020-08-12 by cerna