machinekit/machinekit-hal #301: Checking return value of function call in C code

cerna, 2020-08-25:
When programming in C language, as a general rule, if some function returns a value, you better check it. Given current patterns, this value in more cases than not represents error state (or has the ability to represent an error state). And one better structure the program so that this error checking makes sense when looking at the whole call stack.

Problem is, in Machinekit-HAL, there are numerous parts of the code which **do not** do this. In code static analysis that means bazillion of errors and warnings.

And on more functional side - given that the _Debian-like_ Docker builder images do export the `-Werror` flag for any (GCC) compiler with major version lesser than 9 - the building of packages (and running tests on) Ubuntu Bionic 18.04 LTS needs additional sanitizing of 4 functions. (I don't know which combination of flags causes that [low] number, there are tons of other instances.)

In many cases it is the checking of return value of `snprintf()` to quiet the truncation warnings.

_(This issue is not about discussion if it needs to be done - because it does. [Of course, I am not going to prevent discussion.] But should serve as progress tracker - it's not going to happen in one go.)_

zultron, 2020-08-25:
+1

I'd like to see commands to reproduce this.

cerna, 2020-08-25:
>I'd like to see commands to reproduce this.

Commands to reproduce _what_ exactly? The Ubuntu Bionic build? That is in [my branch](https://github.com/cerna/machinekit-hal/tree/pr-299-rebase). I am going to open the _pull request_ once the Python 3 work is merged in and I have to solve the _no good, horrible, terrible_ ports stuff first (so think up some script which will process the `/etc/apt/sources.list` the right way). But that's off-topic.

You can see the output of static analysis [here](https://scan.coverity.com/projects/machinekit-machinekit-hal) - it's little stale, but good enough.

rene-dev, 2020-10-03:
linuxcnc recently fixed a number of gcc and static analyzer warnings.