machinekit/machinekit-hal #321: Installing machinekit-hal-rt-preempt

jallwine, 2020-10-01:
I'm unable to install machinekit-hal-rt-preempt as it says it depends on "linux-image-rt-armmp but it is not installable". How do I install linux-image-rt-armmp?

cerna, 2020-10-01:
@jallwine,
it should be installable from [this](https://packages.debian.org/stretch-backports/linux-image-rt-armmp) page, but I haven't really tested it.

Fact is the whole `machinekit-hal-rt-preempt` package is meta-package to install the needed kernel packages and is not actually needed (yet, it would actually be nice if the flavours had own packaging) to run the `rt-preempt` flavour of Machinekit.

cerna, 2020-10-01:
@jallwine,
BTW, if you have a better idea or solution, that would be great!

jallwine, 2020-10-01:
I'd be happy to if I had one. Right now I'm just trying to get up and going! I appreciate you walking me through it. I'm guessing that my problems stem from having the wrong kernel/debian version or something like that. I don't have a good understanding of the RT kernel stuff. When I was building stuff months ago, I basically was shooting in the dark trying different kernels and found that Debian Stretch with kernel 4.14.108-ti-rt-r134 seemed to work the best on the Beaglebone AI (in terms of overall system stability and it actually booting). When I have that combination installed and try to install machinekit-hal-rt-preempt, I get an error saying linux-image-rt-armmp is not installable. 

I've tried using Debian Buster on the Beaglebone AI, but haven't had much success getting a real time kernel going. For example, when I install the latest LXQT image listed [here](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_10_.28Buster.29_-_Weekly) it comes with kernel v4.4.19.94-ti-r43. After running the scripts to install the machinekit and emcapplication repositories, then installing machinekit-hal-rt-preempt it does install the linux-image-rt-armmp package and switch my kernel to v4.19.0-11-rt-armmp, but then I can't boot.  

If I can avoid installing machinekit-hal-rt-preempt and install a realtime kernel myself that works (still need to find one), you're saying it should work with the machinekit-hal package only?

cerna, 2020-10-01:
>If I can avoid installing machinekit-hal-rt-preempt and install a realtime kernel myself that works (still need to find one), you're saying it should work with the machinekit-hal package only?

If we are talking in terms of `rt_preempt` (which we generally are, but just for versatility), the HAL looks for [this](https://github.com/machinekit/machinekit-hal/blob/88a5fd3df88329a97111ce1e2afb73789b3a6246/src/rtapi/flavor/rt-preempt.c#L54) and based on that register callbacks for either `rt-preempt` or `vanilla posix` flavours.

>After running the scripts to install the machinekit and emcapplication repositories, then installing machinekit-hal-rt-preempt it does install the linux-image-rt-armmp package and switch my kernel to v4.19.0-11-rt-armmp, but then I can't boot

Probably that Debian kernel is not really a good fit for Beaglebone AI. If you are able to install a RT kernel from other sources, than that's all is needed.

jallwine, 2020-10-01:
Correct, I don't need the desktop on the Beaglebone.

jallwine, 2020-10-01:
Are you wondering why I went with the LXQT image? I imagine some people will want the desktop. I also wanted to try to run the latency test and see what it shows and those seem to be GUI programs.

closed 2020-10-02 by jallwine

cerna, 2020-10-02:
>Are you wondering why I went with the LXQT image? I imagine some people will want the desktop. I also wanted to try to run the latency test and see what it shows and those seem to be GUI programs.

Yes. Optimally this package (Machinekit-HAL) would be terminal only application, but the **HALscope** is the real stopper in this situation. It will need support for runtime API connection (to RTAPI, kinda naming collision).

But I have Python script for TUI latency test from back when I was still trying to resist the Python language. So I will try to rewrite it and post it into codebase.