machinekit/machinekit-hal #337: Increase rate limit error:

the-snowwhite, 2020-11-19:
This message now shows up upon building
> Test RIP version on Debian 10 arm64: Error response from daemon: toomanyrequests: 
You have reached your pull rate limit. You may increase the limit by authenticating and upgrading: https://www.docker.com/increase-rate-limit
Test RIP version on Debian 10 arm64 – Prepare Docker Drone volume: Error

[Since this limit does not apply to teams.](https://github.blog/2020-04-14-github-is-now-free-for-teams/)
I wonder if Machinekit is sucessfully registered as a (free)Team ?

the-snowwhite, 2020-11-19:
https://github.com/machinekit/machinekit-hal/pull/336

cerna, 2020-11-20:
@the-snowwhite,
> Edit: Seems like Github is systematicially decreasing the container resources for free github users.

I don't think this is a GitHub issue, it looks like a Docker issue (_Docker Inc., the company_, not Docker, the technology). They have recently announced [Docker Hub pull rate limiting](https://www.docker.com/blog/what-you-need-to-know-about-upcoming-docker-hub-rate-limiting/). Given how many tutorials and so on Internet recommends using Docker Hub, I think it is **going to be fun**.

There is possibility of [Open-Source account on Docker Hub](https://www.docker.com/community/open-source/application), however I am not sure if it is applicable. If the account is needed for OS organizations hosting their images on Docker Hub or if it is required for pulling any images from Docker Hub. If it was the first option, then Debian or Ubuntu would need to create this account. (Machinekit is not hosting anything on Docker Hub, only in Github Container registry and will be on [QUAY](https://quay.io/organization/machinekit).)

>I wonder if Machinekit is sucessfully registered as a (free)Team ?

I don't think you need to do anything. Just create a hidden repository.

closed 2020-11-27 by the-snowwhite