machinekit/machinekit-hal #338: halcmd seg faulting after disconnecting from ssh

jallwine, 2020-11-30:
I'm running into a strange problem where the halcmd command starts seg faulting after I disconnect from ssh (the python linuxcnc interface also similarly starts failing). The realtime environment and all HAL processes seem to continue to run, but HAL is in a totally broken state. The HAL environment is set up with systemd, so none of the processes are associated with the ssh terminal that is being disconnected. All ssh terminals must be disconnected for this to occur (as opposed to having multiple ssh connections and closing one of them). If I don't ssh to the machine at all, the environment works fine as well (demonstrated via a python process that I can communicate with via WebSockets).

To help eliminate some culprits, I created a dummy service that simply starts the realtime environment and then sleeps:

    realtime start
    sleep infinity

It demonstrates the same behavior with this simple service. When I ssh to the machine and then exit any interactions with the realtime environment fail (halcmd seg faults). I don't see anything helpful in various logs. I've seen this behavior on the most recent version of machinekit-hal as well as older versions as far back as MachineKit before the split HAL/CNC split. I've also seen this behavior on the Beaglebone Black and the Beaglebone AI. 

Any help on where to look for more information or thoughts on what could be causing this would be appreciated!

jallwine, 2020-11-30:
I witnessed this behavior on Debian Stretch using an rt-preempt kernel. I've tested this on a much older version that used Debian Wheezy and the Xenomai kernel and did not see this behavior.

cerna, 2020-11-30:
@jallwine,
so if I understand it straight:

You start it by starting `systemd` service and command `realtime start`, then sometime after that you connect with SSH to that machine and do what, exactly? You execute some command like `halcmd newinst flipflopv2 switch` or you start the long-term, interactive version `halcmd -kf` which reads from `stdin`?

Because `halcmd` is odd program, it communicates with `rtapi_app` by ZeroMQ socket, but also directly connect to the HAL shared memory (this is why it is used for `fork&exec` pattern when loading the non-realtime unmanaged modules). At the start it loads the HAL memory and at the end there is call to `halcmd_shutdown()` and it unloads it.

I would investigate this if you are terminating the SSH connection when the `halcmd` command is still running.

If what I described is not your scenario, next thing I can think of is - under which users are the `realtime` and `halcmd` started? If there isn't some problem with accessing the shared memory element and OS trying to delete it after the user logs off.

jallwine, 2020-12-01:
The first ssh connection doesn't have to do anything related to HAL. If the ssh connection ends, that somehow puts HAL into a broken state. So this is the order of events:

1) At boot, systemd starts a bash script that simply runs `realtime start` and then sleeps forever. The user that it starts it as is the same user that connects via ssh in the next step (in my case, the pocketnc user).
2) I manually ssh to the machine with the pocktenc user.
3) At any point after that, I close the ssh connection. I could close it immediately or ssh in to multiple terminals separately and then close them individually, but when the last terminal is closed is when things break.
4) At this point HAL is in a broken state. This is a bit of a deduction with the bare bones `realtime start` service. When I start up my full HAL environment I have a python process that interfaces with HAL and I'm usually able to interact with HAL via WebSockets. After closing the ssh session in step 3, the python interface to HAL stops working (but I can still communicate via WebSockets to the python process).
5) To verify HAL is broken, I can ssh back in with the pocketnc user and run `halcmd show pin` (or any other halcmd) and it seg faults. At the very least, this seems like a problem that should be addressed by handling the problem more gracefully, but I'm not sure how to debug what is going on.
6) Without a reboot, I can stop the systemd service and start it again via an ssh terminal to put HAL in a good state and when I close the ssh connection, the same problem happens.

cerna, 2020-12-01:
Curious, very curious.

>At this point HAL is in a broken state. This is a bit of a deduction with the bare bones realtime start service. This is a bit of a deduction with the bare bones `realtime start` service.

Could you look if the shared memory segment still exists after you close all shells? You should see after running `ls -la /dev/shm` something like:
```
mars@mars:~/Documents$ ls -la /dev/shm
total 1304
drwxrwxrwt  2 root     root        100 Dec  1 21:36 .
drwxr-xr-x 19 root     root       4340 Dec  1 12:41 ..
-rw-rw----  1 mars     mars     806912 Dec  1 21:36 hal-0-00154711
-rw-rw----  1 mars     mars     524288 Dec  1 21:36 hal-0-00414c32
-rw-r-----  1 www-data www-data   4096 Dec  1 12:41 mono.1107
```
When Machinekit-HAL is running and no `hal*` related `tmpfs` entries when it is down. Maybe run it as another service with logging to file every 10 seconds.

And maybe in the same step looking for the connected processes with `ps -aux | egrep 'rtapi|msgd'`:
```
mars@mars:~/Documents$ ps -aux | egrep 'rtapi|msgd'
mars     21309  0.0  0.0 103340  7744 ?        SLsl 21:36   0:00 msgd:0
root     21314  0.0  0.0 138860  7856 ?        Ssl  21:36   0:00 rtapi:0
mars     21787  0.0  0.0   6212   884 pts/2    S+   21:52   0:00 grep -E rtapi|msgd
```

My thinking is going in the way that maybe (just maybe) with closing the final terminal, the OS cleans up the remaining resources.

>To verify HAL is broken, I can ssh back in with the pocketnc user and run `halcmd show pin` (or any other halcmd) and it seg faults.

By _seg faulting_ you mean something different from: (Right?)
```
mars@mars:~/Documents$ halcmd show pin
halcmd: cant connect to rtapi_app: -1 (uri= uuid=39b7d24d-a7a3-486e-947e-be1db9a3b8a7): rtapi_rpc(): reply timeout

halcmd: the rtapi:0 RT demon is not running - please investigate /var/log/hal.log
halcmd: the msgd:0 logger demon is not running - please investigate /var/log/hal.log
E: 20-12-01 21:55:14 [21927]dangling 'DEALER' socket created at hal/utils/halcmd_rtapiapp.cc:284
```

---

Another option to try is the `shmdrv` kernel driver which creates the shared memory segment directly in kernel space, but I am not sure it is currently in working order. (It would probably need some patching.)

jallwine, 2020-12-01:
Looks like your suspicions were correct about the shared memory getting cleaned up.

Before broken state:

    pocketnc@pocketnc:~$ ls -la /dev/shm
    total 3912
    drwxrwxrwt  2 root     root         100 Dec  1 21:06 .
    drwxr-xr-x 15 root     root        3340 Dec  1 16:15 ..
    -rw-rw----  1 pocketnc pocketnc 2671856 Dec  1 21:06 hal-0-00000064
    -rw-rw----  1 pocketnc pocketnc  806912 Dec  1 21:06 hal-0-00154711
    -rw-rw----  1 pocketnc pocketnc  524288 Dec  1 21:06 hal-0-00414c32
    pocketnc@pocketnc:~$ ps -aux | egrep 'rtapi|msgd'
    pocketnc  6328  0.6  0.9  29300  6224 ?        SLsl 21:06   0:00 msgd:0
    root      6333 44.3 10.4  68924 65364 ?        SLsl 21:06   0:56 rtapi:0
    pocketnc  6414  0.0  0.2   3824  1656 pts/0    S+   21:08   0:00 grep -E rtapi|msgd
    pocketnc@pocketnc:~$ exit

After closing ssh connection and reconnecting:

    pocketnc@pocketnc:~$ ls -la /dev/shm
    total 0
    drwxrwxrwt  2 root root   40 Dec  1 21:08 .
    drwxr-xr-x 15 root root 3340 Dec  1 16:15 ..
    pocketnc@pocketnc:~$ ps -aux | egrep 'rtapi|msgd'
    pocketnc  6328  0.8  1.0  29300  6280 ?        SLsl 21:06   0:01 msgd:0
    root      6333 44.9 10.4  68924 65364 ?        SLsl 21:06   1:08 rtapi:0
    pocketnc  6463  0.0  0.2   3824  1568 pts/0    S+   21:09   0:00 grep -E rtapi|msgd

    
When I run halcmd after the shared memory segments disappear:

    pocketnc@pocketnc:~$ halcmd show pin
    Segmentation fault

Any thoughts on how I might prevent the OS from cleaning up the shared memory?

jallwine, 2020-12-01:
Not sure if this tells you anything, but the `ipcs -a` command gives some extra info.

Before broken state:

    pocketnc@pocketnc:~$ ipcs -a
    
    ------ Message Queues --------
    key        msqid      owner      perms      used-bytes   messages    
    
    ------ Shared Memory Segments --------
    key        shmid      owner      perms      bytes      nattch     status      
    0x000003e9 163840     pocketnc   700        8192       4                       
    0x000003ea 196609     pocketnc   700        170000     5                       
    0x000003eb 229378     pocketnc   700        8192       5                       
    0x000003ec 262147     pocketnc   700        2048       3                       
    0x000003ed 294916     pocketnc   700        131072     3                       
    
    ------ Semaphore Arrays --------
    key        semid      owner      perms      nsems     
    0x000003e9 163840     pocketnc   700        1         
    0x000003ea 196609     pocketnc   700        1         
    0x000003eb 229378     pocketnc   700        1         
    0x000003ec 262147     pocketnc   700        1         
    0x000003ed 294916     pocketnc   700        1         

After ssh connection closes and reconnecting:

    pocketnc@pocketnc:~$ ipcs -a
    
    ------ Message Queues --------
    key        msqid      owner      perms      used-bytes   messages    
    
    ------ Shared Memory Segments --------
    key        shmid      owner      perms      bytes      nattch     status      
    0x000003e9 163840     pocketnc   700        8192       3                       
    0x000003ea 196609     pocketnc   700        170000     6                       
    0x000003eb 229378     pocketnc   700        8192       5                       
    0x000003ec 262147     pocketnc   700        2048       3                       
    0x000003ed 294916     pocketnc   700        131072     3                       
    
    ------ Semaphore Arrays --------
    key        semid      owner      perms      nsems

jallwine, 2020-12-01:
This seems like a likely culprit: https://superuser.com/questions/1117764/why-are-the-contents-of-dev-shm-is-being-removed-automatically

jallwine, 2020-12-01:
That seems to be it. I added uncommented `RemoveIPC=yes` in /etc/systemd/logind.conf and changed it to `RemoveIPC=no` and all seems to work.

jallwine, 2020-12-01:
Thanks for you help tracking that down, @cerna!

closed 2020-12-01 by jallwine