machinekit/machinekit-hal #339: problems building machinekit-hal for Debian Stretch

jallwine, 2020-12-08:
Per the documentation, I run:

    git clone https://github.com/machinekit/machinekit-hal.git
    cd machinekit-hal
    ./scripts/buildcontainerimage.py Debian Stretch armhf

Which results in:

    E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/p/python3.5/libpython3.5-minimal_3.5.3-1+deb9u2_armhf.deb  404  Not Found [IP: 151.101.64.204 80]
    E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/p/python3.5/libpython3.5-stdlib_3.5.3-1+deb9u2_armhf.deb  404  Not Found [IP: 151.101.64.204 80]
    E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/p/python3.5/libpython3.5_3.5.3-1+deb9u2_armhf.deb  404  Not Found [IP: 151.101.64.204 80]
    E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/p/python3.5/libpython3.5-dev_3.5.3-1+deb9u2_armhf.deb  404  Not Found [IP: 151.101.64.204 80]
    E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
    (Reading database ... 15851 files and directories currently installed.)
    Removing machinekit-hal-cross-build-deps:armhf (1.0) ...
    mk-build-deps: Unable to install all build-dep packages
    The command 'bash -c cd /tmp &&                                                                    ./debian/bootstrap -s -p /tmp &&                                              mk-build-deps --build-arch=$(dpkg-architecture -qDEB_BUILD_ARCH)                            --host-arch=${HOST_ARCHITECTURE}                                              -ir -t                                                              "apt-get -o Debug::pkgProblemResolver=yes --no-install-recommends -y"                   ./debian/control' returned a non-zero code: 29
    Traceback (most recent call last):
      File "./scripts/buildcontainerimage.py", line 222, in <module>
        main(args)
      File "./scripts/buildcontainerimage.py", line 171, in main
        args.target[1] if args.target is not None else None)
      File "./scripts/buildcontainerimage.py", line 156, in build_docker_image
        _cwd=self.normalized_path)
      File "/usr/local/lib/python3.7/site-packages/sh.py", line 1503, in __call__
        raise exc
    sh.ErrorReturnCode_29:

If I edit the Dockerfile (scripts/containers/buildsystem/debian/Dockerfile) and change the first `apt-get update` to `apt-get update --fix-missing` (as suggested by the error) the docker container builds and I can bootstrap and configure, but the build command fails. It looks like everything built fine, but when copying files at the end something goes wrong:

    cp: preserving times for 'debian/machinekit-hal-dev//usr/lib/libhal.so': No such file or directory
    dh_install: cp --reflink=auto -a debian/tmp/usr/lib/libhal.so debian/machinekit-hal-dev//usr/lib/ returned exit code 1
    debian/rules:126: recipe for target 'install' failed
    make: *** [install] Error 2
    dpkg-buildpackage: error: fakeroot debian/rules binary-arch gave error exit status 2
    Packages cannot be build because of an error

Any ideas?

the-snowwhite, 2020-12-09:
try:

    scripts/buildcontainerimage.py Debian 9 armhf

cerna, 2020-12-09:
>You need to use number instead of name for $distro try:

Actually, both versions are possible.

>E: Failed to fetch http://security.debian.org/debian-security/pool/updates/main/p/python3.5/libpython3.5_3.5.3-1+deb9u2_armhf.deb  404  Not Found [IP: 151.101.64.204 80]

This is extremely odd. Yes, there is no `deb9u2` version of these packages, which is why it is returning **404**. However, there is a `deb9u3` version. And it uses only standard Debian tools. So it should all work out nicely. I have just tried it and got `Get:33 http://security.debian.org/debian-security stretch/updates/main armhf libpython3.5 armhf 3.5.3-1+deb9u3 [1204 kB]` in my Docker build log.

All I can say at this moment is that maybe there was a networking error. :man_shrugging: 

I was able to build `./scripts/buildcontainerimage.py debian stretch armhf` image without an issue:

```
Successfully built 7db7f8961500
Successfully tagged machinekit-hal-debian-builder-v.armhf_9:latest
Container image build ran successfully to completion!
```

the-snowwhite, 2020-12-09:
Must then be a network issue I also earlier built successfully with: `scripts/buildcontainerimage.py Debian 9 armhf`
~~~
Successfully built dfd6e3020aeb
Successfully tagged machinekit-hal-debian-builder-v.armhf_9:latest
Container image build ran successfully to completion!
~~~

jallwine, 2020-12-09:
I buy that there is something different about my network setup, but it still consistently errors for me when I don't add the --fix-missing flag to the apt-get update command.

The bigger problem was it erroring during the compilation step. I recently updated to macOS Big Sur and my guess is it had something to do with that. I installed a docker update this morning that mentioned Big Sur and minor bugs and it now works.

closed 2020-12-09 by jallwine

cerna, 2020-12-09:
@jallwine,
aha, I don't have any experience with Docker for macOS. As far as I understand, it runs a virtual machine with Linux kernel and the Docker engine on it. I tested it on Linux and `Docker version 20.10.0-beta1, build ac365d7`. I bet that @the-snowwhite did his test on Linux too.

I am still scratching my head why it failed on that `deb9u2` package version. The `apt` ecosystem should always find the latest available version, so either it should fail a lot sooner or don't fail at all.

One of the ways how to test it is to allow Travis CI. Each Travis CI run so far builds the whole image (little wasteful), so if it is able to build it, the problem is probably somewhere else.