machinekit/machinekit-hal #342: Unable to compile in Debian 9

NeoTech, 2021-01-03:
As im trying to compile a run in place instance of machinekit-hal i ran into patrol.

Firstly i get a wall of text with thexe protobuf mal formation errors and then lastly it wont recognize that protobuf is actually installed properly.

This is on Debian 9 (Stretch) - i went down to 9 after i realized getting it running in Debian 10, was completely out of the questions at this point in time.

```
TabError: inconsistent use of tabs and spaces in indentation                                                                                                                                                      --depends_out: protoc-gen-depends: Plugin failed with status code 1.                                                                                                                                              protoc compile dependencies for machinetalk/proto/src/machinetalk/protobuf/message.proto                                                                                                                          Traceback (most recent call last):                                                                                                                                                                                  File "machinetalk/proto/scripts/protoc-gen-depends", line 17, in                                                                                                                                            from google.protobuf.compiler.plugin_pb2 import CodeGeneratorRequest,CodeGeneratorResponse                                                                                                                      File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/compiler/plugin_pb2.py", line 4, in                                                                                                        from google.protobuf import descriptor as _descriptor                                                                                                                                                           File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/descriptor.py", line 47, in                                                                                                                from google.protobuf.internal import cpp_message                                                                                                                                                                File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/internal/cpp_message.py", line 49                                                                                                                  import copyreg                                                                                                                                                                                                                 ^                                                                                                                                                                                                TabError: inconsistent use of tabs and spaces in indentation                                                                                                                                                      --depends_out: protoc-gen-depends: Plugin failed with status code 1.                                                                                                                                              protoc compile nanopb C machinetalk/proto/src/machinetalk/protobuf/message.proto                                                                                                                                                                                                                                                                                                                                                             *************************************************************                                                                                                                                                     *** Could not import the Google protobuf Python libraries ***                                                                                                                                                     *** Try installing package 'python-protobuf' or similar.  ***                                                                                                                                                     *************************************************************                                                                                                                                                                                                                                                                                                                                                              Traceback (most recent call last):                                                                                                                                                                                  File "../lib/python/machinetalk/protobuf/nanopb_generator.py", line 19, in                                                                                                                                  import google.protobuf.text_format as text_format                                                                                                                                                               File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/text_format.py", line 52, in                                                                                                               from google.protobuf.internal import type_checkers                                                                                                                                                              File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/internal/type_checkers.py", line 50, in                                                                                                    from google.protobuf.internal import decoder                                                                                                                                                                    File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/internal/decoder.py", line 86, in                                                                                                          from google.protobuf.internal import encoder                                                                                                                                                                    File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/internal/encoder.py", line 72, in                                                                                                          from google.protobuf.internal import wire_format                                                                                                                                                                File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/internal/wire_format.py", line 38, in                                                                                                      from google.protobuf import descriptor                                                                                                                                                                          File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/descriptor.py", line 47, in                                                                                                                from google.protobuf.internal import cpp_message                                                                                                                                                                File "/home/a/.local/lib/python3.5/site-packages/google/protobuf/internal/cpp_message.py", line 49                                                                                                                  import copyreg                                                                                                                                                                                       [... cut ...]                                                                           python-protobuf is already the newest version (3.0.0-9).                                                                                                                                                          0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.                                                                                                                                                   
--
```

cerna, 2021-01-04:
Could you provide the exact command succession you used? I am able to reach an error state when trying the Debian Stretch, but not on the same step as you.

Run an interactive Docker container:
```
docker run --rm -it debian:buster bash -i
```

And in it:
```
apt update && \
apt upgrade -y && \
apt install -y sudo curl git build-essential fakeroot devscripts && \
echo "ALL ALL = (ALL) NOPASSWD: ALL" >> /etc/sudoers && \
addgroup --gid 1000 mk && \
adduser --uid 1000 --ingroup mk --home /home/mk \
--shell /bin/bash --disabled-password --gecos "" mk && \
su mk -c " \
curl -1sLf \
  'https://dl.cloudsmith.io/public/machinekit/machinekit/cfg/setup/bash.deb.sh' \
  | sudo -E bash && \
cd /home/mk && \
git clone --depth 1 https://github.com/machinekit/machinekit-hal.git machinekit && \
cd /home/mk/machinekit && \
debian/bootstrap && \
mk-build-deps -irs sudo -t \"apt-get -o Debug::pkgProblemResolver=yes --no-install-recommends -y\" && \
cd src && \
./autogen.sh && \
./configure && \
make && \
sudo make setuid && \
../scripts/rip-environment && \
halrun"
```

I get an error on:
```
Syntax checking python script hal_gpio_mcp23017
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 76: ordinal not in range(128)
hal/user_comps/Submakefile:12: recipe for target '../bin/hal_gpio_mcp23017' failed
make: *** [../bin/hal_gpio_mcp23017] Error 1
make: Leaving directory '/home/mk/machinekit/src'
```
Which is new and probably connected to the Python 2 -> Python 3 transfer.

But not your error.

>This is on Debian 9 (Stretch) - i went down to 9 after i realized getting it running in Debian 10, was completely out of the questions at this point in time.

Any reason for it? Usually the Debian community recommends using the latest version available.

NeoTech, 2021-01-04:
Reason i had to to Debian 9, was because machinekit source would fight me on python2 support.
I keep running into issues with it because i need to have developer headers to compile other things.

But yeah, you are most likely right this is Python related.. I keep fighting those dependencies and can get around some of the errors by splicing in packages from backports and ubuntu distro.. which makes this thing really unreliable.

I will close this issue as it cannot lead to a resolution and now i cleaned out that machine and started over on Debian 10, but compiling lcnc instead to get to the -dev package which machinekit archive do not have anymore so i can compile HAL drivers, repackage them and add them into machinekit that way..

closed 2021-01-04 by NeoTech