machinekit/machinekit-hal #343: Missing script.

NeoTech, 2021-01-05:
In my attempts to build this software i found that building the machinekit-cnc part
requires a script. /scripts/build_with_cnc

That is nowhere to be found.. :)

cerna, 2021-01-06:
@NeoTech,
yeah, that's not going to work. That script was removed during the overturn phase to make the EMCApplication work. I do have a plan to make it work again, however it will need some fundamental changes and it has low priority.

**So you have multiple choices:**

1. Use older version of Machinekit-HAL which still has that script
2. Use EMCApplication
3. Patch in hackish way the build artifacts from Machinekit-HAL (mostly shared libraries I think) to the Machinekit-CNC namespace, so you can satisfy dependencies (but I am not sure if some symbols [API] was not slightly changed).

NeoTech, 2021-01-07:
A documentation update would be in order then i suppose. And some instructions on howto get the EMCApplication working..  I noticed it could just be dpkg-buildpackaged as well.

closed 2023-03-29 by NeoTech