machinekit/machinekit-hal #345: correct way to get rt-preempt support

jallwine, 2021-02-02:
What is the correct way to install machinekit-hal with rt-preempt support? When I install the machinekit-hal-rt-preempt package it installs kernel 4.19.0-14-rt-armmp. This doesn't seem to work with the Beaglebone AI (it doesn't seem to have the proper device tree). I'm trying to use kernel 4.19.94-ti-rt-r57, which I have a working device tree for. I'm happy to build from  from source as RIP build for now, but I'm not sure the proper incantations to make that happen either. Any help is appreciated!

cerna, 2021-02-02:
@jallwine,

if you have _rt_preempt_ patched kernel running (and you know it, respective you are sure), then all you need is to install the _machinekit-hal_ package.

The other ones are basically just metapackages for pulling the kernel version too.

jallwine, 2021-02-02:
Gotcha, thanks @cerna!

closed 2021-02-02 by jallwine

jallwine, 2021-02-02:
Forgot we had this discussion in #321

cerna, 2021-02-03:
Yes, generally speaking the packaging leaves lot to be desired. It is not up to the Debian standard as the whole project should be in a lot more packages for it to support tools like `multi-arch` or survive _Lintian_.

So the meta packages are probably not exactly the best solution to this problem, specifically on SBCs where the kernel situation is a lot more shattered than in i686/amd64.