machinekit/machinekit-hal #348: RE: Errors from Running Axis...on BBG and LCD Screen/EMCApplication and MachineKit-Hal

silver2row, 2021-03-05:
Hello,

    ImportError: No module named _hal

This is the error I am receiving while running ./emcapplication/bin/axis and I 'think' this is due to my 'run-in-place' compiled version of machinekit-hal.

I have no way to tell, really, b/c I do not understand what is going on in the world of machinekit-hal and emcapplication for now.

I have successfully built the emcapplication, installed the packages, and can get linuxcnc to run but ./axis is broken due to the errors above.

Seth

P.S. If you are going through a phase of exchanging source and things need improvements, okay. I can start to look for instances of improvements too but I need to have a starting place.

Is the lack of _hal modules due to Python/Python3 conversion?

cerna, 2021-03-08:
@silver2row,
>This is the error I am receiving while running ./emcapplication/bin/axis and I 'think' this is due to my 'run-in-place' compiled version of machinekit-hal.

Yeah, the `Run-In-Place` version of Machinekit-HAL is not supported for EMCApplication right now (or ever will be), it needs to have the modules in the system wide installation location as standard on Debian based system. Theoretically it could be made to work with RIP Machinekit-HAL, but it would require some serious programming gymnastics.

>I have no way to tell, really, b/c I do not understand what is going on in the world of machinekit-hal and emcapplication for now.

The problem is based on where the shared libraries are physically on the computer. You mention `_hal.so`, that is connected to the Python. EMCApplication needs Python2, Machinekit-HAL currently only support Python3, so for Python2 you will need to use older version of Machinekit-HAL.

>I have successfully built the emcapplication, installed the packages, and can get linuxcnc to run but ./axis is broken due to the errors above.

Can you at least open the EMCApplication's AXIS window and run simulation? That should work.

>Is the lack of _hal modules due to Python/Python3 conversion?

I would think so, yes.

silver2row, 2021-03-09:
Hello @cerna ,

Okay. No issue. I will download to an older version of machinekit-hal or do as I said I would in the #general discord channel.

...

For once, I am starting to remember what I learned about `ld` and make/make install. I think once I `sftp'd` the files to the BBB, I forgot to `make` them and install them w/ `make install`.

Still pluggin' at it over here. So, I will update 'every channel' that I have been discussing this issue on and soon.

Seth

P.S. I cannot open the `Axis window` and run simulations w/ `./axis` or via `linuxcnc` and then clicking on the `Axis option` on the LCD Display screen.

closed 2021-03-10 by silver2row