machinekit/machinekit-hal #360: Project URL redirects to a blog

mayl, 2021-12-22:
The current project url is "https://machinekit.io" which redirects to a blog by Michael Haberler (@mhaberler?) which seems to have out of date ssl).  The correct project website seems to be "www.machinekit.io", so I would suggest the github project URL should be updated to reflect url instead.

cerna, 2021-12-22:
@mayl,
problem is, I am not even sure who actually owns and pays for the domain. I think it is @cdsteinkuehler but maybe it is somebody else, the `WHOIS` database is no help.

Michael Haberler left the organization long ago for exciting new things, so it is outside his purview. I think whoever has access to the DNS settings will first have to come forward.

mayl, 2021-12-23:
Hmm, I see fixing this "in general" then might be kind of difficult without help.  

I wasn't clear in my issue, but my original suggestion was a little more specific.  The project URL in the sidebar (see screenshot below) currently points at the personal blog - anyone with write access to this repo should at least be able to change that URL to point to the actual project page I think.
[image]

cerna, 2021-12-23:
This is actually something I can do. So I changed it from `machinekit.io` to `www.machinekit.io`. There seems to be some links on the actual page which have absolute addresses without the `www.` prefix.

mayl, 2021-12-23:
Looks good from here too, the link from GitHub now goes directly to the project page. 

I'll let you decide whether you want to close this or keep it open to track the other domain issues you mentioned.