machinekit/machinekit-hal #368: what are the most stable interfaces working with machinekit-hal?

ebo, 2022-01-18:
I see several, but none of them seem to have been worked on in a couple of years or more.  Maybe I am missing something, but it would be nice to have an interface.

cerna, 2022-01-20:
What is the _interface_ in this instance? A way to access the HAL state? There is the `halcmd` executable, the C API and what was getting the most love recently is the Python API and bindings. (Importunately it is separated into the Pure Python C++ implementation `pyhalmodule` and four `cy*` Cython implementations.)

ebo, 2022-01-20:
Sorry I was not clear.  I meant various GUI's that use machinekit -- 
including EMCApplication, cetus, machineface, machinekit-cnc, and of 
course LinuxCNC and likely others as well.

What I am curious about here is the status of various projects, and 
which ones are the most stable.  Also like you said, even within 
machinkit-hal there are various things that have been "getting the most 
love" while others are likely STARVING for love.  Hmmm I hear a parody 
song in the making...

On Jan 20 2022 4:16 PM, cerna wrote:
> What is the _interface_ in this instance? A way to access the HAL
> state? There is the `halcmd` executable, the C API and what was
> getting the most love recently is the Python API and bindings. [...]

summerlotus513, 2022-02-15:
I'm thinking about this too.
In newer versions of machinekit-hal, there doesn't seem to be a proper GUI to match it.
Even the way of remote communication that was supported in the old version is not available (I tried to use machinetalk+cetus to control it before)