machinekit/machinekit-hal #376: About Building from Source

summerlotus513, 2022-02-15:
when i build [machinekit-hal](https://github.com/machinekit/machinekit-hal) from source, refer to the steps given in the readme
```
git clone https://github.com/machinekit/machinekit-hal.git
cd machinekit-hal
mkdir build
cmake -S . -B ./build
cmake --build ./build
sudo cmake --build ./build --target setuid
cmake --build ./build --target binary_tree_venv
cd build
direnw allow
halrun
```
The command `direnw` can't seem to find it. I can find `direnv` in my system

and, I also got the following error
```
1. Could not find PYPROJECT_BUILD using the following names: pyproject-build
2. Could not find YAPPS2 using the following names: yapps
```

cerna, 2022-02-16:
> The command direnw can't seem to find it. I can find direnv in my system

This is a typo. It surely should be the `direnv`. Thinking about it, maybe the whole mentioning of `direnv` in the README is superfluous. It is useful for multiple builds and directories, to quickly switch between them. But it is a bit harder to set up and for normal user not necessary.

> and, I also got the following error

Do you have the Machinekit dependencies repository installed?

summerlotus513, 2022-02-16:
yes i use direnv
The error is because I am using ubuntu18.04, I have upgraded my system to 20.04, everything is ok, I can already use halrun.
By the way, after machinekit-hal is separated from machinekit, the whole architecture is clearer, which is great. 
But I still want to use the GUI program to control (no way, the teacher asked us to do this),
I have tested the remote control scheme on machinekit before (using cetus, configserver + mkwrapper) and am very satisfied. Using machinekit-hal I can't seem to find any matching gui programs (EMCApplication, cetus neither seem to work), and even with configserver removed from machinekit-hal, mkwrapper doesn't work (ModuleNotFoundError: No module named 'linuxcnc').

So, what kind of GUI program should I use locally? What should I do if I want to use remote control?
Thank you very much~
@cerna