machinekit/machinekit-hal #380: Problems with source code compilation and installation

jack9603301, 2022-03-20:
I want to use manual way to compile the whole machinekit-hal + EMCApplication, but I don't know the right way, I can finish the compilation (either deb package or make install), but can't export machinekit-hal to EMCApplication frontend, please tell me how Do? If I don't provide --with-hal, he can detect LOCAL HAL, but there seems to be a compilation error related to halui when compiling, if possible, please tell me the complete list of compilation instructions.

RVillatore, 2022-07-17:
@jack9603301 right now emcapplication is not updated to run with current machinekit-hal.
In order to compile it, you need to fix dependencies and linker issues.
I'm working on it in my spare time, but might take a wile until I'm done.
Right now I can compile emcapplication for bullseye and foi bionic. I'm also able to generate packages for bionic but havent tested the compiled software yet.

RVillatore, 2022-07-18:
@jack9603301 I 've managed to finish my very bad fix to compile emcapplication for the newest machinekit-hal releaso (bullseye).
Please see my git repo: https://github.com/RVillatore/EMCApplication/tree/machinekit-hal-last-update
You just need to follow the instructions under docker to create a container and build the packages.
Just to let you know, I've only compiled the application, but haven't tested it