machinekit/machinekit-hal #381: About whether machinekit-hal supports qtquickvcp

summerlotus513, 2022-08-20:
hello developer,
The new version of machinekit-hal adjusts the entire architecture I think is worthy of recognition.
In the previous machinekit, I have used QtQuickVcp to create some demos, so that I can control machinekit cross-platform.
But in the new version of machinekit-hal, I didn't find relevant support, or I don't know how to configure it.
Can you tell me how to use Vcp to connect to the controller remotely? How should I configure it?

the-snowwhite, 2022-10-10:
@summerlotus513  the QtQuickVcp stuff only works in the machinekit-hal machinekit-cnc combination python 2 versions.
If you want to build machinekit packages with qtquickvcp support you have to checkout some older revisions of machinekit hal and machinekit-cnc.
machinekit-hal: commit 120c9f765cf12bcd337b66308b64b3146a26ec8d
machinekit-cnc: commit 0a19e8628ecb5c83f2a810747be31b1bf77271e7

summerlotus513, 2023-05-06:
> @summerlotus513 the QtQuickVcp stuff only works in the machinekit-hal machinekit-cnc combination python 2 versions. If you want to build machinekit packages with qtquickvcp support you have to checkout some older revisions of machinekit hal and machinekit-cnc. machinekit-hal: commit [120c9f7](https://github.com/machinekit/machinekit-hal/commit/120c9f765cf12bcd337b66308b64b3146a26ec8d) machinekit-cnc: commit 0a19e8628ecb5c83f2a810747be31b1bf77271e7

okk~thanks.
I still hope that Machinekit-hal can be operated remotely, just like mkwrapper and machinekitalk. I wonder if the current Machinekit-Hal may support this in the future?

the-snowwhite, 2023-05-09:
@summerlotus513 Since the Auther of machinekitalk and machinekit Client / qtquickvcp clearly has abandoned his work
there is only chance for a revival if a new developer steps up to the plate.
A pro pro remote operation I don't know if you are aware that current machinekit-hal has an updated and working ros interface ? 
https://github.com/the-snowwhite/hal_ros_control