machinekit/machinekit-hal #398: Install Machinekit on Debian Bookworm

dpflexvn, 2024-05-22:
Hi,
I followed install instruction: 
1. Install with repository: http://www.machinekit.io/docs/getting-started/installing-packages/
2. Build from source: http://www.machinekit.io/docs/developing/machinekit-developing/
All 2 options did not work.
Any one has experience to install Machinekit on Debian Bookworm ?.
I am using kernel 6.1.0-18-rt-amd64
Thank you.

jallwine, 2025-01-23:
I also am wondering if anyone has built this for Debian Bookworm?

yishinli, 2025-02-01:
I've sucessfuly built Machinekit-hal from source with RPi5 and Raspbian Bookworm 32-bit. 
Here's my [revised files](https://github.com/yishinli/machinekit-hal/commit/f0d63d275ab30072d4ac501b84024405fb82cbbe).
Since, I do not have enough knowledge to add an architecture(RPi5 w/ Bookworm 32-bit) to CMakeList.txt, I would be appreciate if someone can advise/help implementing it.